Koenkk/zigbee2mqtt #12459: Container doesnt start with fatal error

TheQue42, 2022-05-12:
### What happened?

Updated with portainer, havent made any config changes in a long time, and now container didnt start.
I found and commented on an old closed issue, but I dont know if that is even noticed?

```
Using '/app/data' as data directory

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xffaa18ac
```

After some troubleshooting, I noticed that it WILL start, if I used _privileged_ mode, which I would rather not.

I am running in plain docker, on an RPI4.
Linux gizmo 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

### What did you expect to happen?

Normal startup

### How to reproduce it (minimal and precise)

It was running fine as last log shows:

info  2022-05-08 19:45:51: Logging to console and directory: '/app/data/log/2022-05-08.19-45-49' filename: log.txt
info  2022-05-08 19:45:52: Starting Zigbee2MQTT version 1.24.0 (commit #7a2ddf2)
info  2022-05-08 19:45:52: Starting zigbee-herdsman (0.14.20)
info  2022-05-08 19:45:52: zigbee-herdsman started (resumed)
info  2022-05-08 19:45:52: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
info  2022-05-08 19:45:52: Currently 19 devices are joined:
info  2022-05-08 19:45:52: Vibration Sensor (0x00158d0005431e1f): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
info  2022-05-08 19:45:52: DraperiSensor (0x00158d0001e4bb73): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-05-08 19:45:52: IkeaSocket (0x5c0272fffe77c5a6): E1603/E1702/E1708 - IKEA TRADFRI control outlet (Router)
info  2022-05-08 19:45:52: DM Verkstadsdörren (0x00158d000689aae2): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-05-08 19:45:52: Källarförrådet (0x00158d00056e4d94): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-05-08 19:45:52: Rumstemp Elliott (0x00158d000588ccd5): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-05-08 19:45:52: IkeaMS (0x588e81fffe28a023): E1525/E1745 - IKEA TRADFRI motion sensor (EndDevice)
info  2022-05-08 19:45:52: Ikea DimmerButton (0x60a423fffef1be96): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2022-05-08 19:45:52: AltanLampaH (0x84fd27fffea8ba55): LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (Router)
info  2022-05-08 19:45:52: AltanLampaV (0x84fd27fffea8e3f0): LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (Router)
info  2022-05-08 19:45:52: IkeaExtender (0x84fd27fffe4797f7): E1746 - IKEA TRADFRI signal repeater (Router)
info  2022-05-08 19:45:52: DM Källarförrådet (0x00158d000700591d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-05-08 19:45:52: DM Gillestugedörren (0x00158d0006f0d769): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-05-08 19:45:52: MikroLampan (0x84fd27fffe748d23): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
info  2022-05-08 19:45:52: Mikro LampDimmer  (0x50325ffffe634d58): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2022-05-08 19:45:52: Golvtemp Duschen (0x00158d0006ebfbc7): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-05-08 19:45:52: FrysTemp (0x00158d000759b0a3): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-05-08 19:45:52: UteLampa Förrådet (0x50325ffffe2bb4f5): LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (Router)
info  2022-05-08 19:45:52: Elliotts Nattlampa (0x2c1165fffecfbd60): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2022-05-08 19:45:52: Zigbee: disabling joining new devices.
info  2022-05-08 19:45:55: Connecting to MQTT server at mqtt://vimes.disc-world.se
info  2022-05-08 19:45:55: Connected to MQTT server

// And this goes on until it shutsdown today.

Then portainer updated to :latest, and it refused to start.

The old container actually ran in --network=host, but now its running in a normal network namespace with -p X:Y, but using --privileged :-( 

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26720700

### Adapter

Conbee II

### Debug log

Using '/app/data' as data directory

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xffaa18ac

Impact123, 2022-05-12:
> -t=target_release

<https://manpages.debian.org/bullseye/apt/apt.8.en.html>

> Backports are packages taken from the next Debian release (called "testing"), adjusted and recompiled for usage on Debian stable.

<https://backports.debian.org>

buster: `2.3.3-4`
buster-backports: `2.5.1-1`
<https://packages.debian.org/en/buster/libseccomp2>
<https://packages.debian.org/en/buster-backports/libseccomp2>

TheQue42, 2022-05-12:
Superb! Running fine now, without privileged, and with user=1001+dialup.

I guess its about time to upgrade to bullseye soon...

The strange thing is why this issue popped up now, when the container must have been created ages ago. (That is at least my excuse for missing the instruction on libseccomp ;-) ) 

Once again, thanks for the speedy help!

Impact123, 2022-05-12:
> why this issue popped up now

Because you updated :)
[The base image has changed from `node:14*` to `node:16*` in `1.25.0`](https://github.com/Koenkk/zigbee2mqtt/commit/b515c3213f80a3228ca8747b25896377e998bb22#diff-f34da55ca08f1a30591d8b0b3e885bcc678537b2a9a4aadea4f190806b374ddc)
[Also see `1.25.0`'s `Notes` for background](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.25.0)

TheQue42, 2022-05-13:
Okey, so its all up to RTFM, but I cant help being confused, since the 1.25.0 was made on April 1st, and I did read the whole Matter2MQTT stuff, but my watchtown container didnt notify my of an update, even though I think I must have updated zwjs at least once during this time.
Well, whatever! I wont make any promises I cant be sure to keep, but I will _try_ to RTFM better in the future :-)

closed 2022-06-14 by TheQue42