Koenkk/zigbee2mqtt #12461: Aqara roller shade companion E1 in Zigbee2MQTT intermittently errors out to MAC Transaction Expired (240)

nitro41992, 2022-05-12:
### What happened?

I setup the Aqara roller shade companion E1 ([ZNJLBL01LM](https://www.zigbee2mqtt.io/devices/ZNJLBL01LM.html#xiaomi-znjlbl01lm)) on Z2M a couple days ago and initially everything seemed to be working properly. More recently, every now and then, the roller shade stops working with the error (Data request failed with error: 'MAC transaction expired' (240))'

I have other devices using Z2M that are working without any issues so the problem seems to be isolated only to this device. Also to note, after a while, the device works again as intended regardless of the errors still showing up in the Z2M logs. Has anyone experienced anything similar in Z2M?

Edit: Using the Sonoff Zigbee 3.0 with the firmware upgrade

### What did you expect to happen?

The Roller Shade to pull the shades up, down or stop rolling based on the command

### How to reproduce it (minimal and precise)

Set the state in exposes to OPEN,  CLOSE, or STOP

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Zigbee2MQTT:error 2022-05-10 16:55:53: Error: Read [REDACTED By ME] /1
genAnalogOutput(["presentValue{"sendWhen":"immediate","timeout":10000,"
disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,
"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,
"transactionSequenceNumber":null,"writeUndiv":false}) failed 
(Data request failed with error: 'MAC transaction expired' (240))

Zigbee2MQTT:error 2022-05-10 16:56:21: Publish 'set' 'state' to 'window_roller_shade' failed: 
'Error: Write [REDACTED By ME] /1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery"
:false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,
"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed 
(Data request failed with error: 'MAC transaction expired' (240))'

closed 2022-06-19 by github-actions[bot]

lluisd, 2023-03-04:
@nitro41992  did you solved it? are you using docker over synology nas? I have the same problem with same Aqara roller

SyBeer, 2023-11-02:
I have similar problem. From times to times one from my 4 x Aqara roller E1 [ZNJLBL01LM](https://www.zigbee2mqtt.io/devices/ZNJLBL01LM.html#xiaomi-znjlbl01lm)) show the same errors. 
My setup: raspberry Pi 4, Sonoff USB 3.0 on extended usb cable, SSD on USB3.0, HomeAssistant, Zigbee2mqtt.