Koenkk/zigbee2mqtt #12463: [New device support]: Support for Advance Arrival Sensor

iharyadi, 2022-05-12:
### Link

https://community.home-assistant.io/t/zigbee-arrival-sensor-for-car/382529

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b001d866887","nwkAddr":14279,"manufId":0,"manufName":"KMPCIL","powerSource":"Battery","modelId":"tagv1","epList":[1],"endpoints":{"1":{"profId" :260,"epId":1,"devId":12,"inClusterList":[0,1,3,15,32],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"tagv1","manufacturerName":"KMPCIL"}},"genPollCtrl":{"attribut es":{"checkinInterval":14400}},"genBinaryInput":{"attributes":{"presentValue":0,"statusFlags":0}},"genPowerCfg":{"attributes":{"batteryVoltage":30}},"msTemperatureMeasurement":{"attributes" :{"measuredValue":700}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c273ee","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b002 4c273ee","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c273ee","endpointID":1},{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c273ee","en dpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":0,"maxRepIntval":10,"repChange":1},{"cluster":15,"attrId":85,"minRepIntval":0,"maxRepIntval":300,"repChange":1} ,{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"appVersion":4,"stackVersion":73,"hwVersion":3,"dateCode":"130925","zclVersion":1,"interviewC ompleted":true,"meta":{"configured":610118766},"lastSeen":1652396067237,"defaultSendRequestWhen":"active"}

### Comments

Hi Koenkk,  

I make my own Arrival Sensor.  It is basically a presence sensor designed specifically to be used in our car.  The sensor add a few capability to help secure the car.  This is unique to my sensor.  I have commercialized the sensor.   I would love to have the driver included in the core Zigbee2MQTT core.  It will make the experience better for Zigbee2MQTT user in using the new Arrival Sensor.

I have created a external driver.  I and a few member from HA community has tested the driver.  Thank you for your hard work with Zigbee2MQTT, the external driver work nicely.   

I would like to have you consider the external driver that I made to be included in Zigbee2MQTT.  I am just a beginner in writing the driver.  I am open to suggestion to change the driver to fit the Zigbee2MQTT model better.  Please just let me know any changes that you like me to make.   

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const reporting = require('zigbee-herdsman-converters/lib/reporting');

const {
    precisionRound, mapNumberRange, isLegacyEnabled, toLocalISOString, 
    numberWithinRange, hasAlreadyProcessedMessage,
    calibrateAndPrecisionRoundOptions, addActionGroup, postfixWithEndpointName, 
    getKey,batteryVoltageToPercentage, getMetaValue,
} = require('zigbee-herdsman-converters/lib/utils');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const constants = require('zigbee-herdsman-converters/lib/constants');
const libColor = require('zigbee-herdsman-converters/lib/color');
const utils = require('zigbee-herdsman-converters/lib/utils');

const myoptions = {
	options:{
	presence_timeout_dc: () => { return exposes.numeric('presence_timeout_dc').withValueMin(60).withDescription('Time in seconds after which presence is cleared after detecting it (default 60 seconds) while in DC.');},
	 presence_timeout_battery: () => { return exposes.numeric('presence_timeout_battery').withValueMin(120).withDescription('Time in seconds after which presence is cleared after detecting it (default 420 seconds) while in Battery.');},
	},
};

function handlePresence(model, msg, publish, options, meta) {
        const useOptionsTimeout_battery = options && options.hasOwnProperty('presence_timeout_battery');
        const timeout_battery = useOptionsTimeout_battery ? options.presence_timeout_battery : 420; // 100 seconds by default

        const useOptionsTimeout_dc = options && options.hasOwnProperty('presence_timeout_dc');
        const timeout_dc = useOptionsTimeout_dc ? options.presence_timeout_dc : 60;

        const mode = meta.state? meta.state['power_state'] : false;

        const timeout =  mode ?  timeout_dc : timeout_battery;
            // Stop existing timer because motion is detected and set a new one.
        clearTimeout(globalStore.getValue(msg.endpoint, 'timer'));
        const timer = setTimeout(() => publish({presence: false}), timeout * 1000);
        globalStore.putValue(msg.endpoint, 'timer', timer);

	return {presence:true};
};

const myconverters = {
    STS_PRS_251_binary_input: {
        cluster: 'genBinaryInput',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
	    const payload = handlePresence(model, msg, publish, options, meta);
            if (msg.data.hasOwnProperty('presentValue')) {
              const presentValue =  msg.data['presentValue']
              payload.power_state = (presentValue & 0x01)> 0;
              payload.occupancy = (presentValue & 0x04) > 0;
              payload.vibration = (presentValue & 0x02) > 0;
	    }
	    return payload;
	},
    },
    STS_PRS_251_power_presence: {
        cluster: 'genPowerCfg',
        type: ['attributeReport', 'readResponse'],
        options: [myoptions.options.presence_timeout_dc(),myoptions.options.presence_timeout_battery()],
        convert: (model, msg, publish, options, meta) => {
            const payload = handlePresence(model, msg, publish, options, meta);
	    if (msg.data.hasOwnProperty('batteryVoltage')) {
                // Deprecated: voltage is = mV now but should be V
                payload.voltage = msg.data['batteryVoltage'] * 100;

                if (model.meta && model.meta.battery && model.meta.battery.voltageToPercentage) {
                    payload.battery = batteryVoltageToPercentage(payload.voltage, model.meta.battery.voltageToPercentage);
                }
            }
            return payload;
        },
    },
};

const device = {
	zigbeeModel: ['tagv1'],
        model: 'KMPCIL-tag-001',
        vendor: 'KMPCIL',
        description: 'Arrival sensor',
        fromZigbee: [myconverters.STS_PRS_251_power_presence, myconverters.STS_PRS_251_binary_input, fz.temperature],
        exposes: [e.battery(), e.presence(), exposes.binary('power_state',exposes.access.STATE,true,false), e.occupancy(), e.vibration(), e.temperature()],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_1500_2800'},},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            for (const cluster of ['msTemperatureMeasurement','genPowerCfg', 'genBinaryInput']){
               await utils.sleep(2000);
               await endpoint.bind(cluster, coordinatorEndpoint);
            }

            await utils.sleep(1000);
	    const p = reporting.payload('batteryVoltage', 0, 10, 1);
            await endpoint.configureReporting('genPowerCfg', p);	    
           
            await utils.sleep(1000);
            const p2 = reporting.payload('presentValue', 0, 300, 1);
            await endpoint.configureReporting('genBinaryInput', p2);
            
            await utils.sleep(1000);
	    await reporting.temperature(endpoint);
	    await endpoint.read('genBinaryInput', ['presentValue']);
        },
};

module.exports = device;
```

Koenkk, 2022-05-13:
@iharyadi could you create a pr for https://github.com/Koenkk/zigbee-herdsman/pulls ?

iharyadi, 2022-05-13:
@Koenkk,  I will try to make the PR.  May I have your help so that I do not mess up the architecture.  I think I can handle merging the device object I have done it before.

How do you like to merge the supporting code such as myconverters , myoptions  and handlePresence. Of course,  I will try to figure out more appropriate name for it.  I just am not too familiar how do you want to merge them to the framework.

iharyadi, 2022-05-15:
I managed to file PR.  I think this  issue can be closed.

closed 2022-05-15 by iharyadi