Koenkk/zigbee2mqtt #12468: (Data request failed with error: 'No network route' (205))

CReimer, 2022-05-13:
### What happened?

None of my Zigbee devices react to any command

### What did you expect to happen?

Something 😄 

### How to reproduce it (minimal and precise)

Try to change the state of any device on the Zigbee network.

### Zigbee2MQTT version

1.25.1-dev commit: 24cfdf4

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8689925/log.txt)

qweluke, 2022-06-01:
I'm having the sam issue! non of my power plugged devices are working correctly...

```
{"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:info  2022-06-01 11:46:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Światło w sypialni' failed: 'Error: Command 0xa4c1389125274327/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Światło w sypialni"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2022-06-01 11:46:43: Publish 'set' 'state' to 'Światło w sypialni' failed: 'Error: Command 0xa4c1389125274327/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:info  2022-06-01 11:46:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Światło w sypialni' failed: 'Error: Command 0xa4c1389125274327/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Światło w sypialni"},"type":"zigbee_publish_error"}'
```

CReimer, 2022-06-01:
I switched to Slaesh's CC2652RB stick and sent the Sonoff stick back. Slaesh's stick works without problems.

CReimer, 2022-07-03:
There's a lot more info in #12774. So I think we can close this issue und continue there

closed 2022-07-03 by CReimer

Bimboboy007, 2023-05-29:
Gleiches Problem manchmal funktionieren die Geräte als Router und manchmal steigen die Geräte einfach aus No network route' (205)