Koenkk/zigbee2mqtt #12471: Hive motion sensor MOT003 errors in logs, although device works fine

dnedeltchev, 2022-05-14:
### What happened?

Hive motion sensor MOT003 pairs fine and works fine but it has an exclamation warning sign. If you click on the reconfigure button, the following error is displayed in the logs

Failed to configure '0x0xxxxxxxxxxxxxx', attempt 1 (Error: Bind 0x0xxxxxxxxxxxxxx/6 msTemperatureMeasurement from '0x0xxxxxxxxxxxxxx/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

### What did you expect to happen?

Expectation is no errors and warnings.

### How to reproduce it (minimal and precise)

Pair the device and look at the icon in z2m which shows the exclamation. Click on reconfigure and monitor logs.

### Zigbee2MQTT version

 1.25.1-1

### Adapter firmware version

20210708

### Adapter

zzh!

dnedeltchev, 2022-05-16:
> Try to wakeup the device right before sending the configure request.

I have tried this but the results are mixed. Sometimes I would get
reconfigure successful and sometimes I would get the error which I posted.
I have tried with join enabled and disabled. Regardless of what I try the
interview failed and red exclamation mark remain. The devices however do
work as expected. I've had this for a very long time. I'd say probably more
than a year. I can't remember ever not seeing this. Although it's possible
in the very beginning when I started using z2m and hadn't noticed it. It's
not causing any functional issue though so I thought it was something
cosmetic. The firmware build date is 20181205. I have also tried to pair
the sensor with the hive app hoping that the firmware will be updated.
There is no way that I have found to tell what the latest firmware is. Hive
support are a bit rubbish and don't share this info. But the firmware has
remained the same for years.

Koenkk, 2022-05-16:
Can you provide the herdsman debug log when pairing the device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

dnedeltchev, 2022-05-17:
Got the logs. But is it safe to post them here unreacted?

Koenkk, 2022-05-18:
Just the logs starting from the point where you pair the device until it fails doesn't contain any sensitive information (besides friendlynames and the ieeaddr)

dnedeltchev, 2022-05-18:
Device in question has ieeaddr 0x00124b0012facc49. I have tried to trim the logs and leave only lines that have this ID. I am not sure if I managed to catch the exact moment of the pairing in the logs but I do see errors relating to this ID saying pairing failed because it "node descriptor request failed for '0x00124b0012facc49', attempt 6". Let me know if I should do the debug trace again
[acc49-2.log](https://github.com/Koenkk/zigbee2mqtt/files/8718654/acc49-2.log)
[acc49-1.log](https://github.com/Koenkk/zigbee2mqtt/files/8718655/acc49-1.log)
.

Koenkk, 2022-05-21:
Interview error should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Regarding the configure error, could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['MOT003'],
    model: 'MOT003',
    vendor: 'Hive',
    description: 'Motion sensor',
    fromZigbee: [fz.temperature, fz.ias_occupancy_alarm_1_with_timeout, fz.battery, fz.ignore_basic_report,
        fz.ignore_iaszone_statuschange, fz.ignore_iaszone_attreport],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(6);
        const binds = ['genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        // await reporting.temperature(endpoint);
        await endpoint.read('genPowerCfg', ['batteryPercentageRemaining']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.temperature(), e.occupancy(), e.battery_low(), e.tamper(), e.battery()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

To test, first force remove the device, restart z2m and pair it again.

dnedeltchev, 2022-05-21:
I did the test with external converters exactly as you described but the first time I tried to reconfigure the device I got the same error. I restarted z2m a second time and after that I got a success notification. Shall I leave the install as is or do further troubleshooting?

> On 21 May 2022, at 07:47, Koen Kanters ***@***.***> wrote:
>

Koenkk, 2022-05-22:
If interview works now and reconfigure didn't initially that is fine (since z2m will reattempt the configures but not the interviews)

dnedeltchev, 2022-05-22:
Thank you!
I will wait for the next official release and test the pair issue.

dnedeltchev, 2022-06-02:
Issue has been fixed in 1.25.2-1. Closing.

closed 2022-06-02 by dnedeltchev