Koenkk/zigbee2mqtt #12473: Docker Crashing with RangeError: Offset is outside the bounds of the DataView

jamesonuk, 2022-05-14:
### What happened?

Error is encountered which kills the process and stops the docker container zigbee2mqtt is running in

### What did you expect to happen?

error is handled or doesn't occur

### How to reproduce it (minimal and precise)

This has happened twice now.   There is nothing I can see in the logs around this just the error and then the container stopping.

This has only started happing since update to 1.25.1

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

RangeError: Offset is outside the bounds of the DataView
    at DataView.getUint8 (<anonymous>)
    at parseReceivedDataNotification (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:306:30)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
    at Driver.emit (node:events:527:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
    at Parser.emit (node:events:527:28)
    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/slip/src/slip.js:189:18)

jamesonuk, 2022-07-16:
Bumping again

Koenkk, 2022-07-17:
@chrishae could you take a look at this?

ChrisHae, 2022-07-20:
Are you on the newest version? And can you please give an extract from the zigbee-herdsman debug log when the error happens.
https://www.zigbee2mqtt.io/guide/usage/debug.html

jamesonuk, 2022-07-20:
I am now running [1.26.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.26.0) commit: [cd23aea](https://github.com/Koenkk/zigbee2mqtt/commit/cd23aea)
I will need to set this up again but I am fairly sure that as the docker container died there were no logs other than what is mentioned above.

It is intermittent so might take a while for it to trigger again

ChrisHae, 2022-07-21:
I'm asking because it should be fixed around january 8th in zigbee-herdsman. That should be z2mqtt 1.23.0.
The line numbers in the debug output you posted in your initial comment don't match with the actual code.

Could be that an error still exists but a more detailed and up to date excerpt of the debug log would be usefull.

@koennkk The zigbee-herdsman code from the master branch is in the z2mqtt master branch or does he need to check out the dev branch or something like that?

jamesonuk, 2022-07-21:
> I'm asking because it should be fixed around january 8th in zigbee-herdsman. That should be z2mqtt 1.23.0. The line numbers in the debug output you posted in your initial comment don't match with the actual code.

I was definitely on 1.25.1 when I started seeing these errors.   It hasn't happened since yesterday but I have debug logging enabled if it occurs again.   Will keep an eye on this

Koenkk, 2022-07-21:
@jamesonuk good, please provide the error log when on z2m 1.26.0

closed 2022-08-30 by github-actions[bot]