Koenkk/zigbee2mqtt #12475: [Feature request]:

jnxxx, 2022-05-14:
### Is your feature request related to a problem? Please describe

No problem as such, just requesting to expand an existing device with more attributes.

### Describe the solution you'd like

I have just extended my Danfoss Icon regulator with a Zigbee module, and want to give it a try with Z2M instead of the Ally gateway.
I notice some features missing though.

**Floor temperature**
The simplest one is thermostats with floor sensors, the floor temperature is not read by Z2M.
It can be read through Dev console though. Cluster: Temperature (0x0402), attribute measuredValue (0).
I have two types of thermostates, with and without floor sensor:
modellId: 0x8031    Read result of 'msTemperatureMeasurement': {"measuredValue":2219}
modellId: 0x8030    Read result of 'msTemperatureMeasurement': {"measuredValue":-32768}

**Scheduling**
Additionally it would be neat to have scheduling functionality within the regulator. It seems generally this has been left out, and put in manual mode instead.
Likely some people want to control schedule with Home Assistant, but I would prefer to set the schedule in the regulator and use Home Assistant primarily for logging/monitoring.

From the Dev console these are working, just not made made available:

hvacThermostat, programingOperMode R/W (Bit 0 = Simple setpoint/manual (0) or schedule (1))
hvacThermostat, UnoccupiedHeatingSetpoint

Not working from Dev console, but should be available according to [this](https://drive.google.com/file/d/1n5MqOSaBiPYZAXqRsyglrGSwGzOSjZUA/view?usp=sharing):

Get and set weekly schedule
Thermostat Cluster (0x0201), commands 2 & 1
It is unknown what the payload should look like, but hopefully a get will reveal.

hvacThermostat, DanfossFloorMinSetpoint
hvacThermostat, DanfossFloorMaxSetpoint

hvacThermostat, danfossPreheatStatus
hvacThermostat, DanfossScheduleTypeUsed     (to set regular schedule or vacation)

### Describe alternatives you've considered

The alternative is of course to do scheduling using Home Assistant, and let each thermostat be in manual mode.
Or add the regulator to an Ally gateway instead (but I really like to have the output_status pushed over, so I can do some statistics on which rooms are actually heated, how much and when).

### Additional context

Regulator model:
Zigbee: 0x0200
Danfoss: 088U1142

Thermostat with floor sensor:
Zigbee: 0x8031
Danfoss: 088U1082

Thermostat without floor sensor:
Zigbee: 0x8030
Danfoss: 088U1081

nagydavid, 2022-11-06:
I will look into for the floor heating setpoint, during the coming weeks, since I just extended my network as well, but about two thing:
The scheduling protocol is a bit harder to achieve through the regulator, and through the UI, climate card, etc, you won't be able to add the unoccupied heating setpoint anyway. Scheduling can be done by the scheduling card from hacs in a much intuitive way, rather than set extra additional setpoint values , which anyway HA will not make use of.

nagydavid, 2022-11-06:
@Koenkk can you pls reopen this request. Thanx

closed 2022-12-15 by github-actions[bot]

resk10, 2023-08-29:
@nagydavid did you ever get around to looking at this? I am also missing the IR floor temperature as @jnxxx requested initially.