Koenkk/zigbee2mqtt #12481: some problems went wrong with TUYA TS0002

huahuazi, 2022-05-15:
### What happened?

when I click L1 on， then L2 will on, too.
when I click L1 off， then L2 will off, too.

I cannot control one of them in ha
However, when I press the switch on the wall physically, the switch works normally, and L1 and L2 will not be turned on or off at the same time. However, when I use ha to switch L1 and L2, they will be turned on and off at the same time. The two switches like be linked.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

sonoff

### Debug log

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Asia/Shanghai",
    "os_name": "Linux",
    "os_version": "5.10.108",
    "supervisor": "2022.05.1",
    "host_os": "Home Assistant OS 7.6",
    "docker_version": "20.10.9",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "xiaomi_gateway3": {
      "version": "2.0.9",
      "requirements": [
        "zigpy>=0.33.0"
      ]
    },
    "auto_backup": {
      "version": "0.10.1",
      "requirements": []
    },
    "esxi_stats": {
      "version": "0.6.4",
      "requirements": [
        "pyvmomi==7.0.3"
      ]
    },
    "colorfulclouds": {
      "version": "1.2.5",
      "requirements": []
    },
    "nodered": {
      "version": "1.0.8",
      "requirements": []
    },
    "xiaomi_miot": {
      "version": "0.5.7",
      "requirements": [
        "construct==2.10.56",
[... 522 lines cut ...]
                  "topic": "zigbee2mqtt/0xa4c138a75a17fa91/l2/set"
                },
                {
                  "payload": "ON",
                  "qos": 0,
                  "retain": false,
                  "time": "2022-05-15T06:20:06.041813+00:00",
                  "topic": "zigbee2mqtt/0xa4c138a75a17fa91/l2/set"
                },
                {
                  "payload": "OFF",
                  "qos": 0,
                  "retain": false,
                  "time": "2022-05-15T06:20:07.011644+00:00",
                  "topic": "zigbee2mqtt/0xa4c138a75a17fa91/l2/set"
                }
              ]
            }
          ]
        }
      ],
      "triggers": []
    }
  }
}

huahuazi, 2022-05-17:
> Should be fixed now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

When will this version be updated to the official version? not the dev version

Impact123, 2022-05-17:
[Judging from the release history so far](https://github.com/Koenkk/zigbee2mqtt/tags) I'd say `June 1`

closed 2022-06-25 by github-actions[bot]

BogdanValea, 2022-07-12:
hello all, any news on when this issue is gonna be solved in the master branch ? thanks all for the great effort.

dmitry-pervushin, 2022-10-24:
Apparently not in 1.28.0-1 ? :)