Koenkk/zigbee2mqtt #12482: [New device support]: Unsupported TS011F _TZ3000_cjrngdr3

evsmorodin, 2022-05-15:
### Link

https://aliexpress.ru/item/1005004128359494.html

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xa4c13806778eac9d","nwkAddr":36677,"manufId":4417,"manufName":"_TZ3000_cjrngdr3","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":0,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":76,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1652602900150,"defaultSendRequestWhen":"immediate"}

### Comments

The device does not show any parameters and they cannot be controlled

[image]
[image]
[image]
[image]

Koenkk, 2022-05-16:
`_TZ3000_cjrngdr3` is already supported in the latest dev.

closed 2022-05-16 by Koenkk

evsmorodin, 2022-05-31:
I added a converter for this device

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const libColor = require('zigbee-herdsman-converters/lib/color');
const utils = require('zigbee-herdsman-converters/lib/utils');

const definition = {
    fingerprint: [ {modelID: 'TS011F', manufacturerName: '_TZ3000_cjrngdr3'},],
    model: 'PJ-MINI-ZSW01',
    vendor: 'Dongguan Pinja Technology Co., Ltd.',
    description: 'Smart plug (with power monitoring by polling) for installation in a socket',
    ota: ota.zigbeeOTA,
    fromZigbee: [fz.on_off,fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory,
        fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    options: [exposes.options.measurement_poll_interval()],
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage'),
        exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
            .withDescription('Plug LED indicator mode'), e.child_lock()],
    onEvent: tuya.onEventMeasurementPoll,
};

module.exports = definition;
```
Sensor data is displayed, the state changes, everything is fine with this... But link quality is always 0 lqi...

What could be wrong?

Koenkk, 2022-05-31:
@evsmorodin what adapter are you using?

evsmorodin, 2022-06-01:
I took it from here [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js)

I tried `model: 'TS011F_plug_1',` and `model: 'TS011F_plug_3',`  but the result is the same

Koenkk, 2022-06-01:
@evsmorodin I mean USB coordinator (e.g. conbee, zzh, sonoff adapter)

evsmorodin, 2022-06-02:
Today, after updating to 1.25.2, the link quality value appeared: 33 lqi. I will continue to monitor the changes, thanks