Koenkk/zigbee2mqtt #12488: CCT5010-0001/550B1012 no control

gbjmkm, 2022-05-15:
### What happened?

I have a 550B1012, in Z2M it detected as CCT5010-0001. I assume it is the same device sold under different name from SE.
I am not able to control it within Z2M or HA.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB stick

### Debug log

debug 2022-05-15 13:57:26Received Zigbee message from '0x804b50fffecd2202', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
info 2022-05-15 13:57:26MQTT publish: topic 'zigbee2mqtt/0x804b50fffecd2202', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":1,"dimmer_mode":null,"linkquality":63,"state":"OFF"}'
debug 2022-05-15 13:57:26Received Zigbee message from '0x804b50fffecd2202', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 3 with groupID 0
info 2022-05-15 13:57:26MQTT publish: topic 'zigbee2mqtt/0x804b50fffecd2202', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":1,"dimmer_mode":null,"linkquality":63,"state":"OFF"}'
warning 2022-05-15 13:57:31Failed to ping '0x804b50fffecd2202' (attempt 1/2, Read 0x804b50fffecd2202/242 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32270 - 242 - 207 - 0 - 1 after 10000ms))
warning 2022-05-15 13:57:57Failed to ping '0x804b50fffecd2202' (attempt 2/2, Read 0x804b50fffecd2202/242 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32270 - 242 - 208 - 0 - 1 after 10000ms))
debug 2022-05-15 13:58:00Active device '0x804b50fffecd2202' was last seen '0.58' minutes ago.
debug 2022-05-15 13:59:30Received MQTT message on 'zigbee2mqtt/0x804b50fffecd2202/set' with data '{"state":"ON"}'
debug 2022-05-15 13:59:30Publishing 'set' 'state' to '0x804b50fffecd2202'
debug 2022-05-15 13:59:31Received MQTT message on 'zigbee2mqtt/0x804b50fffecd2202/set' with data '{"state":"ON"}'
debug 2022-05-15 13:59:31Publishing 'set' 'state' to '0x804b50fffecd2202'
debug 2022-05-15 13:59:32Received MQTT message on 'zigbee2mqtt/0x804b50fffecd2202/set' with data '{"state":"OFF"}'
debug 2022-05-15 13:59:32Publishing 'set' 'state' to '0x804b50fffecd2202'
debug 2022-05-15 13:59:34Received MQTT message on 'zigbee2mqtt/0x804b50fffecd2202/set' with data '{"brightness":29}'
debug 2022-05-15 13:59:34Publishing 'set' 'brightness' to '0x804b50fffecd2202'
error 2022-05-15 13:59:53Publish 'set' 'state' to '0x804b50fffecd2202' failed: 'Error: Command 0x804b50fffecd2202/242 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32270 - 242 - 215 - 6 - 11 after 10000ms)'
debug 2022-05-15 13:59:53Error: Command 0x804b50fffecd2202/242 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32270 - 242 - 215 - 6 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
info 2022-05-15 13:59:53MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x804b50fffecd2202' failed: 'Error: Command 0x804b50fffecd2202/242 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 32270 - 242 - 215 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x804b50fffecd2202"},"type":"zigbee_publish_error"}'
error 2022-05-15 14:00:16Publish 'set' 'state' to '0x804b50fffecd2202' failed: 'Error: Command 0x804b50fffecd2202/242 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32270 - 242 - 216 - 6 - 11 after 10000ms)'
debug 2022-05-15 14:00:16Error: Command 0x804b50fffecd2202/242 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32270 - 242 - 216 - 6 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
info 2022-05-15 14:00:16MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x804b50fffecd2202' failed: 'Error: Command 0x804b50fffecd2202/242 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 32270 - 242 - 216 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x804b50fffecd2202"},"type":"zigbee_publish_error"}'

closed 2022-06-22 by github-actions[bot]

taugusti, 2022-09-07:
Did you solve this issue? I have the same on two different Schneider / LK Wiser dimmers. 
Detected as CCT5010-0001.
The funny thing is that when I touch the "test" button on the dimmer (that toggles the light), the state is immediately reported in the zigbee2mqtt "exposes" tab.

gbjmkm, 2022-09-22:
Hi
It seems to be running. I have no idea on how and why.
When I add the unit I get a lot of failed communication but after leaving it for some time it starts to operate as intended.
It could be related to restart of hass.