Koenkk/zigbee2mqtt #12489: [New device support]: ROB_200-029-0

maxi1406, 2022-05-15:
### Link

https://www.robbshop.nl/robb-smarrt-zigbee-rolluikschakelaar

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0x84ba20fffed30dbe","nwkAddr":755,"manufId":4107,"manufName":"ROBB smarrt","powerSource":"Mains (single phase)","modelId":"ROB_200-029-0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,258,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0,"zclVersion":3}},"genPollCtrl":{"attributes":{}},"genAnalogOutput":{"attributes":{}},"genOnOffSwitchCfg":{"attributes":{}},"genLevelCtrl":{"attributes":{"currentLevel":127}},"haDiagnostic":{"attributes":{"numberOfResets":93}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r47","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1652614209646,"defaultSendRequestWhen":"immediate"}

### Comments

I Added the device but it is not supported.
The ROB_200-029-0 is the new version of ROB_200-010-0 (this device is supported)

Koenkk, 2022-05-16:
Please check if it works with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ROB_200-029-0"'],
    model: 'ROB_200-029-0',
    vendor: 'ROBB',
    description: 'Zigbee curtain motor controller',
    meta: {coverInverted: true},
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position()],
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

maxi1406, 2022-05-16:
It almost works, I needed to change one thing in your code.
I think it's a typo.

The code say's:
`zigbeeModel: ['ROB_200-029-0"'],`

But it needs to be"
`zigbeeModel: ['ROB_200-029-0'],`

There is a extra " in the name, after this change is works

Koenkk, 2022-05-17:
Great, added it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-17 by Koenkk