Koenkk/zigbee2mqtt #12490: sonoff ZBmini firmware 2.0.0 work fine but last Seen problem

steve41200, 2022-05-15:
### What happened?

last seen don't actualise alone (only when trigger device virtually)

### What did you expect to happen?

last seen works like other rooter on my network

### How to reproduce it (minimal and precise)

just plug the ZBMINI firmware 2.0.0 and wait they never update last seen

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20211217

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

### Debug log

I don't know what put it.
Nu bug particularly. I wait Lon time and see never update the last update

I get resume the problem
[image]

I have 6 ZBMINI sonoff 3 on firmware 2.0.0 ( date: 20210320)who have the same problem.

They works fine when I trigger them into HomeAssistant or Z2M. It update the last seen parameter, but last seen never actualise alone like other rooter.
I have 3 other ZBMINI with no number firmware information but date firmware older (20200927). The last seen parameter update alone like other rooter.
Is their is a trick to change firmware ?
Without that problem they works fine.
I use the last seen for tell me each day if a device is offline. So they tell me this 3 are offline if they don't were trigger one day ...

I have try to plug on just near the coordinator but I have the same effect. No update last seen if no trigger

I have the latest version of add on z2m Current version: 1.25.1-1
the coordinator is Texas Instruments LAUNCHXL-CC1352P-2

steve41200, 2023-06-08:
Hi. The only solution i ve found is desactive thé last seen sensor on them .

RezzZ, 2023-06-09:
i removed the device from z2m, pushed some wires more in the box and re-adjusted the zbmini in the ceiling box and the devices seems to be stable online now for half a day... Theory, perhaps the internal on/off switch (by sqeezing the zbmini) or the reset button was constantly pushed by pressure from outside, wires inside the box and the light fixture on the other side... time will tell

distante, 2023-11-04:
> i removed the device from z2m, pushed some wires more in the box and re-adjusted the zbmini in the ceiling box and the devices seems to be stable online now for half a day... Theory, perhaps the [...]

Did you confirmed this? I have an 20210320 that is making me problems.

RezzZ, 2023-11-05:
> Did you confirmed this? I have an 20210320 that is making me problems.

for me that was the solution plus what i wrote above about changing the timeout in z2m