Koenkk/zigbee2mqtt #12491: [New device support]: ZTB23 Zemismart switcher

santanar00, 2022-05-15:
### Link

https://www.zemismart.com/products/tb21-zb-1

### Database entry

{"id":14,"type":"EndDevice","ieeeAddr":"0x003c84fffed6dcd3","nwkAddr":53826,"manufId":4098,"manufName":"_TZ3000_wu0shw0i","powerSource":"Mains (single phase)","modelId":"TS0013","epList":[1,3,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cd4a029","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cd4a029","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cd4a029","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1652624050718,"defaultSendRequestWhen":"immediate"}

### Comments

Hello team, how are you?

Please can you help to adding the Zemismart Switcher ZTB223 to compatible devices in ZigbeeToMQTT? 

[image]

The link provided above is for the same product, but with a similar model.

Thank you very much!

closed 2022-06-22 by github-actions[bot]

arnans, 2023-10-19:
I also want to use this switch with Zigbee2MQTT. Can we add support for it?  @Koenkk

Koenkk, 2023-10-20:
should be detected out of the box as a https://www.zigbee2mqtt.io/devices/TS0013.html

jsobell, 2024-09-25:
It is recognised, but does not respond to individual states for the multiple gangs.
Sending light-on or off for any individual light causes all outputs to change.