Koenkk/zigbee2mqtt #12493: [New device support]: Curtain swithch Bseed TS0601

rbenhass, 2022-05-15:
### Link

https://pl.aliexpress.com/item/1005003391392059.html?spm=a2g0o.store_pc_allProduct.8148356.4.66be1d70Ch8BCx&pdp_npi=2%40dis%21EUR%21%E2%82%AC%2044%2C96%21%E2%82%AC%2022%2C48%21%21%21%21%21%402100bdde16526258930288492e5211%21%21sh

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0x003c84fffe7ff224","nwkAddr":45116,"manufId":4098,"manufName":"_TZE200_yenbr4om","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0007\u000f\r\u0014*\u0012\u0010\r\u0014*\u0012","65506":31,"65508":0,"stackVersion":0,"dateCode":"","appVersion":68}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d3a74","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d3a74","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1560235388},"lastSeen":1652642063722,"defaultSendRequestWhen":"immediate"}

### Comments

I re-open a new request for supporting this device as the other bug are closed. I'm using z2m addon v1.25.1-1 with HA
For now this device is partial supported So _TZE200_yenbr4om can be added to the list of manufacturer I'm using attached converter from: https://github.com/Koenkk/zigbee-herdsman-converters/blob/6640de0174b16a76e1746d323e3c3522cd7e8aeb/devices/tuya.js#L784. 
- I am able only to open/close/stop the cover.
- **Calibration** is not supported  with logs: Publish 'set' 'calibration' to 'curtain01' failed: 'Error: Write 0x540f57fffeecb96a/1 closuresWindowCovering({"tuyaCalibration":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2022-05-15 21:33:19Error: Write 0x540f57fffeecb96a/1 closuresWindowCovering({"tuyaCalibration":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:6325:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)

- **Backlight** mode is not working: Publish 'set' 'backlight_mode' to 'curtain01' failed: 'Error: Write 0x540f57fffeecb96a/1 genOnOff({"tuyaBacklightMode":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2022-05-15 21:41:34Error: Write 0x540f57fffeecb96a/1 genOnOff({"tuyaBacklightMode":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:6364:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
info 2022-05-15 21:41:34MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'backlight_mode' to 'curtain01' failed: 'Error: Write 0x540f57fffeecb96a/1 genOnOff({\"tuyaBacklightMode\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"curtain01"},"type":"zigbee_publish_error"}'

- **motor_reversal** not supported: Publish 'set' 'motor_reversal' to 'curtain01' failed: 'Error: Write 0x540f57fffeecb96a/1 closuresWindowCovering({"tuyaMotorReversal":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2022-05-15 21:47:01Error: Write 0x540f57fffeecb96a/1 closuresWindowCovering({"tuyaMotorReversal":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:6339:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_yenbr4om'
        },
    ],
    model: 'TS0601_cover',
    vendor: 'BSEED',
    description: 'Zigbee curtain switch',
    fromZigbee: [fz.tuya_cover, fz.tuya_backlight_mode, fz.tuya_cover_options],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options,tz.tuya_cover_calibration, tz.tuya_cover_reversal, tz.tuya_backlight_mode],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')
                .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0)
                    .withValueMax(255)
                    .withDescription('Motor speed')),
            exposes.binary('calibration', ea.ALL, 'ON', 'OFF'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF'),
            exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),
            exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time')],
};

module.exports = definition;
```

rbenhass, 2022-05-20:
Anything I can do to make progress?

rbenhass, 2022-06-06:
how can I make this progress any additional information is needed?

Maggi0r, 2022-06-06:
Wo hast du das eingetragen ich habe dass in die Tuya.js am Anfang nach den ganze Libs eingefügt aber er wird trotzdem nicht erkannt.

rbenhass, 2022-06-10:
@Koenkk BSEED technical department replied that the UART protocol is not currently support and that the
ZigBee shade switches also do not support turning off the backlight.
So finally I can only open/close/stop and get current position with attached converter. Can we add this devise/manufacturer as it's converter already exist?

Koenkk, 2022-06-11:
@rbenhass can you update your external converter such that it only contains the features it supports? (especially the `exposes` section of it)

rbenhass, 2022-06-11:
@Koenkk updated

Koenkk, 2022-06-12:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-12 by Koenkk

Anatoly1976, 2022-06-20:
Set position still  doesn't work.

rbenhass, 2022-06-20:
@Anatoly1976 I never said that setting position is working. With this converter we can only **get** the postion

rbenhass, 2022-06-20:
Hello @itsolon,
I agree with you that this curtain switch is a carp the backlight is awful and it cannot be set off it's very annoying to have this light at night. For me it's the only solution I found to automatize tow switches one stick to the other. And now with Z2M addon on HA I can create my automation and do what I want to do.
About Calibration it can be done manually pressing pause button then when it start blinking press up/down check this video: https://www.youtube.com/watch?v=9nEowDoKwaw

itsolon, 2022-06-21:
> 

Thank you very much for calibration hint i will try that tomorrow if time