Koenkk/zigbee2mqtt #12500: Aqara roller shade companion E1

GamerClassN7, 2022-05-16:
### What happened?

Device dnt report state back, but everithing seems ok, even controls work just state report dont 

### What did you expect to happen?

After i set  curtain value state should be reported back not staty on 0 or 100 depending on open close 

### How to reproduce it (minimal and precise)

click one of this buttons and see that slider dont move
[image]

### Zigbee2MQTT version

 1.25.1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (zStack3x0)

Drafteed, 2022-05-16:
In order for this device to work (fully), at least the following firmware is required on your adapter:

CC2530/CC2531: 20211115
CC1352/CC2652: 20211114
Conbee II: [0x26720700](http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26720700.bin.GCF)

GamerClassN7, 2022-05-17:
> In order for this device to work (fully), at least the following firmware is required on your adapter:

Hello, is there waz how to update my Sonoff zigbee stick  ower on board usb ?

GamerClassN7, 2022-05-17:
also after updating firmware do i need to rejoin all devices ?

GamerClassN7, 2022-05-20:
> > i update stick to version
> > [image]
> > however behavioral is still same
> 
> You will need to repair it after upgrading your adapter firmware.
> 
> https://www.zigbee2mqtt.io/devices/ZNJLBL01LM.html

Still not working as it should but it is much bether  :)

Otnow, 2022-06-03:
@GamerClassN7, I think [this](https://github.com/Koenkk/zigbee2mqtt/issues/12647) is the same problem as you.

If so, please write the [following debug log](https://github.com/Koenkk/zigbee2mqtt/issues/12647#issuecomment-1143623963).

closed 2022-07-12 by github-actions[bot]