Koenkk/zigbee2mqtt #12507: [New device support]: Moes/Tuya Smart Button ERS-10TZBVB-AA

gorbunovav, 2022-05-16:
### Link

https://aliexpress.ru/item/1005004204208927.html

### Database entry

{"id":25,"type":"EndDevice","ieeeAddr":"0x385b44fffe3c1bf1","nwkAddr":6233,"manufId":4098,"manufName":"_TZ3000_kjfzuycl","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":1,"65534":0,"modelId":"TS004F","manufacturerName":"_TZ3000_kjfzuycl","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"manuSpecificTuya_3":{"attributes":{"53265":"�\u0000\u0003�\u0003�\u0000Z\u0003�\u0003�\u0000\u0003�\u0000d\u0000�\u0003�\u0003�\u0000\u0001�\u0003 \u0001\u000e\u0003�\u0003�\u0001\u0001�\u0000\u0000\u0000�\u0000�\u0000�"}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f8256","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f8256","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":927531739},"lastSeen":1652734204419,"defaultSendRequestWhen":"immediate"}

### Comments

The device is reported as unsupported.

I've modified the converter for the Tuya Wireless Switch, which has the same model number and similar features per button: https://www.zigbee2mqtt.io/devices/TS004F.html (I've just removed extra buttons).

The default mode for this device is the `dimmer mode`. This converter switches device into the `scene` mode, in which it just reports button actions: 

 - single
 - double
 - hold (hold the button for about 5 seconds)

It is still possible to switch the mode with quick triple button press.
The provided converter version doesn't support the dimmer mode.

In the dimmer mode:

- first single press turns the light on
- consequential single presses are switching brightness / color temperature presets.
- double press turns off the light

Product page, found by the serial number:
https://expo.tuya.com/product/1050040

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

module.exports = [
    {
        fingerprint: [{modelID: 'TS004F', manufacturerName: '_TZ3000_kjfzuycl'}],
        model: 'ERS-10TZBVB-AA',
        vendor: 'Moes',
        description: 'Smart Button',
        exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold'])],
        fromZigbee: [fz.battery, fz.tuya_on_off_action],
        toZigbee: [tz.tuya_operation_mode],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genBasic', [0x0004, 0x000, 0x0001, 0x0005, 0x0007, 0xfffe]);
            await endpoint.write('genOnOff', {'tuyaOperationMode': 1});
            await endpoint.read('genOnOff', ['tuyaOperationMode']);
            try {
                await endpoint.read(0xE001, [0xD011]);
            } catch (err) {/* do nothing */}
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        },
    },
];
```

Koenkk, 2022-05-26:
Added by @kirovilya  in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4285

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) (and the next release on 1 June)

closed 2022-05-26 by Koenkk

MiAutomations, 2022-06-21:
> @MiAutomations yes, it is available

Thank you for your reply !

I tried use the device, now it is recognized when I pair it. But I can't view the way to use in the automations once there are nothing in the actions only the battery but is not reporting the value and the single action is not working also

[image]

[image]

Can you please let me know if you need to so something in oder to get this to work ? 

I already deleted the device and try to pair again with sucess 

Thank you and best regards

gorbunovav, 2022-06-21:
@MiAutomations the `operation_mode` should be `event`

MiAutomations, 2022-06-21:
Great thank you I will check