Koenkk/zigbee2mqtt #12519: LifeControl MCLH-08 no battery charge readings

VikeDragon, 2022-05-17:
### What happened?

LifeControl MCLH-08 no battery charge readings.
The sensor does not transfer the charge of two AA batteries to Zigbee2Mqtt.
In Exposes only the parameters of temperature, humidity, VOCs, eco2, communication quality - there is no charge parameter.
I ask you to add charge readings for the sensor. Please help me solve the problem.

### What did you expect to happen?

Get the percentage of battery charge, as for the end device

### How to reproduce it (minimal and precise)

Connect the sensor to Zigbee2mqtt and check the battery charge data

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (zStack3x0)

### Debug log

debug 2022-05-17 20:34:26Received Zigbee message from '0x00158d0000dab5d7', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"maxMeasuredValue":1124,"measuredValue":2607,"minMeasuredValue":4995,"tolerance":110}' from endpoint 1 with groupID 0
info 2022-05-17 20:34:26MQTT publish: topic 'zigbee2mqtt/0x00158d0000dab5d7', payload '{"eco2":1124,"humidity":49.95,"last_seen":"2022-05-17T20:34:28+05:00","linkquality":142,"temperature":26.07,"voc":110}'

VikeDragon, 2022-05-18:
Greetings Koenkk!
Thank you for taking the time to solve my problem. Here is the database part.db for this device, where it can be seen that the charge value is formed as a percentage :

{"id":53,"type":"EndDevice","ieeeAddr":"0x00158d0000dab5d7","nwkAddr":17129,"manufId":4151,"manufName":"Nexturn","powerSource":"Battery","modelId":"VOC_Sensor","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,513,1026,1029],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"VOC_Sensor","manufacturerName":"Nexturn","powerSource":3,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"2016-06-01","swBuildId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2654,"minMeasuredValue":5379,"maxMeasuredValue":772,"tolerance":56}},"genPowerCfg":{"attributes":{"**batteryPercentageRemaining":95**}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"2016-06-01","swBuildId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1652898764182,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-05-20:
Should be supported now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

VikeDragon, 2022-05-20:
> Should be supported now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Downloaded zigbee-herdsman-converters lifecontrol.js and configured. Zigbee2mqtt I see a new battery parameter has been added, repairing device, but I haven't received any values yet.
Disabled zigbee2mqtt addon and enabled the edge version, but it gives an error "s6-overlay-suexec: fatal: can only run as pid 1"
What am I doing wrong? Can it wait longer with the converter on the stable version until the battery percentage appears?

Koenkk, 2022-05-21:
> Can it wait longer with the converter on the stable version until the battery percentage appears?

Seems to be the best thing to do; next release will be on 1 June

VikeDragon, 2022-05-21:
> > Can it wait longer with the converter on the stable version until the battery percentage appears?
> 
> Seems to be the best thing to do; next release will be on 1 June

Thank you for your quick support! We are waiting for the release ...
Thank you for everything you do!

VikeDragon, 2022-05-22:
Thank you, a day later the charge began to be displayed, apparently depends on the sensor!

VikeDragon, 2022-05-22:
Please add for MCLH-08 and MCLH-07
		meta: {battery: {dontDividePercentage: **true**}},

Koenkk, 2022-05-22:
Done!

closed 2022-05-22 by Koenkk

VikeDragon, 2022-05-23:
Thank you! Super!