Koenkk/zigbee2mqtt #12523: RTCGQ14LM - Illuminance not showing in Home Assistant

m0dded, 2022-05-17:
### What happened?

Aqara P1 Motion sensor [RTCGQ14LM] device support was added in 1.25.1 - however it is not reporting illuminance in HA.
Illuminance is shown in Z2M in dashboard.

I reported this bug to HA - see here: https://github.com/home-assistant/core/issues/71848 - however they have closed it asking to report here.

### What did you expect to happen?

Illuminance to show in HA

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Otnow, 2022-05-18:
Support for `illuminance_lux` has already been [added](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4206) and is currently available in the [dev version of z2m](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), but will be available in the stable version early next month.

If you have such an opportunity, then I ask you to test the operation of the `trigger indicator` option using an [external converter](https://github.com/Koenkk/zigbee2mqtt/issues/11990#issuecomment-1126577607) (_it supports illuminance_lux_) and write a log for improve support this sensor.

m0dded, 2022-05-18:
Thank you - if i add this external converter - will this break all of the P1 Sensors that I have currently paired using the 1.25.1 integration?

Otnow, 2022-05-18:
The proposed external converter will change the functionality of all sensors of this model, but should not break, but improve, therefore testing is required.

In any case, if something goes wrong, you can disable the external converter and everything will be as it was before it.

closed 2022-06-25 by github-actions[bot]