Koenkk/zigbee2mqtt #12526: Ubisys Firmware not found

Danit2, 2022-05-18:
### What happened?

I can not Update the Firmware for my Ubisys devices.

Because i need to update first to the last 1.9.x version befor i can update to the 2.x version.

Two days agoZ2M have show me the last 1.9.x version and now i become an error.
`Failed to check if update available for 'Badezimmer_Spot' (No image available for imageType '0x7b21' with hardware version 5)`

### What did you expect to happen?

I can first update to the last 1.9.x version and after to the 2.x version.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20211217

### Adapter

Sonoff USB

### Debug log

debug 2022-05-18 11:45:03: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Badezimmer_Spot","transaction":"7xqwn-3"}'
info  2022-05-18 11:45:03: Checking if update available for 'Badezimmer_Spot'
info  2022-05-18 11:45:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Badezimmer_Spot'","meta":{"device":"Badezimmer_Spot","status":"checking_if_available"},"type":"ota_update"}'
debug 2022-05-18 11:45:03: Check if update available for '0x001fee0000007ff3' (D1 (5503))
debug 2022-05-18 11:45:03: Using endpoint '232'
debug 2022-05-18 11:45:07: Received Zigbee message from 'Badezimmer_Spot', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":26411534,"imageType":31521,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2022-05-18 11:45:07: MQTT publish: topic 'zigbee2mqtt/Badezimmer_Spot', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":10,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":138,"capabilities_configurable_curve":true,"capabilities_forward_phase_control":true,"capabilities_overload_detection":true,"capabilities_reactance_discriminator":true,"capabilities_reverse_phase_control":true,"last_seen":"2022-05-18T11:45:07+02:00","linkquality":72,"minimum_on_level":20,"mode_phase_control":"automatic","power":0,"state":"OFF","status_capacitive_load":true,"status_forward_phase_control":false,"status_inductive_load":true,"status_overload":false,"status_reverse_phase_control":true,"update":{"state":"idle"},"update_available":false}'
debug 2022-05-18 11:45:07: Got OTA request '{"fieldControl":1,"manufacturerCode":4338,"imageType":31521,"fileVersion":26411534}'
debug 2022-05-18 11:45:08: OTA ubisys: got firmware page, status: 200, data.length: 80862
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B39-0000-0004-02000300-spo-fmi4.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B31-0000-0006-02010300-spo-fmd.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B38-0000-0004-02010300-spo-rmd.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B34-0000-0007-02000300-spo-fmsh.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B37-0000-0004-02000300-spo-rmsh.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B3A-0000-0005-02020300-m7b-r0.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B32-0000-0001-02000300-spo-fms-rev0-1.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B32-0002-0007-02000300-spo-fms.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B35-0000-0004-02000300-spo-rms.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B33-0000-0006-02000300-spo-fms2.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B36-0000-0004-02000300-spo-rms2.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B3B-0000-0001-02000300-m7b-h10.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B2C-0000-0001-01010300-ld6.ota.zigbee
debug 2022-05-18 11:45:10: OTA ubisys: image found: /wp-content/uploads/10F2-7B0D-0000-0001-011C0300-m7b-h1.ota.zigbee
info  2022-05-18 11:45:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Badezimmer_Spot' (No image available for imageType '0x7b21' with hardware version 5)","meta":{"device":"Badezimmer_Spot","status":"check_failed"},"type":"ota_update"}'
info  2022-05-18 11:45:10: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Badezimmer_Spot"},"error":"Failed to check if update available for 'Badezimmer_Spot' (No image available for imageType '0x7b21' with hardware version 5)","status":"error","transaction":"7xqwn-3"}'
error 2022-05-18 11:45:10: Failed to check if update available for 'Badezimmer_Spot' (No image available for imageType '0x7b21' with hardware version 5)
debug 2022-05-18 11:45:10: AssertionError [ERR_ASSERTION]: No image available for imageType '0x7b21' with hardware version 5
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/ubisys.js:32:5)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:194:39)

sjorge, 2022-05-18:
Hmmm I managed to get mine to 2.0.0 ... but I do think I updated to 1.9.3 and then the day after there was another one to 2.0.0.

Looks like the older firmwares are still listed on the page, but I guess the html parser is not picking them up.

Danit2, 2022-05-18:
Yes i have missed to update to the last 1.9.x version and now there on a other website. 

Can someone update the script to the new pages? 

Can someone explain to me how i can do the update manual with the local ota file? 
I dont realy know how to do the index file.

Danit2, 2022-05-18:
i have try it with this:
```
[
    {
        "url": "https://www.ubisys.de/unterstuetzung/support-firmware/archiv-der-firmware-d1/wp-content/uploads/10F2-7B21-0000-0006-0194020E-spo-fmd.ota.zigbee"
    },
    {
        "url": "https://www.ubisys.de/wp-content/uploads/10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee"
    }
]
```

Save it as `firmware.json` and add this line in my config file:
```
ota:
  zigbee_ota_override_index_location: firmware.json
```

but it does not work.

Danit2, 2022-05-18:
i have also try with this:
```
[
    {
        "url": "10F2-7B21-0000-0006-0194020E-spo-fmd.ota.zigbee",
        "force": true
    },
    {
        "url": "10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee",
        "force": true
    }
]
```
in my `firmware.json` file and i have put the firmware files to the same directory as the config file is. 

But it seems that Z2M do not load the new firmware index.

Danit2, 2022-05-21:
I have found a solution for my problem.
I could now update to the new 2.x version.
But it would be good if someone could adapt Ubisys' OTA script to work with the new structure of the homepage.

sjorge, 2022-05-21:
It’s fine to leave this open until someone who actually understands the weird page parser can have a look at it.

Danit2, 2022-06-21:
This is not stale. 
This is still a problem.

svenso, 2022-06-23:
I actually found the official download page for ubisys fw updates used by official ubisys bridge: 
http://fwu.ubisys.de/smarthome/OTA/release/index 

DL of the files for example: 
http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B01-0000-0006-01110206-spo-fmd.ota.zigbee

parsing here is simpler..
Unfortunately https is not working..

closed 2022-07-23 by Koenkk

zivillian, 2022-07-27:
sadly it's not fixed fo all devices and firmware version - I've got a new C4 with firmware 20210305-DE-FB0 which can not be upgraded with the current implementation. The problem is, that the search is aborted after the first applicable firmware (which in my case is the currently installed one).