Koenkk/zigbee2mqtt #12529: OpenHABian install

misko903, 2022-05-18:
### What happened?

I can't see the zigbee2mqtt in the installation menu of openhabian-config, as described here:

https://www.zigbee2mqtt.io/guide/installation/04_openhabian.html

mosquitto is installed and working long time already.

### What did you expect to happen?

to have "optional components" and select "Zigbee2MQTT" in the menu.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211110

### Adapter

cc2531

Koenkk, 2022-05-18:
@Larsen-Locke could you help with this?

Larsen-Locke, 2022-05-18:
Is your raspberry connected to the internet? If so openhabian-config updates itself to the latest version after being started.
Which options do you see under "optional components"?
E.g. the last option should be something with EVCC. If that is missing too, then openhabian-config is not up-to-date.

gualcosta, 2022-05-19:
I've got the same issue, I am running a RPI 4 with OpenHab 3 (upgraded from 2) and even after updating the system to the latest version did not show the option to install Zigbee2Mqttt.

PatrikStaehli, 2022-05-20:
I've just found out: you need to select openhabian-config branch 'main' instead of 'openHAB3' (which seems to be the default) in the first menu (01 | Select Branch).

Regards
Patrik

Larsen-Locke, 2022-05-20:
yes, that's it. You have to choose main in the branch menu. I have to add that in the doc.

Larsen-Locke, 2022-05-20:
Added in the docs.
PR is [https://github.com/Koenkk/zigbee2mqtt.io/pull/1391](https://github.com/Koenkk/zigbee2mqtt.io/pull/1391)

closed 2022-06-28 by github-actions[bot]

misko903, 2022-10-11:
> PR is Koenkk/zigbee2mqtt.io#1391

Confirming it works. However, I have installed the Linux version, now I
need to figure out how to install OpenHABian ver...

Updating Zigbee2MQTT... FAILED (stop systemctl)

                            │ There was an error or interruption during the
execution of:                  │
                            │   "20 | Optional Components"
                                │
                            │
                               │
                            │ Please try again. If the error persists,
please read                         │
                            │ /opt/openhabian/docs/openhabian-DEBUG.md or
                               │
                            │
https://github.com/openhab/openhabian/blob/main/docs/openhabian-DEBUG.md
how │
                            │ to proceed.

Thanks for pointing out how to have it on the menu anyway! that was it...

miso

misko903, 2022-10-11:
I have started with v3.0, now on the latest stable 3.2 - updated to latest
patches:
*openHABian Configuration Tool —
[openHAB3]{2022-03-27T13:22:11-06:00}(1305ac7)*
Latest item is Remove HABApp - scrolling down makes no difference.
[image: image.png]

Thanks,
miso

Larsen-Locke, 2022-10-13:
Well, if there is no option zigbee2mqtt your openhabian is not up-to-date.
The tool offers automated update when connected to the internet.
The current version is  [main]{2022-09-25T17:39:16+02:00}(3935166) .
To get the latest updated you have to switch to main branch
The problem with the update of zigbee2mqtt should be resolved with the next update of openhabian. The PR is just being reviewed now. I'll keep you infomed as soon as it's merged.
For future issues: Please raise in openhabian not zigbee2mqtt, cause there is nothing that Koen can do about that.

Larsen-Locke, 2022-10-15:
PR was merged. Issue can be closed