Koenkk/zigbee2mqtt #12531: [New device support]:

bdkacz, 2022-05-18:
### Link

https://zigbee.blakadder.com/Lidl_14156408L.html

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0x2c1165fffe6111f1","nwkAddr":60187,"manufId":4098,"manufName":"_TZ3210_c2iwpxf1","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502B","manufacturerName":"_TZ3210_c2iwpxf1","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1652903252747,"defaultSendRequestWhen":"immediate"}

### Comments

Hi. 

Please add support for linked device. 

I try to write converter, but after add attached external converter z2m fails to start

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
	// Since a lot of Tuya devices use the same modelID, but use different data points
	// it's usually necessary to provide a fingerprint instead of a zigbeeModel
	fingerprint: [{
		// The model ID from: Device with modelID 'TS0601' is not supported
		// You may need to add \u0000 at the end of the name in some cases
		modelID: 'TS0502B',
		// The manufacturer name from: Device with modelID 'TS0601' is not supported.
		manufacturerName: '_TZ3210_c2iwpxf1'
	}, ],
	model: '14156408L',
	vendor: 'Lidl',
	description: 'Light',
	fromZigbee: [
		fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
		fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
	],
	toZigbee: [
		tz.tuya_data_point_test, // Another debug converter
	],
	
};

module.exports = definition;
```

### Supported color modes

color temperature

bdkacz, 2022-05-18:
Here is working converter

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [{modelID: 'TS0502B', manufacturerName: '_TZ3210_c2iwpxf1'}],
model: '14156408L',
description: 'Livarno Home Ceiling LED light',
vendor: 'Lidl',
extend: extend.light_onoff_brightness(),
};

module.exports = definition;

bdkacz, 2022-05-18:
Here is working converter :

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [{modelID: 'TS0502B', manufacturerName: '_TZ3210_c2iwpxf1'}],
model: '14156408L',
description: 'Livarno Home Ceiling LED light',
vendor: 'Lidl',
extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;

Koenkk, 2022-05-20:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-20 by Koenkk