Koenkk/zigbee2mqtt #12542: NAS-AB02B2: Intermittent ignored commands

IAmTheMilkManMyMilkIsDelicious, 2022-05-20:
### What happened?

Due to the way [NAS-AB02B2](https://www.zigbee2mqtt.io/devices/NAS-AB02B2.html#neo-nas-ab02b2) is setup in z2m (or firmware) only one command can be sent at a time. When using the siren for multiple purposes (door bell, house alarm, notification beeps) you have to send each command (duration, melody, volume, alarm) one after another.

For example a door bell script:

1. payload: '{ "duration": 4 }'
    topic: zigbee2mqtt/Siren/set
2. payload: '{ "melody": 18 }'
    topic: zigbee2mqtt/Siren/set
3. payload: '{ "volume": "medium" }'
    topic: zigbee2mqtt/Siren/set
4. payload: '{ "alarm": true }'
    topic: zigbee2mqtt/Siren/set

The issue the device has is it will randomly ignore commands. There seems to be no specific command it will or won't ignore. Sometimes I will run this script and it won't set the duration, causing the alarm bell to play forever. Other times it doesn't set the volume and it's too quiet. Many times it just won't play a sound at all.

I look through the 

### What did you expect to happen?

When running the script I mentioned, it should always play the correct melody for the correct duration

### How to reproduce it (minimal and precise)

I used homeassistant so my script looks like this:

```
alias: Set Door Bell
sequence:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/Siren/set
      payload: '{ "duration": 4 }'
  - service: mqtt.publish
    data:
      payload: '{ "melody": 18 }'
      topic: zigbee2mqtt/Siren/set
  - service: mqtt.publish
    data:
      payload: '{ "volume": "medium" }'
      topic: zigbee2mqtt/Siren/set
  - service: mqtt.publish
    data:
      payload: '{ "alarm": true }'
      topic: zigbee2mqtt/Siren/set
mode: single
icon: mdi:bell
```

But I would also have another script such as:
```
alias: Set Siren House Alarm Mode
sequence:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/Siren/set
      payload: '{ "duration": 900 }'
  - service: mqtt.publish
    data:
      payload: '{ "melody": 7 }'
      topic: zigbee2mqtt/Siren/set
  - service: mqtt.publish
    data:
      payload: '{ "volume": "high" }'
      topic: zigbee2mqtt/Siren/set
  - service: mqtt.publish
    data:
      payload: '{ "alarm": true }'
      topic: zigbee2mqtt/Siren/set

mode: single
icon: mdi:volume-vibrate
```

To reproduce, I simply run one script then the other a few times. I will always run into random issues where the device ignores certain commands.

### Zigbee2MQTT version

1.25.0 commit: 6f1460e4

### Adapter firmware version

20210120

### Adapter

zzh (CC2652R Stick)

### Debug log

The debug log doesn't show any warnings or errors and based on my scripts actually does successfully send every command to the device:

```
Publishing 'set' 'alarm' to 'Siren'
Received MQTT message on 'zigbee2mqtt/Siren/set' with data '{ "alarm": true }'
Publishing 'set' 'alarm' to 'Siren'
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":13}],"seq":52992}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":13}],"seq":53248}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":13}],"seq":53504}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[15],"type":"Buffer"},"datatype":4,"dp":21}],"seq":53760}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[15],"type":"Buffer"},"datatype":4,"dp":21}],"seq":54016}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":5}],"seq":54528}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":13}],"seq":54784}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Siren/set' with data '{ "melody": 16 }'
Publishing 'set' 'melody' to 'Siren'
Received MQTT message on 'zigbee2mqtt/Siren/set' with data '{ "volume": "high" }'
Publishing 'set' 'volume' to 'Siren'
Received MQTT message on 'zigbee2mqtt/Siren/set' with data '{ "alarm": true }'
Publishing 'set' 'alarm' to 'Siren'
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[16],"type":"Buffer"},"datatype":4,"dp":21}],"seq":55040}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":5}],"seq":55296}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":13}],"seq":55552}' from endpoint 1 with groupID 0
Received Zigbee message from 'Siren', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":13}],"seq":55808}' from endpoint 1 with groupID 0
```

IAmTheMilkManMyMilkIsDelicious, 2022-06-20:
Still an issue on current version 1.25.2

closed 2022-07-29 by github-actions[bot]

alb1981, 2022-09-30:
The problem exists. Zigbee2MQTT 1.27.0-1.

alb1981, 2022-09-30:
Zigbee2MQTT Current version: 1.27.2-1
The problem still exists.

IAmTheMilkManMyMilkIsDelicious, 2022-10-08:
Yes I still have this issue too, I think this device is either unpopular or this issue only affects a small amount of users. I think I will have to replace it with the [HEIMAN HS2WD-E](https://www.zigbee2mqtt.io/devices/HS2WD-E.html), hopefully that device may be more reliable.

jerrm, 2022-10-09:
I have several of the units.  I experienced the issue.  I now have a script that has been reliable, it adds a 40ms delay between commands, and issues each command twice, ie:  melody, wait, melody, duration, wait, duration, wait, volume, wait, volume, wait, alarm, wait, alarm.

I use standard HA service calls (select_option, set_number, turn_on) vs direct mqtt settings, but doubt that makes a difference.

Obviuosly this is a woraround, not 100% solution, but I haven't had problems since.  YMMV.

jhron, 2022-10-28:
> I have several of the units. I experienced the issue. I now have a script that has been reliable, it adds a 40ms delay between commands, and issues each command twice, ie: melody, wait, melody, [...]

I have the same experience with this sensor. It is not reliable. I have to send only one item per payload and every message 2x with some delay in between. And last is always "alarm": true. Last message can be send only once (I am still testing it, so I can be wrong)
In addition I have to always send everything. Otherwise when I send just "alarm": true, it sometimes set volume and duration back to low and 30s, and sometimes it is just blinking but there is no sound at all from the device. 

Originally I had NAS-AB02B0 and it was working ok (until it felled to ground and power connector was broken).

Nafanya10, 2023-07-31:
The problem still exists.
The issue the device has is it will randomly ignore commands.
I send each command (duration, melody, volume, alarm) one after another, but the random failures still happen.
Please help fix this issue.

Model: Alarm (NAS-AB02B2)
Manufacture: Zigbee TZE204_t1blo2bj
Zigbee2MQTT Current version: 1.32.1
Home Assistant Core 2023.7.2

albarts, 2024-04-18:
> I have several of the units. I experienced the issue. I now have a script that has been reliable, it adds a 40ms delay between commands, and issues each command twice, ie: melody, wait, melody, duration, wait, duration, wait, volume, wait, volume, wait, alarm, wait, alarm.

Still same issue with this Siren. 
Adding those delays and sending twice each command seems to work ok until a proper fix.

dominique6591, 2024-08-22:
I had the same issue. Adding delays between each command seems to work now. Delay values are based on engineering feeling only. No rational. Commands are sent once.
[image]

lecourtb, 2024-09-23:
I have same issue. Going to try to add a sleep between each commands.

chriswtb, 2024-12-11:
same here :( I would use two devices for alarm and doorbell but even then, the melody/duration is always reset again.

chriswtb, 2024-12-11:
quite interesting. while looking for an alternative Alarm module, i came across the newer ZA03 which seems to be based on the AB02.... Having the same problems:
https://community.home-assistant.io/t/za03-sirene-in-automation-help-needed/790383/5
https://github.com/Koenkk/zigbee2mqtt/issues/23529

tjaege, 2025-09-13:
I was confused too when testing this device. After some trial and error I figured out the following:

duration must be sent as a single payload (solo).
Example: {"duration": 5}
The device will store this internally, but it will not show up in the state (it usually reports null).

melody and volume must also be sent as separate payloads.
If you try to send them together, one of the values will be ignored.

Once those are set, you can trigger the alarm with:
{"alarm": true}

So in short:
Send melody (separately).
Send volume (separately).
Send duration (separately, only when you need to change it).
Finally send alarm: true.

Hope this helps others avoid the same confusion.