Koenkk/zigbee2mqtt #12543: Hive 1613v Energy and Temperature not reported

realroywalker, 2022-05-20:
### What happened?

According to the documentation the hive switch should report the temperature and energy usage (https://www.zigbee2mqtt.io/devices/1613V.html#hive-1613v)

The device detects and joins fine but the energy and temperature status always remains as null when the device is drawing power, only the instant power value is shown.

[image]

Here is a screenshot of the device details once joined to the network (Tumble Dryer Switch in the logs):
[image]

### What did you expect to happen?

The energy and temperature values to be reported

### How to reproduce it (minimal and precise)

Join the device as normal and observe the values reported while a device is consuming power via the plug

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

 20210120

### Adapter

Egony cc2652p

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8738738/log.txt)

realroywalker, 2022-07-10:
Now running 1.26.0 and can confirm I still have the same issue - energy and temperature both report null all the time

deskyeti, 2022-07-10:
Problem still existed on 1.26.0-1 although I’ve gone back to 1.25 for now due to other issues.

Koenkk, 2022-07-11:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SLP2', 'SLP2b', 'SLP2c'],
    model: '1613V',
    vendor: 'Hive',
    description: 'Active plug',
    fromZigbee: [fz.on_off, fz.metering, fz.temperature],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(9);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering', 'msTemperatureMeasurement']);
        await reporting.onOff(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.instantaneousDemand(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.temperature(endpoint);
    },
    exposes: [e.switch(), e.power(), e.energy(), e.temperature()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Koenkk, 2022-07-11:
> `failed (Status 'UNSUPPORTED_ATTRIBUTE')`

Seems the device doesn't support temperature, are you sure it does? (if not we should remove it)

realroywalker, 2022-07-11:
I can't guarantee either way unfortunately as I don't have access to the device via any other means than z2m - I was simply going by the wiki pages on z2m that say it exposes temperature 😁 

But at least the energy value is working, so thank you for that! - I assume the change from the external converter file goes into the next z2m release, so after that it's no longer needed to have the extra file?

Koenkk, 2022-07-12:
Pushed the fix, it will be included in the next release. I've removed the temperature, I think this device doesn't support it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

deskyeti, 2022-10-11:
I will test this as well. I have never seen the devices report temperature.

similair, 2022-12-29:
Hi There,
I am on version 1.28.4 and have the issue above, any ideas how to fix it please.

similair, 2023-01-09:
See [https://github.com/Koenkk/zigbee2mqtt/issues/15808](url)