Koenkk/zigbee2mqtt #12544: What can be a problem for 'friendly_name is already in use' in HA

Masterz69, 2022-05-20:
### What happened?

Key is stuck on device and it left network by long pressing key.
Device disappears in Z2M device list & dashboard.
But still exists in HA device list...

Name for device was same in Z2M & HA - f.e. XXX

Then I deleted device in HA.
From now there is no device XXX or any related entity in HA, MQTT topics, Z2M.

And now can't rename new device as XXX.

Getting error 'friendly_name XXX is already in use'.

Restarted Z2M add-on, the HA Core. Same staff.
See NO device XXX in HA & Z2M, but can't assign this name.

Pls advice - where can be a cause. May be I can provide some info to troubleshoot.

### What did you expect to happen?

Ability to user name which previously used for device, which left network by any reason.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

[image]

```
debug 2022-05-20 15:23:34: Received MQTT message on 'zigbee2mqtt/bridge/request/device/rename' with data '{"from":"plg MultiX","homeassistant_rename":false,"to":"plg MultiV","transaction":"74ea6-1"}'
error 2022-05-20 15:23:34: Request 'zigbee2mqtt/bridge/request/device/rename' failed with error: 'friendly_name 'plg MultiV' is already in use'
debug 2022-05-20 15:23:34: Error: friendly_name 'plg MultiV' is already in use
    at Object.changeFriendlyName (/app/lib/util/settings.ts:667:15)
    at Bridge.renameEntity (/app/lib/extension/bridge.ts:453:18)
    at Bridge.deviceRename (/app/lib/extension/bridge.ts:222:21)
    at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:118:63)
    at EventEmitter.emit (events.js:412:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:122:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27)
    at WebSocket.emit (events.js:400:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1137:20)
```

Masterz69, 2022-05-20:
The issue was:
* after device left the network by holding pairing key on device,
* IEEE and device name still left in devices.yaml file...

Solution: delete device entry from devices.yaml file.

closed 2022-05-20 by Masterz69

Dekadinious, 2023-12-15:
I can add that this has moved to configuration.yaml