Koenkk/zigbee2mqtt #12548: [New device support]: Somgoms ZigBee Smart Touch Fan Switch

akorfmann, 2022-05-21:
### Link

https://www.aliexpress.com/item/4001242513879.html

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0x84fd27fffe5fff1a","nwkAddr":46556,"manufId":4098,"manufName":"_TZE200_fvldku9h","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"         $\u001b*\u0012         $\u001b*\u0012","65506":31,"65508":0,"modelId":"TS0601","ManufacturerName":"_TZE200_fvldku9h","stackVersion":0,"dateCode":"","appVersion":68}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,""stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1653107038460,"defaultSendRequestWhen":"immediate"}

### Comments

I was trying to add the device from the documentation pages but it's a bit out of my wheelbase and I'm kinda lost. The fan controller shows up as the generic Tuya "TS0601" device name and shows up as unsupported in my zigbee2mqtt dashboard. It is a simple fan controller; up/increase speed, down/decrease speed and an on/off toggle. any help would be greatly appreciated as I can't find many zigbee fan controllers out there and I quite like this one (it has a beeper as well as backlight touch controls). (I also have homeassistant installed on a separate device if that makes a difference)

closed 2022-06-28 by github-actions[bot]

Big-Szu, 2023-04-19:
Can this issue by re-opened? I have the same switch and I'd love to have it added.

Big-Szu, 2023-04-21:
@akorfmann FYI https://github.com/Koenkk/zigbee2mqtt/issues/17397

CIPH3RD, 2024-11-16:
bit late here, but I have the same switch, this works for all features of the fan switch:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_fvldku9h']),
    model: 'ZB-1SRW-US',
    vendor: 'Somgoms',
    description: 'Fan switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    whiteLabel: [{vendor: 'Somgoms', model: 'ZB-1SRW-US'},
        tuya.whitelabel('Somgoms', 'ZB-1SRW-US', 'Fan switch', ['_TZE200_fvldku9h']),
    ],
    exposes: [
        e.binary('backlight_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Backlight switch'),
        e.binary('switch_fan', ea.STATE_SET, 'ON', 'OFF').withDescription('Fan On Off'),
        e.numeric('fan_speed', ea.STATE_SET).withValueMin(0).withValueMax(100).withValueStep(1).withUnit('%').withDescription('Fan Speed'),
        e.enum('relay_status', ea.STATE_SET, ['off', 'on', 'memory']).withDescription('Recover state after power outage'),
        e.numeric('countdown_fan', ea.STATE_SET).withValueMin(0).withValueMax(43200).withValueStep(1).withUnit('s').withDescription('Max ON time in seconds'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'switch_fan', tuya.valueConverter.onOff],
            [2, 'countdown_fan', tuya.valueConverter.countdown],
            [4, 'fan_speed', tuya.valueConverter.raw],
            [11, 'relay_status', tuya.valueConverter.raw],
            [13, 'backlight_switch', tuya.valueConverter.onOff],
        ],
    },
};
module.exports = definition;`