Koenkk/zigbee2mqtt #12555: PIR TS0202_TZ3000_mcxw5ehu reports occupied every ~4h

chrisrock1984, 2022-05-21:
### What happened?

every ~4 hours this sensor reports occupied -> I have 3 of them and all are behaving the same. This can be seen in the history of home assistant.
This is also the same with the Edge version. I am running now the non edge version

### What did you expect to happen?

not reporting occupied when no one is in these rooms.

### How to reproduce it (minimal and precise)

just check the history of home assistant 
[image]

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

zhh

Koenkk, 2022-05-21:
could you provide the debug log of:
- occupancy triggered correctly 
- occupancy triggered incorrectly

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

chrisrock1984, 2022-05-22:
Ok,
I have logs now.
Sensors **0xa4c138fcae2f042e**  and **0xa4c13856556dd**
had a "ghost" detection at the very end of the logs. I did a real detection, by going into the room about 15 to 20 minutes later.

[_addon_45df7312_zigbee2mqtt_logs (7).txt](https://github.com/Koenkk/zigbee2mqtt/files/8749516/_addon_45df7312_zigbee2mqtt_logs.7.txt)

Koenkk, 2022-05-22:
I need to know the exact line numbers. I'm especially interested in the `Received Zigbee message from '0xa4c138fcae2f042e'` message of both the non triggered and triggered occupancy:true publish.

chrisrock1984, 2022-05-23:
non triggered detection -> starting line 109026
triggered detection -> starting line 119718
both for 0xa4c138fcae2f042e

I hope this is the correct one
thx for you help!!

Koenkk, 2022-05-23:
Found the message that are send by the device:

Non triggered
```
Zigbee2MQTT:debug 2022-05-22 15:01:50: Received Zigbee message from '0xa4c138fcae2f042e', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
```

Triggered
```
Zigbee2MQTT:debug 2022-05-22 15:21:33: Received Zigbee message from '0xa4c138fcae2f042e', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
```

As you can see, the value of `zonestatus` is identical between both messages meaning that we cannot distuingish them. Looks to me that the device is malfuctioning.

chrisrock1984, 2022-05-23:
Too bad
All 3 of them then are not working correctly ;-/

Thx for your help anyway

I will have a closer look tomall of the messages as I know now what to look for

chrisrock1984, 2022-05-23:
I have filtered out everything eles - the only thing which looks different is in line 7.
Instead of "0" it's "{"localTime":706546910}"

[filtered.txt](https://github.com/Koenkk/zigbee2mqtt/files/8757470/filtered.txt)

chrisrock1984, 2022-05-23:
a second pir has the same localTime -> line 1

what is the difference between "ReadResponse" and "DefaultResponse" ? Sounds like, it tries to read the state of the PIR without the need to. 
This kind of PIR does not send a false on its own, does it? There is a timer in zigbee2mqtt wich sends the false to MQTT as far as I can remember. With this "ReadResponse" it looks like, that it also reads the "always" true occupancy and this is then published every 4 hours to MQTT which leads to the non triggered occupancy.

Might this be the case or am I totally wrong here ;-)

[filtered2.txt](https://github.com/Koenkk/zigbee2mqtt/files/8757519/filtered2.txt)

chrisrock1984, 2022-05-24:
Thx for clarification.

I have one question left - how would the PIR behave, when using it with a tuya gateway. According to your explanation it should also wrongly detect a motion every 4h, or?

I can test this, as I have a tuya gateway and will let you know.

chrisrock1984, 2022-05-25:
Same problem on the tuya gw.
Every 4h I get a motion detection message.
Poor tuya

This can be closed I guess

Koenkk, 2022-05-25:
> Same problem on the tuya gw.

Unfortunately as expected.

closed 2022-05-25 by Koenkk