Koenkk/zigbee2mqtt #12561: problem waking up on wireless switch legand

apages2, 2022-05-22:
### What happened?

When the legrand button is on standby, if I press one of the buttons, there is a latency of several seconds for the action to take place.

### What did you expect to happen?

When this same switch is linked to the official Legrand gateway, the action is almost immediate, it would be nice if the behavior were identical on zigbee2mqtt

### How to reproduce it (minimal and precise)

Wireless switch legrand 067724 or 067723 who is bound with wired switch legrand 067772

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

0x26720700

### Adapter

Conbee2

wurff, 2022-05-22:
I have the same behavior. The delay is sometimes several seconds after long standby.

moshemalawach, 2022-06-07:
Same behavior here, with both dev and stable releases.

Loic691, 2022-07-04:
Same behavior with J2M 1.26.0.
I haven't this problem when I used this device with offical zigbee jeedom plugin

apages2, 2022-08-05:
The problem is not present with a legrand gateway

closed 2022-09-13 by github-actions[bot]