Koenkk/zigbee2mqtt #12563: Systems stops responding: " zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request"

mhjansen79, 2022-05-22:
### What happened?

Just moved from Deconz over to Z2M. Migration went great. Added 135 items without issue. 

System runs fine, and them all of a sudden Zigbee devices no longer are responsive to inputs. 

When I look at the Z2M log within HA Addons I find the following:

zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request

When I do a restart of the add-on, I see the following:

2022-05-22 11:33:47: Zigbee: disabling joining new devices.
2022-05-22T09:33:47.356Z zigbee-herdsman:controller:log Disable joining
2022-05-22T09:33:47.358Z zigbee-herdsman:deconz:adapter zclFrame to all - zclFrame.payload:
2022-05-22T09:33:47.361Z zigbee-herdsman:deconz:adapter { options: 10, commisioningWindow: 0 }
2022-05-22T09:33:47.361Z zigbee-herdsman:deconz:adapter sendZclFrameToAll - message send
2022-05-22T09:33:47.364Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 8 request id: 1
2022-05-22T09:33:47.366Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 1 status: 6
2022-05-22T09:33:47.366Z zigbee-herdsman:deconz:driver networkstate: 1 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 0
2022-05-22T09:33:50.240Z zigbee-herdsman:deconz:adapter checking coordinator simple descriptor
2022-05-22T09:33:50.244Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-05-22T09:33:50.249Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request
2022-05-22T09:33:50.253Z zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request

My assumption is that it has to do with this line: 

2022-05-22T09:33:47.366Z zigbee-herdsman:deconz:driver networkstate: 1 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 0

Restarting the add-on does not fix the issue. Not until I restart HA completely does it resolve itself. 

Does anyone have any suggestions on how to fix this? 

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

 0x26720700

### Adapter

ConBee II

closed 2022-07-29 by github-actions[bot]

risteon, 2022-10-02:
Same issue here. Running zigbee2mqtt in docker. Sometimes it works, sometimes is does not and produces above mentioned error message.

spitfire, 2023-05-24:
Same here. Decided to stop working without an upgrade or configuration change. Everything looks like it's initialized properly until I've enabled zigbee_herdsman_debug.