Koenkk/zigbee2mqtt #12566: SONOFF zigbee 3.0 usb dongle plus Keeps crashing

Veldkornet, 2022-05-22:
### What happened?

As a continuation of https://github.com/Koenkk/zigbee2mqtt/issues/11216, every so often all devices become unavailable and I get messages like the following:

```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"wjftl-3","value":true}'
Info Zigbee: allowing new devices to join.
Error Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'SRSP - ZDO - mgmtPermitJoinReq after 6000ms'
Debug Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"SRSP - ZDO - mgmtPermitJoinReq after 6000ms","status":"error","transaction":"wjftl-3"}'
```

I currently run both MQTT and Z2M on HomeAssistant OS on a RaspberryPi 4 with a [ARGON one M.2](https://www.argon40.com/products/argon-one-m-2-case-for-raspberry-pi-4?variant=39730210275393). The Sonoff adapter is plugged into a [Nedis Powered USB HUB](https://nedis.com/en-us/product/computer/usb-hubs/usb-3-0/550705940/usb-hub-5-ports-qc30-usb-32-gen1-mains-powered-usb-powered-5-gbps-5x-usb), which is plugged into one of the usb 2.0 ports of the RPi.
The only items plugged into the hub are the SONOFF and an Aeotec ZStick.

This issue does not occur with the ConBee II.

I’ve been through the troubleshooting topic listed [here](https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time); and I’ve added `usbcore.autosuspend=-1`, but this hasn’t seem to make any difference.

It can crash every few hours or every 1-2 days, I don’t find a clear pattern.

### What did you expect to happen?

Everything should stay running and the coordinator should not crash.

### How to reproduce it (minimal and precise)

unclear, it just crashes on it’s own.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220507

### Adapter

SONOFF Zigbee USB Dongle Plus (zStack3x0)

Koenkk, 2022-05-22:
Can you provide the zigbee herdsman log starting from the point where you start z2m until it fails?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Veldkornet, 2022-05-25:
@Koenkk See log [here](https://drive.google.com/file/d/128fQbWjZof387dA9cFyhnmX0jq_pTB3W/view?usp=sharing). Looks to have started at `2022-05-25T03:48:50.628Z`. I have about 2 days worth of that debug log before the issue arose, but let me know if you need more info.

Koenkk, 2022-05-25:
- Could you try disabling the availability feature? I don't see any indication that something goes wrong in the coordinator firmware but this should reduce the stress on it.
- Does anything appear in the `dmesg` logging when the crash happens?

Veldkornet, 2022-05-26:
I've disabled Availability before, but it didn't make a difference. Only that there is a less visual way for me to see that there is an issue. 

A bit difficult to look back in dmesg...

However, I bought a new SONOFF and have moved everything over to it instead. Will see now if it works any better or has the same issues.

dajomu1, 2022-05-29:
I had the same problem with my Electrolama stick. Check if you have the Zigbee Home Automation integration installed. When I disabled the integration everything worked as it should.

Veldkornet, 2022-05-30:
Nope, same thing happens with the new dongle. It was going strong for the first couple of days, but then last night... ☠️

Also, no I do not have ZHA. 

Again, the issue does not occur when I use the ConBee II.

Bascht74, 2022-06-24:
Same here. And some Philips motion sensors get partley disconnected (still shows connection, but it sends no updated sensor data). After hard restart (power off/on) everything is fine until it happens again after some days. Pairing of new devices is not possible as well in that state...

Veldkornet, 2022-06-24:
At the moment for me, it happens once to twice per day. Most of the time, I just restart the add-on and it’s fine.
Now and then I also need to re-plug the stick.

It’s easy to spot because all of the devices become unavailable in HomeAssistant

Veldkornet, 2022-07-25:
Issue still occurs. Still haven't really discovered why...

Veldkornet, 2022-08-26:
FYI, I've updated the firmware to 20220726 and it seems to work better, but it still happens. 

Ie, longer periods between when it breaks. Now it's maybe once a week.

Minoux78, 2022-09-05:
Hello,
@Veldkornet
Where do you find thé 20220726 firmware?
For me the laste update for SONOFF Zigbee USB Dongle Plus (zStack3x0) is the 20220219.

Thanks for.your feedback

Veldkornet, 2022-09-05:
> Hello,
> 
> @Veldkornet
> 
> Where do you find the 20220726 firmware?
> 
> For me the last update for SONOFF Zigbee USB Dongle Plus (zStack3x0) is the 20220219.
> 
> 
> 
> Thanks for your feedback 

Here -> https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

popy2k14, 2022-09-15:
@Veldkornet Exact same issue here!
The cherry on top, i have bought the sonoff to replace the conbee II because sometimes deconz addon has stabilitty issues :-)

I am also running; 20220219
Did you got the stick stable?

popy2k14, 2022-09-15:
@Koenkk is this a known issue and solved with an newer firmware?

popy2k14, 2022-09-15:
@Koenkk i have enabled the herdsman logging on my hass.io supervised addon and tried to gather the logs as described in the link. There are now herdsman entries in it.

When the issue occured i will grab them an come back here.

Did you found an issue in @Veldkornet  log above?

@Veldkornet is the new beta firmware 20220726 stable for you?

Veldkornet, 2022-09-15:
The new firmware works better, but it still happens unfortunately.

inertia666, 2022-09-15:
I am having constant crashes with this thing with both the stable and develop branches. I can get it running a few minutes then it crashes out again.

popy2k14, 2022-09-16:
@Veldkornet thx for the info.

@Koenkk it happened again, here is the log file: https://drive.google.com/file/d/1NjKBKp9Qz-TPUm7kYP6pZS8MHRjekf8a/view?usp=sharing

Koenkk, 2022-09-16:
@popy2k14 the log doesn't stop with a crash, just with:

```
Zigbee2MQTT:error 2022-09-16 06:24:48: Failed to execute LQI for 'Balkon indirekt'
Zigbee2MQTT:info  2022-09-16 06:24:48: Network scan finished
Zigbee2MQTT:info  2022-09-16 06:24:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[],"nodes":[{"definition":null,"failed":["lqi"],"friendlyName":"Coordinator","ieeeAddr":"0x00124b00257cb272","lastSeen":null,"networkAddress":0,"type":"Coordinator"},{"definition":{"description":"Hue dimmer switch","model":"324131092621","supports":"battery, action, action_duration, linkquality","vendor":"Philips"},"friendlyName":"Dimmschalter Balkon","ieeeAddr":"0x0017880103e5e9a3","lastSeen":1663270959464,"manufacturerName":"Philips","modelID":"RWL021","networkAddress":24406,"type":"EndDevice"},{"definition":{"description":"Hue White A60 Single bulb E27/B22","model":"8718696449691","supports":"light (state, brightness), effect, linkquality","vendor":"Philips"},"failed":["lqi"],"friendlyName":"Balkon indirekt","ieeeAddr":"0x0017880103169f9c","lastSeen":1663267144444,"manufacturerName":"Signify Netherlands B.V.","modelID":"LWB010","networkAddress":43725,"type":"Router"},{"definition":{"description":"Hue motion outdoor sensor","model":"9290019758","supports":"temperature, occupancy, battery, illuminance_lux, illuminance, motion_sensitivity, led_indication, occupancy_timeout, linkquality","vendor":"Philips"},"friendlyName":"Bewegungsmelder Balkon","ieeeAddr":"0x00178801064432a9","lastSeen":1663272061394,"manufacturerName":"Philips","modelID":"SML002","networkAddress":49383,"type":"EndDevice"}]}},"status":"ok","transaction":"hhopd-1"}'
```

- I expect that the real crash is logged after this, can you provide the log of this?
- I see you are doing a network scan, does the crash also happen when not doing network scans?

@Veldkornet do you by any chance of have SONOFF sensors on your network (I mean battery powered sensors; not the coordinator itself)

@inertia666 I need herdsman debug logs of the crash, otherwise I cannot help.

popy2k14, 2022-09-16:
@Koenkk 
That's the whole log file. There isn't anything else in the log!
Network scans working over the day without an problem.
I am using this to see the LQI's and when there are no lines between the devices Z2M crashed.
In the morning i have done an scan to see if it crashed.

When it help's i can change my "test" and do not do an network scan the comming days and just try to see if the devices are offline.

Please let me know how i should proceed with my test.

inertia666, 2022-09-19:
I'm happy to report that my USB stick is working perfectly well now. I had misunderstood the instructions and exposed the device to two different Docker containers (Both Home Assistant and Zigbee2MQTT. Presumably, this was causing the crash. Sorry for reporting that I was experiencing crashing.

Koenkk, 2022-09-21:
@popy2k14 does z2m still answer the network map scan? If yes then z2m it self has no crashed. I do not see the finish of the network scan in your logs.

popy2k14, 2022-09-21:
@Koenkk yes, it returns an error in red.

Koenkk, 2022-09-23:
@popy2k14 no problem!

@Veldkornet is your setup stable now? If not do you by any chance of have SONOFF sensors on your network (I mean battery powered sensors; not the coordinator itself). If it is stable you can close this issue.

popy2k14, 2022-09-23:
@Koenkk i had not a single SonOff battery powered sensor on my network. Just an hue dimmer swicth gen 1, hue e27 bulb & hue outdoor motion sensor. If you need the exact types, i can get them for you out of my deconz. Just let me know.

Veldkornet, 2022-09-24:
No, it still crashes. 💩

I do not have other SONOFF devices on my network. 

I have Frient (Develco), Innr, OSRAM, Ledvance. 

I had some Aqara, but they fell off the network every time the stick crashed, so I chucked those out the window.

Koenkk, 2022-09-25:
Do you by any chance have a CC2531 sniffer and can sniff the traffic when the coordinator crashes? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

adfx, 2022-10-22:
Hey guys,

need to join in here. Same Stick, same problem (al least i think): 
`usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0`

Z2M won't start. Full Log here:

```
me@rasPi3B $ ./start_z2m_w-debug.sh 

To detach the container press ctrl-p ctrl-q

c02b55caa1a31be6284961409f4347c9699fc3b43e2ed0b1b88db55a8f7a0003
Zigbee2MQTT:debug 2022-10-22 14:52:10: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2022-10-22 14:52:10: Logging to console and directory: '/app/data/log/2022-10-22.14-52-10' filename: log.txt
Zigbee2MQTT:debug 2022-10-22 14:52:10: Removing old log directory '/app/data/log/2022-10-22.14-47-04'
Zigbee2MQTT:info  2022-10-22 14:52:10: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-10-22 14:52:10: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2022-10-22 14:52:10: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0","rtscts":false}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"rtscts":false,"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +24ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload +274ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +261ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +1s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352 +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +819ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +819ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2022-10-22 14:52:31: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-22 14:52:31: Failed to start zigbee
Zigbee2MQTT:error 2022-10-22 14:52:31: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-22 14:52:31: Exiting...
Zigbee2MQTT:error 2022-10-22 14:52:32: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
```

What is interesting in my case, it has to be an issue with the RasPi3b, because same dongle with same container works like a charme on my LapTop. I copied config and databases to my laptop and pulled new z2m:latest. I then tried to start older versions of z2m without success on my rPi.
I was thinking of permissions but sadly they're all the same. Same id, groups and so on. No other Container is using the adapter in first place. Nothing else running. 

Any help VERY appreciated as I'm out of ideas where to look next.

adfx, 2022-11-03:
Maybe @Koenkk can have a look here? Thx in advance

adfx, 2022-11-03:
maybe @Koenkk can have a look here? Thanks in advance

It has to be an issue with the adapter or permissions. If I start without any conf.yaml or database let's say a fresh install and start with --device=/dev/ttyUSB0 z2m won't pass over the adapter at all. Result: 

```
me@RasPi3B:/docker/zigbee2mqtt $ ./start_z2m_w-debug.sh 

To detach the container press ctrl-p ctrl-q

ab4af1f5f01e0486b5e64a869b98154e0016f340be5252063c17e67cf8468173
Zigbee2MQTT:debug 2022-11-03 18:23:09: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2022-11-03 18:23:09: Logging to console and directory: '/app/data/log/2022-11-03.18-23-09' filename: log.txt
Zigbee2MQTT:debug 2022-11-03 18:23:09: Removing old log directory '/app/data/log/2022-11-03.18-12-24'
Zigbee2MQTT:info  2022-11-03 18:23:09: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-11-03 18:23:09: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2022-11-03 18:23:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{}}'
  zigbee-herdsman:adapter No path provided, auto detecting path +0ms
Zigbee2MQTT:error 2022-11-03 18:23:10: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-11-03 18:23:10: Failed to start zigbee
Zigbee2MQTT:error 2022-11-03 18:23:10: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-03 18:23:10: Exiting...
Zigbee2MQTT:error 2022-11-03 18:23:11: Error: No path provided and failed to auto detect path
    at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:81:23)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:129:24)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
```

srvoets, 2022-11-22:
> Hey guys,
> 
> need to join in here. Same Stick, same problem (at least i think): `usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0`
> 
> Z2M won't start. Full Log here:
> 
> ```
> me@rasPi3B $ ./start_z2m_w-debug.sh 
> 
> 
> To detach the container press ctrl-p ctrl-q
> 
> 
> c02b55caa1a31be6284961409f4347c9699fc3b43e2ed0b1b88db55a8f7a0003
> Zigbee2MQTT:debug 2022-10-22 14:52:10: Loaded state from file /app/data/state.json
> Zigbee2MQTT:info  2022-10-22 14:52:10: Logging to console and directory: '/app/data/log/2022-10-22.14-52-10' filename: log.txt
> Zigbee2MQTT:debug 2022-10-22 14:52:10: Removing old log directory '/app/data/log/2022-10-22.14-47-04'
> Zigbee2MQTT:info  2022-10-22 14:52:10: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
> Zigbee2MQTT:info  2022-10-22 14:52:10: Starting zigbee-herdsman (0.14.62)
> Zigbee2MQTT:debug 2022-10-22 14:52:10: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0","rtscts":false}}'
>   zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"rtscts":false,"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
>   zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
>   zigbee-herdsman:adapter:zStack:znp:log Serialport opened +24ms
>   zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
>   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +0ms
>   zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload +274ms
>   zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +261ms
>   zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +1s
>   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
>   zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352 +1s
>   zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +819ms
>   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +819ms
>   zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
>   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
>   zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
>   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
> Zigbee2MQTT:error 2022-10-22 14:52:31: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2022-10-22 14:52:31: Failed to start zigbee
> Zigbee2MQTT:error 2022-10-22 14:52:31: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
> Zigbee2MQTT:error 2022-10-22 14:52:31: Exiting...
> Zigbee2MQTT:error 2022-10-22 14:52:32: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
>     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
>     at Zigbee.start (/app/lib/zigbee.ts:58:27)
>     at Controller.start (/app/lib/controller.ts:101:27)
>     at start (/app/index.js:109:5)
> ```
> 
> What is interesting in my case, it has to be an issue with the RasPi3b, because same dongle with same container works like a charme on my LapTop. I copied config and databases to my laptop and pulled new z2m:latest. I then tried to start older versions of z2m without success on my rPi. I was thinking of permissions but sadly they're all the same. Same id, groups and so on. No other Container is using the adapter in first place. Nothing else running.
> 
> Any help VERY appreciated as I'm out of ideas where to look next.

add the following to your config

adapter: ezsp

^^ I saw someone post this, and it fixed 4 days of trying to make it work, but now Z2M crashes every 15 minutes.

adfx, 2022-11-22:
Thanks for your help. I'll try this later and come back then

adfx, 2022-12-01:
So now is later :)

Finally found another container interfering with z2m. In my case raspberrymatic (Homematic container'd) needs to be switched off before starting z2m. For any reason, it looks like Homematic is taking over the stick. So it's a matter of the starting order. I'll write another docker-compose.yml which will start both container in there respective order. Clear to me now why the container was working on my laptop. No other container there interfering with z2m.

@srvoets your idea with adapter: ezsp didn't help. If set, adapter will be accessible but z2m won't start. I removed the entry and now it works like a charme. Thanks for your help anyways.

srvoets, 2022-12-01:
> So now is later :)
>
> Finally found another container interfering with z2m. In my case raspberrymatic (Homematic container'd) needs to be switched off before starting z2m. For any reason, it looks [...]

Ahhh, I pulled my hair out for days seeing the same exact timeout error over and over again, and saw people mention make sure nothing else is running that could use stick, removed z-wave uninstalled it, had same issue, so I tried it in a docker container in unraid, same issue.

only thing that made it work was adapter: ezsp

So you are running this adapter without the above 'adapter' option in your serial config?  Very interesting that yours works, and mine does not, is yours running stock firmware?  I tried for several nights to flash it using the TI tool, python app, remove case holding buttons, never got it into bootloader mode to take the new firmware.

I really wish I did though because my zigbee network runs like dogshit now, I might go back to conbee until I can get this flashed and then set it up as a router.

My other consideration is migrating Z2m from from my unraid build inside an HA VM to a spare Pi that could be placed more centrally.

However, in the original location, conbee worked without issue, I blame this whacky stick.

I even unplugged my steelseries headphone\mouse wireless adapter, but no difference.  At least it "runs"

adfx, 2022-12-11:
@srvoets FYI my docker-compose.yaml which is working:

`version: "3.0"
services:
  zigbee2mqtt:
      container_name: zig2mq
      restart: always
      image: koenkk/zigbee2mqtt
      volumes:
        - /home/eddy/docker/zigbee2mqtt/data:/app/data:rw
        - /run/udev:/run/udev:ro
      ports:
        - 8082:8080
      environment:
        - TZ=Europe/Berlin
      devices:
        - /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_089e9136cd3aec11a38ba4957a0af07f-if00-port0:/dev/ttyUSB0`

and configuration.yaml:

`homeassistant: false
permit_join: true
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://xxx.xxx.xxx.xxx
  user: MYUSER
  password: MYPASSWD
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: debug
devices:
  '0xGeneratedByZ2M:
    friendly_name: '0xGeneratedByZ2M'`

And take care of the starting order. As said in my case Z2M needs to be started first. Then Stick is attached to z2m. Hope that helps.

Just had a look to the preview. Looks like format of .yaml's lost to copy paste. Sorry for that

klaus1k, 2022-12-15:
Hello, I also switched to Sonoff Dongle-P, which ran like a charm for a week. But it stopped working almost yesterday. now its crashing almost immediately after start. Curious thing ist - didnt change anything...

Some information (everything the same for working state and crashing state):
- i followed all the instructions in Z2M FAQ if stick is crashing
- I use it with a USB extension cable.
- I have disabled access from homeassistant container to ttyUSB0
- is use it with the newest dev firmware 20221102

dmesg log shows the following:
`at start:
[  492.124327] usb 1-1.1: new full-speed USB device number 3 using xhci_hcd
[  492.233880] usb 1-1.1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  492.233884] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  492.233887] usb 1-1.1: Product: Sonoff Zigbee 3.0 USB Dongle Plus
[  492.233888] usb 1-1.1: Manufacturer: Silicon Labs
[  492.233890] usb 1-1.1: SerialNumber: 0001
[  492.790586] usbcore: registered new interface driver usbserial_generic
[  492.790602] usbserial: USB Serial support registered for generic
[  492.794352] usbcore: registered new interface driver cp210x
[  492.794377] usbserial: USB Serial support registered for cp210x
[  492.794432] cp210x 1-1.1:1.0: cp210x converter detected
[  492.799162] usb 1-1.1: cp210x converter now attached to ttyUSB0
...
[  539.416965] cp210x ttyUSB0: usb_serial_generic_write_start - error submitting urb: -19
[  539.418737] cp210x ttyUSB0: failed to get comm status: -19
[  539.418750] cp210x ttyUSB0: failed set request 0x7 status: -19
[  539.418763] cp210x ttyUSB0: failed set request 0x12 status: -19
[  539.418767] cp210x ttyUSB0: failed set request 0x0 status: -19

then recurring:

[  547.177746] cp210x ttyUSB0: failed set request 0x0 status: -19
[  547.177753] cp210x ttyUSB0: cp210x_open - Unable to enable UART
...
[  548.934547] cp210x ttyUSB0: failed set request 0x0 status: -19
[  548.934557] cp210x ttyUSB0: cp210x_open - Unable to enable UART`

Any ideas what i could try next? Am a little frustrated right now - migrated from deconz with conbeeII to Z2M with conbeeII (still some problems with lights not responding) and finally to sonoff Dongle-P. Was so happy that finally everything was working as it was supposed to. (including availability)

Koenkk, 2022-12-15:
Please try `20221214` from https://github.com/Koenkk/Z-Stack-firmware/issues/383#issuecomment-1351696706 . `20221102` is known to be unstable (sometimes)

klaus1k, 2022-12-15:
> 

Thanks for the reply! Installed it right away, but sadly the errors didn't go away. I have the feeling that the problem may be at driver/linux usb side. Will look into it, i really hope to find a solution!

adfx, 2022-12-17:
@klaus1k 
Indeed this looks like a driver issue. Have a try starting z2m with root privileges. If that helps you'll need to dig deeper in how to start containers as non-root. 

@srvoets 
I finally found the root cause. My z2m instance restarted right after homematic instance was running. So starting order wasn't the right way to fix my issue. 
For all who are having same issues and using cuxd as addon for homematic I'll advice finding cuxd.ini. There's a simple line needed which disallows making use of tty devices. Just add: TTYPARAM=NONE anywhere to that cuxd.ini. Later if a serial device is needed to be obtained by cuxd then simply define the new device in cuxd.ini and that's it. Hope this helps others avoiding a long story of pulling hairs.

klaus1k, 2022-12-17:
Tried running with root privileges, sadly didn't help. So i tried unloading and reloading `usbserial` and `cp210x` modules. Also with no effect.

Then i noticed, that i get an error from `xhci_hcd` right before the first error with `cp210x` pops up. Maybe the issue lies there? any ideas what i could try? I'm running Z2M in docker on Ubuntu 20.04 on a Minisforum HM50.

> [  151.536962] xhci_hcd 0000:05:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000f address=0xfff9e000 flags=0x0000]
[  151.536974] xhci_hcd 0000:05:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000f address=0xfff9e000 flags=0x0000]
[  151.536978] xhci_hcd 0000:05:00.0: WARNING: Host System Error
[  151.536984] xhci_hcd 0000:05:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000f address=0xfff9e000 flags=0x0000]
[  151.537474] cp210x ttyUSB0: usb_serial_generic_write_start - error submitting urb: -19
[  151.538729] cp210x ttyUSB0: failed to get comm status: -19
[  151.538736] cp210x ttyUSB0: failed set request 0x7 status: -19
[  151.538743] cp210x ttyUSB0: failed set request 0x12 status: -19
[  151.538745] cp210x ttyUSB0: failed set request 0x0 status: -19

lnagel, 2022-12-18:
> Do you by any chance have a CC2531 sniffer and can sniff the traffic when the coordinator crashes? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

I just joined this thread due to having crashes with the SONOFF ZigBee 3.0 adapter. I've upgraded today to `20221214` firmware. I also bought a zigbee sniffer adapter, so let's see. If I get crashes with the new firmware, then I'll setup the sniffer as well. If you have any other tips for how to monitor / debug this, then let us know.

Veldkornet, 2022-12-23:
> Please try `20221214` from https://github.com/Koenkk/Z-Stack-firmware/issues/383#issuecomment-1351696706 . `20221102` is known to be unstable (sometimes)

I've applied `20221220`, will let it run for a while and let you know how it goes.

Veldkornet, 2022-12-23:
> > Please try `20221214` from https://github.com/Koenkk/Z-Stack-firmware/issues/383#issuecomment-1351696706 . `20221102` is known to be unstable (sometimes)
> 
> I've applied `20221220`, will let it run for a while and let you know how it goes.

Nope, already crashed.

Koenkk, 2022-12-23:
@Veldkornet can you provide the herdsman debug log starting from the moment you start z2m until it fails?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

lnagel, 2023-01-23:
> > Do you by any chance have a CC2531 sniffer and can sniff the traffic when the coordinator crashes? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531
> 
> I just joined this thread due to having crashes with the SONOFF ZigBee 3.0 adapter. I've upgraded today to `20221214` firmware. I also bought a zigbee sniffer adapter, so let's see. If I get crashes with the new firmware, then I'll setup the sniffer as well. If you have any other tips for how to monitor / debug this, then let us know.
> 
> The nice news is that after 8 hours of downtime and then upgrading to `20221214`, then all my devices reconnected automatically. I was already dreading the network rebuild whilst driving to location.

The `20221214` version has been stable without crashes for a month. But I remember now that I added additional routers at the same time as the upgrade which may have increased the network reliability or otherwise altered the used topology. I will try to restore the original network topology and continue testing.

Veldkornet, 2023-01-31:
> @Veldkornet can you provide the herdsman debug log starting from the moment you start z2m until it fails?
> 
> 
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Right after that, I went full nuclear ☢️ Completely removed anything Z2M related, reset all devices, started from scratch. 
New PAN ID, Ext PAN ID etc. Still on that newer firmware. 

It hasn't gone down once since then, more than a month already. I've even started to revive some old sensors and add them back to the network. Going strong 💪🏼 

I was hesitant to report back at first, in case I jinx it 😄

closed 2023-03-11 by github-actions[bot]

popy2k14, 2023-03-12:
@Koenkk Back in buisiness. I have now an Sonoff Zigbee 3.0 USB Dongle Plus with latest 20221226 firmware.
Added the three devices back which made the stick crash back than.

And yes, also with the new firmware, it crashes!

Here is the herdsman log:

`Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.394Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.394Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.397Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,137,79]
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.398Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,137,79]
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.398Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,137] - 79
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.398Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":137}
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.398Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.411Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,8,0,48,127,11,1,0,127,0,171,159,47,0,0,5,24,60,11,4,0,48,127,29,137]
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.411Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,8,0,48,127,11,1,0,127,0,171,159,47,0,0,5,24,60,11,4,0,48,127,29,137]
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.411Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,8,0,48,127,11,1,0,127,0,171,159,47,0,0,5,24,60,11,4,0,48,127,29] - 137
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.411Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":32560,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":127,"securityuse":0,"timestamp":3121067,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,60,11,4,0]}}
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.412Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":60,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":4,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":32560,"endpoint":11,"linkquality":127,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: Zigbee2MQTT:info  2023-03-12 19:54:02: MQTT publish: topic 'zigbee2mqtt/Balkon indirekt (Z2M)', payload '{"brightness":254,"color_mode":"xy","last_seen":"2023-03-12T19:54:02+01:00","linkquality":127,"power_on_behavior":null,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:02.414Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: Zigbee2MQTT:info  2023-03-12 19:54:02: MQTT publish: topic 'zigbee2mqtt/Alle', payload '{"brightness":254,"color_mode":"xy","state":"ON"}'
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: Zigbee2MQTT:info  2023-03-12 19:54:02: MQTT publish: topic 'zigbee2mqtt/Balkon indirekt (Z2M)', payload '{"brightness":254,"color_mode":"xy","last_seen":"2023-03-12T19:54:02+01:00","linkquality":127,"power_on_behavior":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: Zigbee2MQTT:info  2023-03-12 19:54:02: MQTT publish: topic 'zigbee2mqtt/Alle ohne Weihnachtsbeleuchtung', payload '{"brightness":254,"color_mode":"xy","state":"ON"}'
Mar 12 18:54:02 homeassistant addon_45df7312_zigbee2mqtt[461]: Zigbee2MQTT:info  2023-03-12 19:54:02: MQTT publish: topic 'zigbee2mqtt/Balkon', payload '{"brightness":254,"color_mode":"xy","state":"ON"}'
Mar 12 18:54:03 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:03.231Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,33,68,129,0,0,0,252,95,239,2,242,0,142,0,206,103,48,0,0,13,29,11,16,101,0,4,0,0,48,0,33,0,0,95,239,29,153]
Mar 12 18:54:03 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:03.231Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,33,68,129,0,0,0,252,95,239,2,242,0,142,0,206,103,48,0,0,13,29,11,16,101,0,4,0,0,48,0,33,0,0,95,239,29,153]
Mar 12 18:54:03 homeassistant addon_45df7312_zigbee2mqtt[461]: 2023-03-12T18:54:03.231Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 33 - 2 - 4 - 129 - [0,0,0,252,95,239,2,242,0,142,0,206,103,48,0,0,13,29,11,16,101,0,4,0,0,48,0,33,0,0,95,239,29] - 153
[... cut ...]
It crashed (doesnt work) arround: 18:54:12
Yes, i have made an network scan afterwards to check if any device is responding, sadly not.

BUT this time i added the devices in the following order:

 * Philips 324131092621 (Dimmer RWL021)
 * Philips	8718696449691 (Light LWB010)
 * Philips	9290019758 (Motion sensor SML002)

It was without an crash ~6h but after i added the last devic (SML002) it crashed twice in short time (minutes).

Maybe there is an issue with the hue motion sensor?

How can i help to debug this?

popy2k14, 2023-03-12:
Found the following in my dmesg around that time it failed:

```
➜  ~ dmesg | grep cp210x
[12.03.2023 18:52:26] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[12.03.2023 18:52:26] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[12.03.2023 19:52:30] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[12.03.2023 19:52:30] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
➜  ~
```

That seemed to be an long ongoing kernel bug which is not solved up until now.
See:

https://github.com/raspberrypi/linux/issues/2406
https://github.com/raspberrypi/linux/issues/5211
https://forum.odroid.com/viewtopic.php?t=40719

Ps.: Running an Odroid N2+

The issue seems to affect various manufactures of USB uarts, see the issues linkes above.

What seems to help is using an active powered usb hub in between, not because of the high power draw of the usb-uart devices, rather to wokaround the issue.

If the issue pops up again, i will throw an unpowered usb hub in between and see if the issue is gone.

What's a little bit wierd is, that my conbee II with deconz addon running good for months.
conbee II is using an ftdi.
Maybe deconz is doing some reopening when uart/stick is lost!?

Does zigbee2mqtt some retries of this sort?
If not, this would be a nice feature.

popy2k14, 2023-03-14:
Today it crashed again, in dmesg are the exact same messages. Will migrate to an x86 hardware and see if the issue is gone. Sorry for spamming this thread. But maybe this info still help someone, with the same issues.

Will post an update after the migration with the exact same config (HA full backup) when it's done.

popy2k14, 2023-03-19:
Running the new x86 hardware with an HA backup for an day and so far so good, no crash!
Seems the issue is really kernel arm based.

popy2k14, 2023-04-01:
After 2 weeks, Z2M on x86 with sonoff dongle p is rock solid!

MOST2K2, 2023-05-09:
> @klaus1k Indeed this looks like a driver issue. Have a try starting z2m with root privileges. If that helps you'll need to dig deeper in how to start containers as non-root.
>
> @srvoets I finally [...]

You saved my day, big thanks!

EpicLPer, 2023-05-16:
This might be a "stupid" thing to blurt into this conversation here, but everything was fine and dandy for the past months without any hiccups or anything (apart from some huge delayed response times from devices but that's beyond the point).  
A few days ago my grandpa did something on our electrical, which tripped the breaker and despite having a UPS somehow made my Sonoff Dongle Plus crash... and ever since then it tends to crash randomly without any warnings. 2 days ago it crashed without anyone tripping our breaker, and yesterday it did the same.

Just as a "this is my experience" kind of info dump here :)

Humberd, 2025-12-05:
I still confirm that my dongle keeps crashing. I updated to the latest firmware without a success.

popy2k14, 2026-01-09:
@Humberd Do you see the kernel log as stated above in your dmesg?
If yes, than it's the same issue i faced above and it not fixed in the linux kernel/driver.
If not, it could be a firmware issue. Please try older firmware 202211226 as stated by the developer here: https://github.com/Koenkk/Z-Stack-firmware/discussions/545#discussioncomment-15407652

popy2k14, 2026-01-09:
@Humberd thx for this info.
Conbee 3 uses an EFR32MG21 rf frontend and an other uart/serial to USB chip.
Maybe that's the reason why the kernel/zstack bug not comes up.