Koenkk/zigbee2mqtt #12567: [Feature request]: Allow automatically updating firmware for devices with OTA upgrades

carlos-sarmiento, 2022-05-22:
### Is your feature request related to a problem? Please describe

I run a large number of Philips Hue devices and keeping them up-to-date is a rather involved process on Z2M because I need to update each device one-by-one. Since the network can't really withstand more than one device updating at the same time, I have to babysit the process.

### Describe the solution you'd like

It would be great if there could be an option to either do automatic ota upgrades or a button to queue firmware updates for devices so that as soon as one is done, the next one gets updated.

### Describe alternatives you've considered

I don't see any alternatives for my request. 

### Additional context

N/A

mike-scofield, 2022-06-18:
Agreed.  I have a bunch of homebuilt devices, defined through my own `external_converters` section, which periodically request FW images.  These images are hosted locally in my data/ directory so the FW requests will not generate internet traffic.

I would like for zigbee2mqtt to automatically upgrade _these_ devices when appropriate, since I know it is safe.  I'll acknowledge that it might not be a good idea to always enable FW updates by default for _all_ devices.

Browsing through the code, I see that onZigbeeEvent() is hardwired to decline OTA updates even when the device asks for one:

```
        // Respond to the OTA request:
        // - In case we don't support OTA: respond with NO_IMAGE_AVAILABLE (0x98) (so the client stops requesting OTAs)
        // - In case we do support OTA: respond with ABORT (0x95) as we don't want to update now.
        const endpoint = data.device.zh.endpoints.find((e) => e.supportsOutputCluster('genOta'));
        if (endpoint) {
            // Some devices send OTA requests without defining OTA cluster as input cluster.
            await endpoint.commandResponse('genOta', 'queryNextImageResponse', {status: supportsOTA ? 0x95 : 0x98});
        }
```

The way to get an OTA update kicked off, instead, goes through onMQTTMessage() -> device.definition.ota.updateToLatest() from zigbee-herdsman-converters.  It would be nice if we could optionally configure this logic to be automatically triggered during the device-initiated check.

sjorge, 2022-06-18:
This should be opt-in though, I have a bunch of device I absolutely do not want to upgrade. e.g. some of the IKEA buttons and I can't imagine I'm the only one.

Or maybe even filtered so you can exclude devices from automatically getting updates so you get best of both worlds.

carlos-sarmiento, 2022-07-19:
I'd like to keep this feature alive. Is there a way to avoid getting it marked as stale?

carlos-sarmiento, 2022-08-19:
Is there any chance this feature could be built into the system? I'm rather keen on it

carlos-sarmiento, 2022-09-20:
Keeping it alive

carlos-sarmiento, 2022-10-21:
Is there a way to call the API for zigbee2mqtt to at least write an external script that can update devices?

fabicodes, 2022-10-29:
> Is there a way to call the API for zigbee2mqtt to at least write an external script that can update devices?

You can find the OTA specific MQTT Topics [here](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#manually-check-if-an-update-is-available).
I've just created a script by the way which at least works for me. You can find it here: [fabicodes/zigbee2mqtt_ota_updater](https://github.com/fabicodes/zigbee2mqtt_ota_updater). There still may be bugs around, but as it's just getting a device list, checking for available OTA Updates and then sequentially performing those it shouldn't be a big issue.

carlos-sarmiento, 2022-12-03:
@fabicodes your script is pretty cool. Thanks for sharing

closed 2023-01-10 by github-actions[bot]

filikun, 2023-06-01:
If you are using Home Assistant the OTA's are exposed in the state of the device and could be automated to update whenever a new firmware update is available.

filikun, 2023-06-01:
> I think it would be great if we can get an entity from Z2M that indicates if an upgrade is in progress. That way we could automate the upgrades from HA

I think it does that already? When an update is present it shows up in the HA UI and there is an exposed update state on the device itself. I'm in the progress of updating a device right now and it looks like this

[image]

carlos-sarmiento, 2023-06-01:
I meant it as a global signal from Z2M so an automation can tell if it should start updating another component or wait until that update is ready.

0BanZai0, 2024-10-23:
60 Updates pending rn 😅 I'd live this feature! Sometime with a clever script out there?