Koenkk/zigbee2mqtt #12574: [New device support]: Bouton Intelligent Boulanger

jgaret, 2022-05-23:
### Link

https://www.boulanger.com/ref/8009289

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x0c4314fffe1bc1a9","nwkAddr":32544,"manufId":4872,"manufName":"Sourcing & Creation","powerSource":"Battery","modelId":"EB-SB-1B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2048,"inClusterList":[0,1,3,2821,4096,64769],"outClusterList":[3,4,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"EB-SB-1B","manufacturerName":"Sourcing & Creation","powerSource":3,"zclVersion":3,"appVersion":41,"stackVersion":6,"hwVersion":2,"dateCode":"20210618","swBuildId":"2.09.01"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":41,"stackVersion":6,"hwVersion":2,"dateCode":"20210618","swBuildId":"2.09.01","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1653305897363,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new device page, but after adding the converter, I don't have any log on any action I try on the button (single click, long click or other actions), I am not really sure of what I should put in the exposes key (i've retrieved the example from here : https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/freepad_ext.js)

I have the following log after pairing :
zigbee2mqtt    | Zigbee2MQTT:info  2022-05-23 13:38:17: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Bouton intelligent","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["button_*_hold","button_*_single","button_*_double","button_*_triple","button_*_quadruple","button_*_release"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"EB-SB-1B","options":[],"supports_ota":false,"vendor":"Sourcing & Creation"},"friendly_name":"0x0c4314fffe1bc1a9","ieee_address":"0x0c4314fffe1bc1a9","status":"successful","supported":true},"type":"device_interview"}'
zigbee2mqtt    | Zigbee2MQTT:info  2022-05-23 13:38:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Bouton intelligent","friendly_name":"0x0c4314fffe1bc1a9","model":"EB-SB-1B","supported":true,"vendor":"Sourcing & Creation"},"type":"pairing"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2022-05-23 13:39:03: Received Zigbee message from '0x0c4314fffe1bc1a9', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":571015174,"imageType":45,"manufacturerCode":4872}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2022-05-23 13:39:03: Device '0x0c4314fffe1bc1a9' requested OTA
zigbee2mqtt    | Zigbee2MQTT:debug 2022-05-23 13:41:30: Saving state to file /app/data/state.json

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['EB-SB-1B'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'EB-SB-1B', // Vendor model number, look on the device for a model number
    vendor: 'Sourcing & Creation', // Vendor of the device (only used for documentation and startup logging)
    description: 'Bouton intelligent', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.action(['button_*_hold', 'button_*_single', 'button_*_double', 'button_*_triple', 'button_*_quadruple', 'button_*_release'])], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

closed 2022-06-30 by github-actions[bot]

Grojojo40, 2023-12-10:
this device was added as part of 1.30.2 but is still discovered as unsupported.

jvanoost, 2024-01-14:
Bonjour,
Dsl je déterre un vieux sujet mais je galère a inclure ce bouton dans z2m.
Tu as une technique @jgaret? Merci

vieurou, 2025-01-18:
> Bonjour, Dsl je déterre un vieux sujet mais je galère a inclure ce bouton dans z2m. Tu as une technique [@jgaret](https://github.com/jgaret)? Merci

+1 j'en ai acheter plusieurs en pensant que c'était ZigBee, et donc qu'il serait compatible, mais a part de l'apairer à une ampoule en direct, je n'arrive à rien

jvanoost, 2025-01-20:
> +1 j'en ai acheter plusieurs en pensant que c'était ZigBee, et donc qu'il serait compatible, mais a part de l'apairer à une ampoule en direct, je n'arrive à rien

bonjour,
J'ai réussi l'an dernier à l'inclure dans z2m mais après pas mal de bataille.
J'ai passé le bouton en mode appairage, puis j'appuyais sur le bouton chaque seconde jusqu'à ce que la config soit terminée dans z2m.
ça a fonctionné pendant un moment, puis un jour il a perdu la connexion (je pense que c'est la pile qui m'a laché) et après ça, je n'ai jamais réussi à le remettre dedans... finalement j'ai mis un autre bouton