Koenkk/zigbee2mqtt #12580: Frontend port doesn't change, despite changing the configuration.yaml file

rainbowbreeze, 2022-05-24:
### What happened?

Home Assistant Supervisor 2022.05.3 env

Despite the port value I put in configuration.yaml file, z2m logs always show frontend loads on port 8099

```
frontend:
  port: 8080
```

or

`frontend: true`

or

```
frontend:
  port: 7543
```

All the other param changes in configuration.yaml are taken into account when I reload z2m add-on (_for example, I tried to change the adapter port, the log level, etc_)

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

0x26720700

### Adapter

Conbee II

### Debug log

```
info  2022-05-24 01:43:25: Connecting to MQTT server at mqtt://core-mosquitto:1883
debug 2022-05-24 01:43:25: Using MQTT login with username: addons
info  2022-05-24 01:43:26: Connected to MQTT server
info  2022-05-24 01:43:26: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
[...]
info  2022-05-24 01:43:26: MQTT publish: topic 'zigbee2mqtt/0x0004740000937ec5/availability', payload 'online'
debug 2022-05-24 01:43:26: Active device '0x0004740000939ee8' was last seen '2.08' minutes ago.
info  2022-05-24 01:43:26: MQTT publish: topic 'zigbee2mqtt/0x0004740000939ee8/availability', payload 'online'
info  2022-05-24 01:43:26: MQTT publish: topic 'zigbee2mqtt/901/availability', payload 'online'
info  2022-05-24 01:43:26: Started frontend on port 0.0.0.0:8099
```

Koenkk, 2022-05-24:
The port is always forced to `8099`: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/blob/8c857174e244a3d0fe38793efb46249535874cb2/common/rootfs/etc/services.d/zigbee2mqtt/run#L9, Home Assistant ingress assumes this port (so it doesn't make sense to specify any other). https://developers.home-assistant.io/docs/add-ons/presentation/#ingress

Koenkk, 2022-05-26:
Added this, assuming this can be closed now.

closed 2022-05-26 by Koenkk