Koenkk/zigbee2mqtt #12587: ZBMINI-L stop responding

ccorderod, 2022-05-24:
### What happened?

ZBMINI-L stop responding. This may happen a few minutes after restarting Zigbee2mqtt or a few days.

ZBMINI-L on firmware 1.1.1. Previously on 1.0.5 with same behavior.

This problem also appears on ZB2MQTT 1.25.0 with the same adapter running previous firmware version.

### What did you expect to happen?

ZBMINI-L to stay responsive

### How to reproduce it (minimal and precise)

(re)start Zigbee2mqtt and wait. After a few minutes or a few days, ZBMINI-L start sending messages every second or so and stop responding.
Restarting Zigbee2mqtt fix the problem.

### Zigbee2MQTT version

1.25.1-dev commit: c00768b7

### Adapter firmware version

20220507

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

### Debug log

`2022-05-24 23:23:55Received Zigbee message from 'LucesMueblesCocina', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Info 2022-05-24 23:23:55MQTT publish: topic '10/49/LucesMueblesCocina', payload '{"last_seen":"2022-05-24T23:23:55+02:00","linkquality":65,"power_on_behavior":"off","state":"OFF","update":{"state":null},"update_available":null}'
Debug 2022-05-24 23:23:55Received Zigbee message from 'LucesMueblesCocina', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2022-05-24 23:23:55MQTT publish: topic '10/49/LucesMueblesCocina', payload '{"last_seen":"2022-05-24T23:23:56+02:00","linkquality":65,"power_on_behavior":"off","state":"OFF","update":{"state":null},"update_available":null}'
Info 2022-05-24 23:23:55MQTT publish: topic '10/49/LucesMueblesCocina', payload '{"last_seen":"2022-05-24T23:23:56+02:00","linkquality":65,"power_on_behavior":"off","state":"OFF","update":{"state":null},"update_available":null}'
Debug 2022-05-24 23:23:56Received Zigbee message from 'LucesMueblesCocina', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Info 2022-05-24 23:23:56MQTT publish: topic '10/49/LucesMueblesCocina', payload '{"last_seen":"2022-05-24T23:23:56+02:00","linkquality":65,"power_on_behavior":"off","state":"OFF","update":{"state":null},"update_available":null}'
2022-05-24 23:23:57Received Zigbee message from 'LucesMueblesCocina', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2022-05-24 23:23:57MQTT publish: topic '10/49/LucesMueblesCocina', payload '{"last_seen":"2022-05-24T23:23:57+02:00","linkquality":72,"power_on_behavior":"off","state":"OFF","update":{"state":null},"update_available":null}'
Info 2022-05-24 23:23:57MQTT publish: topic '10/49/LucesMueblesCocina', payload '{"last_seen":"2022-05-24T23:23:57+02:00","linkquality":65,"power_on_behavior":"off","state":"OFF","update":{"state":null},"update_available":null}'
Debug 2022-05-24 23:23:57Received Zigbee message from 'LucesMueblesCocina', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Info 2022-05-24 23:23:57MQTT publish: topic '10/49/LucesMueblesCocina', payload '{"last_seen":"2022-05-24T23:23:58+02:00","linkquality":65,"power_on_behavior":"off","state":"OFF","update":{"state":null},"update_available":null}'`

Koenkk, 2022-05-25:
- Do you have `permit_join: true` set in your `configuration.yaml`? If yes try disabling this.
- With the next z2m release this device gets ota support, maybe an OTA update of this device will fix it. You can already check it out in the dev branch: (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ccorderod, 2022-05-26:
Hi Koenkk.
permit_join is false in my config.
The z2m release i am using is the dev branch. Ota support is enabled on this release as i upgraded the zbmini-l firmware from 1.0.5 to 1.1.1

streaky, 2022-05-27:
Have the same issue with same device but permit join was on, I actually just turned it off before finding this ticket and will switch to dev branch. I don't think anything about this comment is very useful but just confirming that it has happened to me also (second time) now - will probably switch to dev and see if it helps also.

ccorderod, 2022-05-30:
@Koenkk, is there is any way I can help you to investigate this issue? it is really annoying ...

Koenkk, 2022-05-30:
@ccorderod disappointing that the update doesn't fix it. I cannot do anything myself here, but if you can provide a sniff starting from the point where the device works until it fails I can provide that to Sonoff so they can investigate.

closed 2022-07-07 by github-actions[bot]

isabido, 2022-08-18:
Same problem here too, after restarting HA/Z2M for some maintenance task, the ZBMINI-Ls stop communicating. I have two drives, and one of them even sitting in the configuration.yaml is not visible in the map/device list.

For me these devices have been a real waste of money, apart from a disappointment that they cannot function as a router, they could at least give the possibility of feeding it with neutral and that they were more stable.

ZBMINI-L version 1.1.1
Z2M 1.27.0 commit: a9b8808
Sonof Dongle Coordinator + 20220726

kostasdimos, 2022-09-22:
Same problem here too with three zbmini-l updated to 1.1.1. All of then are not responding 

I deleted one of these and tried to repair it.
Now  the devise seems as "not supported"

ZBMINI-L version 1.1.1
Sonoff ZbBridge Tasmota 11.0.0.1
Zigbee2MQTT: 1.27.2-1

michaelbeyhs, 2022-10-11:
Same issue here.
I pair the device. It works perfectly for a while.
Then for no apparent reason anywhere from a few hours to a day the device becomes unresponsive.
When I look at the status in z2m I see it get updated about once a second, but I’m unable to control the device.

Sometimes the device randomly turns on and off the connected lamp.

I have tried 3 different devices with the same result.

interestingly I have another device connected to another lamp, that does not exhibit this behavior.

I plan on swapping the two devices to see if it maybe has anything to do with the wiring or connected lamps

michaelbeyhs, 2022-10-11:
I connected one of the devices that was unreliable to a different lamp where a second device was reliable, and this previously flaky device now seems to work reliably there.

I’m wondering if it has something to do with the wiring or connected lamps. Maybe device is unable to leak the necessary current through the connected lamp and somehow crashes. I wasn’t able to find any information about requirements of the connected load.

michaelbeyhs, 2022-10-11:
Here’s a screen recording of this situation
Device is unresponsive, but state is updated multiple times a second

https://user-images.githubusercontent.com/12388804/195034674-173ecda3-78b1-4bf9-8e77-5b513cf2bed6.MOV

kostasdimos, 2022-10-11:
One more thing...

I paired my three devices with with another ZBbrige (without tasmota) and ewelink.
Even with the ewelink i have some disconnections but less often.

when offline, if i locally change the switch status the device becomes "connected" again on the ewelink.
It seems that the problem is on switch side an not on the z2q.

vw-noise, 2022-11-27:
any updates on this?
I have the same issues with 4 of these devices. they are running fine for about one day - that's it.

Firmware 1.1.1. and Zigbee2MQTT 1.28.2-1

michaelbeyhs, 2022-11-28:
Not from my side. 
I gave up and moved to Shelly 1L devices.

elmasryahmed, 2022-12-09:
I’d the same issue with the ZBMINI-L connected to Z2M or Hubitat, it’s a device signal issue, I solved it by adding an (IKEA) repeater almost 3 feet away, now it’s working fine.

vw-noise, 2022-12-11:
thx for your answers.

in the meantime I tried some things and found out that
it depends on the load produced by the devices to be switched.
The problems occur because the power supply of my LEDs has only 3.5 watts.
If I hang a normal light bulb on the switch, then it works perfectly for several days.
The same occurs with the normal Mini Switch from Sonoff.

I came up with it, because the Shellys is also always referred to the bypass, which you should include if a load of less than 20 watts, if you do not use a neutral conductor.
Have now also connected to test a Shelly 1l with neutral - which works great with my power supplies, but just over Wifi and not over Zigbee.

schollex, 2023-11-25:
> I’d the same issue with the ZBMINI-L connected to Z2M or Hubitat, it’s a device signal issue, I solved it by adding an (IKEA) repeater almost 3 feet away, now it’s working fine.

I can also confirm that's a signal issue.
I have two ZBMINI-L.
The one with bigger distance to the coordinator regularly becomes unreachable.
I interchanged the connected loads => no change.
I interchanged the 2 ZBMINI => now the one which was nearer and then got farer from the coordinator became regularly unavailable.

The available ZBMINI-L has a singal strength of 110, the weak one 70-90.
Beside the weak one is a battery powered door contact SNZB-01 => no problems yet!