Koenkk/zigbee2mqtt #12588: Jasco 43132 does not show Energy or Power values

haforme, 2022-05-25:
### What happened?

According to the documentation the Jasco 43132 should report power and energy usage.
https://www.zigbee2mqtt.io/devices/43132.html#jasco-43132

The device joins the zigbee network without any issues but the power and energy readings always show null when the connected device is drawing power.

[image]

### What did you expect to happen?

That the Energy and Power values be populated when the attached device is drawing power.

### How to reproduce it (minimal and precise)

Join the device as normal and observe the values reported while a attached device is drawing power.

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

0x26580700

### Adapter

Conbee2

haforme, 2022-06-25:
This is still an active issue.

Koenkk, 2022-06-26:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['43132'],
    model: '43132',
    vendor: 'Jasco',
    description: 'Zigbee smart outlet',
    extend: extend.switch(),
    fromZigbee: [...extend.switch().fromZigbee, fz.metering],
    exposes: [e.switch(), e.power()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.instantaneousDemand(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

jronnen7, 2022-06-29:
@Koenkk aha, I figured out my issue, the above code example does work, I was editing the wrong configuration.yaml file that is why I wasn't able to get it to work yesterday.

I also found a way to include energy by providing the `currentSummDelivered`.

I landed with the following

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['43132'],
    model: '43132',
    vendor: 'Jasco',
    description: 'Zigbee smart outlet (test)',
    extend: extend.switch(),
    fromZigbee: [...extend.switch().fromZigbee, fz.metering],
    exposes: [e.switch(), e.power(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        
        await Promise.all([
            reporting.onOff(endpoint),
            reporting.readMeteringMultiplierDivisor(endpoint),
            reporting.instantaneousDemand(endpoint),
            reporting.currentSummDelivered(endpoint)
        ])
    }
};

module.exports = definition;
```

@haforme notice working power and energy values now 🎉 
[image]

Koenkk, 2022-06-29:
cool, i've integrated the changes!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-29 by Koenkk