Koenkk/zigbee2mqtt #12589: [Feature request]: Hue Dimmer - Multiple press actions

BlunderCig, 2022-05-25:
### Is your feature request related to a problem? Please describe

It is fairly difficult to implement and manage automations to handle multiple button presses. I am trying to create a setup for my Hue Dimmer where you press the on button a number of times to select a "mode", and a single press toggles the light. I have this sort-of working, but every time I change modes I am toggling the light a number of times because I cannot figure out how to make an automation in HA wait for a second button press, and cancel if that button press happens, but toggle the light if there is not a second button press. I understand this is partially a user problem, but it should not be this tricky. I believe this could be much, much better if zigbee2mqtt would report multiple presses.

### Describe the solution you'd like

I would like actions for multiple button presses. Maybe 5 per button to be somewhat reasonable, or potentially even a dropbox where you select the number of clicks you'd like to see reported. This way you get the option of using however many you like while avoiding the clutter of seeing several press options if you do not want so many.  

### Describe alternatives you've considered

I have tried to create this via automations in Home Assistant. It does appear to be possible, but it is cumbersome and I am having trouble with it. I have also considered using a different Zigbee software which supports multiple key presses, but I find zigbee2mqtt so much better in every other regard and really do not wish to switch. 

### Additional context

None

andymk3, 2022-05-30:
@BlunderCig just seen your post on Reddit while searching for this myself. Having now seen it's possible with ZHA, I'd be very happy if someone could make it work with Z2M.

I'm slowly moving all my Hue stuff onto Z2M away from the Hue hub, but the lack of this feature is a bit of a killer right now.

mrjjr1980, 2022-05-30:
Just found out a workaround: make a input_select with a few options. make an automation to set a scene when an each different option is set. Make an automation triggerd by a button press of the hue dimmer and call service to input_select.select_next, now you have the same behaviour. Additionally you can reset de input_select to the first value after a couple of minutes so always the same next scene is called. You can also make an if-then to first switch on the lights based on an off state.

BlunderCig, 2022-06-03:
I've also found workarounds now. It has taken a bit of time to dial in "wait for time to pass" actions as well as time-based state conditions to stop the lights from toggling when trying to change "modes".  

However the majority of these automations would still end up being quite messy and now I've got everything working I can see why it might not be a huge priority to implement this feature. Also, different people may have different tolerances for delays. It would require a lot of settings being added to Z2M which evidently can just be handled elsewhere for now.

closed 2022-07-12 by github-actions[bot]

soeren-gasch, 2024-02-28:
Would love this one, too