Koenkk/zigbee2mqtt #12596: aqara p1 motion sensor - bug

heapxor, 2022-05-26:
### What happened?

It is often reporting 656xx lux of illuminance. 
Basically unusable for any type of automation ... as a result that lux are completely reporting nonsense values.

motion_sensitivity
detection_interval
both are not able to be changed .... 

### Zigbee2MQTT version

1.25.1-1 

### Adapter firmware version

0.0.0_0005

### Adapter

sonoff usb 3.0 stick

Otnow, 2022-05-26:
> It is often reporting 656xx lux of illuminance. Basically unusable for any type of automation ... as a result that lux are completely reporting nonsense values.

Please record a [debug log](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee2mqtt-debug-logging) that captures the moment of the problem.

***

> motion_sensitivity detection_interval both are not able to be changed .... Publish 'set' 'motion_sensitivity' to 'motion-sensor-xxx' failed: 'Error: Write 0x54ef44100043e436/1 aqaraOpple({"268":{"value":3,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'

Immediately after changing the settings of the sensor, you need to press the button on it so that it wakes up and accepts the changes.

heapxor, 2022-05-26:
@otnow thanks,
it was also mentioned here >
https://www.reddit.com/r/homeassistant/comments/ud22gp/comment/i6jk62g/?utm_source=share&utm_medium=web2x&context=3
https://community.smartthings.com/t/aqara-motion-sensor-lux-readings/242209/10
https://github.com/home-assistant/core/issues/34392

I set in configuration 
advanced:
  log_level: debug
but where is that log? the moment when it happens is random  - it absolutely doesnt read lx number correctly ... interval of reading are random, it goes from 0 to 6554 randomly etc... ;(

Otnow, 2022-05-26:
@heapxor, Aqara motion sensors report illuminance only _at the moment of motion detection, when a button is pressed and without any conditions, once every ~ 55 minutes_, i.e. they do not react to the fact of changing the level of illuminance.

The [RTCGQ14LM](https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html) model uses a converter from [RTCGQ12LM](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html), for which **0 Lux** illuminance _at the moment of motion detection_ corresponds to the message from the sensor `{"illuminance": 65536}`, i.e. `illuminance = msg.data['illuminance'] - 65536`.

Apparently, the manufacturer in the _RTCGQ14LM_ model has changed the formula for calculating illuminance.

Accordingly, we need to see exactly what values ​​the sensor will send at different levels of illuminance.

Please record the following 3 debug logs:

- the moment the movement starts and then once button press in a completely dark place (or you can cover the sensor with something opaque) to see the value that the sensor will send with a guaranteed illuminance level of **0 Lux**

- the moment the movement starts and then once button press inn medium light conditions (usual conditions)

- the moment the movement starts and then once button press in high light conditions (for example, you can bring the sensor close to a bright lighting lamp)

heapxor, 2022-05-26:
@Otnow 
yes, thanks. i was not aware of that... but makes sense now. Even tho / there was motion etc...and it shows 
[image]
instead of ~0lx.

i might be replicate 
1) 
`debug 2022-05-26 21:59:15Received Zigbee message from 'motion-sensor-chodba-aqara', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":131070}' from endpoint 1 with groupID 0
debug 2022-05-26 21:59:15lumi.motion.ac02: unknown key illuminance with value 131070
debug 2022-05-26 21:59:15lumi.motion.ac02: Processed data into payload {}
info 2022-05-26 21:59:15MQTT publish: topic 'zigbee2mqtt/motion-sensor-chodba-aqara', payload '{"battery":100,"detection_interval":20,"illuminance":65534,"linkquality":87,"motion_sensitivity":"high","occupancy":true,"state":"OFF","temperature":27,"update":{"state":null},"update_available":null,"voltage":3120}'
debug 2022-05-26 21:59:23Received Zigbee message from 'motion-sensor-chodba-aqara', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.motion.ac02"}' from endpoint 1 with groupID 0
debug 2022-05-26 21:59:28Received Zigbee message from 'motion-sensor-chodba-aqara', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,29,12,3,40,27,4,33,0,0,5,33,1,0,8,33,5,1,10,33,130,37,12,32,1,19,32,0,20,32,0,100,16,0,101,33,254,255,105,32,20,106,32,3,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2022-05-26 21:59:28lumi.motion.ac02: Processed buffer into data {"1":3101,"3":27,"4":0,"5":1,"8":261,"10":9602,"12":1,"19":0,"20":0,"100":0,"101":65534,"105":20,"106":3,"107":0}
debug 2022-05-26 21:59:28lumi.motion.ac02: unknown key 8 with value 261
debug 2022-05-26 21:59:28lumi.motion.ac02: unknown key 12 with value 1
debug 2022-05-26 21:59:28lumi.motion.ac02: unknown key 19 with value 0
debug 2022-05-26 21:59:28lumi.motion.ac02: unknown key 20 with value 0
debug 2022-05-26 21:59:28lumi.motion.ac02: unknown key 107 with value 0
debug 2022-05-26 21:59:28lumi.motion.ac02: Processed data into payload {"voltage":3101,"battery":100,"temperature":27,"state":"OFF","illuminance":65534,"detection_interval":20,"motion_sensitivity":"high"}
debug 2022-05-26 21:59:28lumi.motion.ac02: Processed data into payload {"voltage":3101,"battery":100,"temperature":27,"state":"OFF","illuminance":65534,"detection_interval":20,"motion_sensitivity":"high"}
info 2022-05-26 21:59:28MQTT publish: topic 'zigbee2mqtt/motion-sensor-chodba-aqara', payload '{"battery":100,"detection_interval":20,"illuminance":65534,"linkquality":80,"motion_sensitivity":"high","occupancy":true,"state":"OFF","temperature":27,"update":{"state":null},"update_available":null,"voltage":3101}'`

2)
`debug 2022-05-26 22:03:56Received Zigbee message from 'motion-sensor-chodba-aqara', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":65552}' from endpoint 1 with groupID 0
debug 2022-05-26 22:03:56lumi.motion.ac02: unknown key illuminance with value 65552
debug 2022-05-26 22:03:56lumi.motion.ac02: Processed data into payload {}
info 2022-05-26 22:03:57MQTT publish: topic 'zigbee2mqtt/motion-sensor-chodba-aqara', payload '{"battery":100,"detection_interval":20,"illuminance":16,"linkquality":83,"motion_sensitivity":"high","occupancy":true,"state":"OFF","temperature":27,"update":{"state":null},"update_available":null,"voltage":3101}'
debug 2022-05-26 22:04:04Received Zigbee message from 'motion-sensor-chodba-aqara', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.motion.ac02"}' from endpoint 1 with groupID 0
debug 2022-05-26 22:04:09Received Zigbee message from 'motion-sensor-chodba-aqara', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,31,12,3,40,27,4,33,0,0,5,33,1,0,8,33,5,1,10,33,130,37,12,32,1,19,32,0,20,32,0,100,16,0,101,33,15,0,105,32,20,106,32,3,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2022-05-26 22:04:09lumi.motion.ac02: Processed buffer into data {"1":3103,"3":27,"4":0,"5":1,"8":261,"10":9602,"12":1,"19":0,"20":0,"100":0,"101":15,"105":20,"106":3,"107":0}
debug 2022-05-26 22:04:09lumi.motion.ac02: unknown key 8 with value 261
debug 2022-05-26 22:04:09lumi.motion.ac02: unknown key 12 with value 1
debug 2022-05-26 22:04:09lumi.motion.ac02: unknown key 19 with value 0
debug 2022-05-26 22:04:09lumi.motion.ac02: unknown key 20 with value 0
debug 2022-05-26 22:04:09lumi.motion.ac02: unknown key 107 with value 0
debug 2022-05-26 22:04:09lumi.motion.ac02: Processed data into payload {"voltage":3103,"battery":100,"temperature":27,"state":"OFF","illuminance":15,"detection_interval":20,"motion_sensitivity":"high"}
debug 2022-05-26 22:04:09lumi.motion.ac02: Processed data into payload {"voltage":3103,"battery":100,"temperature":27,"state":"OFF","illuminance":15,"detection_interval":20,"motion_sensitivity":"high"}
info 2022-05-26 22:04:09MQTT publish: topic 'zigbee2mqtt/motion-sensor-chodba-aqara', payload '{"battery":100,"detection_interval":20,"illuminance":15,"linkquality":87,"motion_sensitivity":"high","occupancy":true,"state":"OFF","temperature":27,"update":{"state":null},"update_available":null,"voltage":3103}'
info 2022-05-26 22:04:18MQTT publish: topic 'zigbee2mqtt/motion-sensor-chodba-aqara', payload '{"battery":100,"detection_interval":20,"illuminance":15,"linkquality":87,"motion_sensitivity":"high","occupancy":false,"state":"OFF","temperature":27,"update":{"state":null},"update_available":null,"voltage":3103}'`

3)
[... cut ...]
info 2022-05-26 22:07:04MQTT publish: topic 'zigbee2mqtt/motion-sensor-chodba-aqara', payload '{"battery":100,"detection_interval":20,"illuminance":199,"linkquality":83,"motion_sensitivity":"high","occupancy":true,"state":"OFF","temperature":27,"update":{"state":null},"update_available":null,"voltage":3154}'
info 2022-05-26 22:07:15MQTT publish: topic 'zigbee2mqtt/motion-sensor-chodba-aqara', payload '{"battery":100,"detection_interval":20,"illuminance":199,"linkquality":83,"motion_sensitivity":"high","occupancy":false,"state":"OFF","temperature":27,"update":{"state":null},"update_available":null,"voltage":3154}'`

is that OKay for fixing the code? thanks

Otnow, 2022-05-26:
@heapxor, thanks, but I still can't understand the logic of high illuminance values, because I do not see the dynamics of their changes:

1. high value (**65534** presumably corresponds to **0 Lux**): `{"illuminance":131070}`

2. normal value (**16 Lux**): `{"illuminance":65552}`

3. normal value (**180 Lux**): `{"illuminance":65716}`

I expected to see different high light values in all 3 cases, but it was only in 1 case.

Ideally, we need to [sniff the traffic](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) between the sensor and the native Aqara hub in order to see the exact correspondence between the received illuminance value from the sensor and the value displayed in the _Aqara Home_ app.

Do you have such an opportunity?

TheJulianJES, 2022-05-27:
ZHA users are experiencing the same issue: https://github.com/zigpy/zha-device-handlers/pull/1541#issuecomment-1139206008

Otnow, 2022-05-27:
@TheJulianJES, thanks. Do I understand correctly that high illuminance values ​​come from the sensor only in the dark?

TheJulianJES, 2022-05-27:
@Otnow This is what it looks like to me at the moment. There are still multiple different (very large) values that are sent in the complete darkness. I'm not sure if they have a special meaning.

I'm currently trying to figure out the number above which illuminance should be set to 0 (and how and when these values are sent in general).

Otnow, 2022-05-27:
@TheJulianJES, good. Does the sensor report very low illuminance values like **1, 2, 3 Lux**?

heapxor, 2022-05-27:
@otnow to me 65534 something was reported mostly when i simulated dark/ covered sensor.

it got flipped to 65x again... there is dark at moment
other sensor reports low lx
[image]

Otnow, 2022-05-27:
@heapxor, ok, thanks.

I think the quick fix is to consider high illuminance values as **0 Lux**.

Accordingly, it is necessary to find out the upper limit, above which the values ​​are considered high.

I suggest that **1000 Lux** should be considered the upper limit, as this value is the maximum in the _Aqara Home_ app when specifying a condition in automation for a colleague [RTCGQ12LM](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html):

[image]

Does anyone have the opportunity to see this value in the _Aqara Home_ app for **RTCGQ14LM**?

TheJulianJES, 2022-05-27:
@Otnow I got 0, 3, and 4 as values (1 and 2 are probably also possible then) **with motion/attrid=274** in the dark after the substraction here: https://github.com/zigpy/zha-device-handlers/blob/4c067ac7af513daa953f0de83c70209520b15926/zhaquirks/xiaomi/aqara/motion_ac02.py#L52 (ZHA).
(For some reason it's reporting as 1 lx in Home Assistant though, but that's probably some other calculation error.)

The high values really only seem to occur when the motion sensor is in the dark without any motion for quite some time.
They seem to only be sent on the ``MotionCluster`` (instead of the ``OppleCluster`` where motion+lux is sent).

(I'm still testing)

TheJulianJES, 2022-05-27:
@Otnow 1000 lx might be a bit low for the limit. I reached ~2000 lx when just pointing a flashlight at the sensor.
Perhaps ``0xFFAA``/``65450`` makes sense (and everything above sets the illuminance to 0)

Anto79-ops, 2022-05-27:
Try putting your device directly in the Sun to get the full spectrum of visible light. Flashlights can give narrow light spectrums especially if they're led-based. 

I do this myself but my devices are already mounted and that sticky tape is so sticky that I'm afraid that I might ruin my drywall or the device itself LOL

Otnow, 2022-05-27:
@TheJulianJES, thanks. It turns out that manufacturer has increased the upper limit of illuminance for **RTCGQ14LM**.

To clarify the upper limit of illuminance, I additionally [asked a question on Reddit](https://www.reddit.com/r/homeautomation/comments/uz4nif/what_is_the_upper_limit_of_illumination_for_aqara/).

I also found this screenshot in [one of the reviews](https://notenoughtech.com/home-automation/aqara-motion-sensor-p1/):

[image]

Judging by it, the upper limit of illuminance is presumably **10000 Lux**, but better to clarify.

Otnow, 2022-05-28:
On Reddit, they [answered](https://www.reddit.com/r/Aqara/comments/uz4evb/comment/iabt1ai/) that for **RTCGQ14LM**, the maximum illuminance value in the automation condition in the _Aqara Home_ app is **1000 Lux**, the same as for **RTCGQ12LM**.

Considering that **RTCGQ14LM** is actually capable of measuring values above 1000 Lux, it is erroneous to focus on any boundary values ​​in the _Aqara Home_ app and it becomes possible to set the upper limit only experimentally.

Does anyone have any new additional information on the current issue?

TheJulianJES, 2022-05-29:
I currently have every value higher than ``0xFFFA``/``65530`` set the illuminance to ``0`` and it seems to work well so far.
I'd probably use a smaller limit like  ``0xFFAA``/``65450`` though (or even smaller).

See https://github.com/zigpy/zha-device-handlers/pull/1587#discussion_r883765285 for reference

Otnow, 2022-05-29:
@TheJulianJES, thanks.

Created a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4299) to fix the current problem and set **65000 Lux** as the upper limit of illuminance to stay away from the high values.

Anto79-ops, 2022-05-29:
FWIW, initally even the aqara home app had the same issue of high illuminance reporting. Tried to find there release notes of the Aqara Home app, but couldn't find them.

https://www.reddit.com/r/Aqara/comments/szh4dc/aqara_p1_motion_sensor_launches_in_china/iaeorm5/?context=3

Otnow, 2022-05-29:
@Anto79-ops, thanks. This looks like a software bug in the sensor's firmware, which is likely to be fixed in the new version.

closed 2022-05-30 by Koenkk

pleaseRedo, 2022-06-02:
I ordered 7 p1 sensors, and 3 of them from(aqara home apps) are reporting the same non-sense lux readings.  Guess it is firmware's issue.

phananhtai2k4, 2025-12-16:
1c 5f 11 9c 0a f7 00 41 30 01 21 77 0c 03 28 1c
04 21 00 00 05 21 01 00 08 21 0a 01 0a 21 00 00
0c 20 01 13 20 00 14 20 00 64 10 00 65 21 6a 02
69 20 1e 6a 20 02 6b 20 00
This is the ZCL message I received from the sensor. For key 101 (0x65), I always get a very small value, always less than 1000 even though the environment is very bright. This makes sense since it is of type uint16.

But why does the Zigbee2MQTT code look like this?

js
case "101":
    ...
    else if (["RTCGQ12LM", "RTCGQ14LM", "RTCGQ15LM"].includes(model.model)) {
        // Sometimes RTCGQ14LM reports high illuminance values in the dark
        assertNumber(value);
        const illuminance = value > 65000 ? 0 : value;
        payload.illuminance = illuminance;
    }
I see that the value of key 101 cannot be greater than 65000. Could someone please help me understand this?