Koenkk/zigbee2mqtt #12597: TuYa TS0601_dimmer stopped dimming

galzil, 2022-05-26:
### What happened?

After updating to the latest firmware my dimmer stopped dimming as it did before, it still goes on and off but changes to the brightness level make no affect on the light itself.

### What did you expect to happen?

Changing brightness level in Z2M changes the actaul brightness

### How to reproduce it (minimal and precise)

Zigbee Model TS0601
Zigbee Manufacturer _TZE200_dfxkcots
Description Zigbee smart dimmer
Changing brightness level

### Zigbee2MQTT version

1.25.1-dev commit: f79fb16e

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

Debug 2022-05-26 09:51:57Received Zigbee message from '0x842e14fffe13ceae', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Debug 2022-05-26 09:52:01Received Zigbee message from '0x842e14fffe13ceae', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0
Debug 2022-05-26 09:52:04Received Zigbee message from '0x842e14fffe13cea4', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0
Debug 2022-05-26 09:52:07Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data 'ON'
Debug 2022-05-26 09:52:08Publishing 'set' 'state' to '0x60a423fffeec5e61'
Debug 2022-05-26 09:52:08Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:08MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":69,"state":"ON"}'
Debug 2022-05-26 09:52:08Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:08MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":69,"state":"ON"}'
Debug 2022-05-26 09:52:08Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:08MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":69,"state":"ON"}'
Debug 2022-05-26 09:52:09Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:09MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":72,"state":"ON"}'
Debug 2022-05-26 09:52:09Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:09MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":69,"state":"ON"}'
Debug 2022-05-26 09:52:09Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:09MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":69,"state":"ON"}'
Debug 2022-05-26 09:52:09Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:09MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":69,"state":"ON"}'
Debug 2022-05-26 09:52:09Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:09MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":72,"state":"ON"}'
Debug 2022-05-26 09:52:10Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:10MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":72,"state":"ON"}'
Debug 2022-05-26 09:52:10Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:10MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":188,"level":591,"linkquality":72,"state":"ON"}'
Debug 2022-05-26 09:52:10Received Zigbee message from '0x842e14fffe13cea4', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Debug 2022-05-26 09:52:15Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data '{"brightness":208}'
Debug 2022-05-26 09:52:15Publishing 'set' 'brightness' to '0x60a423fffeec5e61'
Debug 2022-05-26 09:52:15Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data 'ON'
Debug 2022-05-26 09:52:15Publishing 'set' 'state' to '0x60a423fffeec5e61'
Debug 2022-05-26 09:52:15Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,51],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:15MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":208,"level":591,"linkquality":76,"state":"ON"}'
Debug 2022-05-26 09:52:15Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:15MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":208,"level":591,"linkquality":76,"state":"ON"}'
Debug 2022-05-26 09:52:15Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data '{"brightness":218}'
Debug 2022-05-26 09:52:15Publishing 'set' 'brightness' to '0x60a423fffeec5e61'
Debug 2022-05-26 09:52:15Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data 'ON'
Debug 2022-05-26 09:52:15Publishing 'set' 'state' to '0x60a423fffeec5e61'
Debug 2022-05-26 09:52:15Received Zigbee message from '0x842e14fffe13ceae', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":64}' from endpoint 1 with groupID 0
Debug 2022-05-26 09:52:15Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,90],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:15MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":218,"level":591,"linkquality":72,"state":"ON"}'
[... 19 lines cut ...]
Debug 2022-05-26 09:52:19Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:19MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":185,"level":591,"linkquality":76,"state":"ON"}'
Debug 2022-05-26 09:52:19Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data '{"brightness":107}'
Debug 2022-05-26 09:52:19Publishing 'set' 'brightness' to '0x60a423fffeec5e61'
Debug 2022-05-26 09:52:19Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data 'ON'
Debug 2022-05-26 09:52:19Publishing 'set' 'state' to '0x60a423fffeec5e61'
[... cut ...]
Debug 2022-05-26 09:52:25Received MQTT message on 'zigbee2mqtt/0x60a423fffeec5e61/set' with data 'OFF'
Debug 2022-05-26 09:52:25Publishing 'set' 'state' to '0x60a423fffeec5e61'
Debug 2022-05-26 09:52:25Received Zigbee message from '0x60a423fffeec5e61', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-05-26 09:52:25MQTT publish: topic 'zigbee2mqtt/0x60a423fffeec5e61', payload '{"brightness":254,"level":591,"linkquality":76,"state":"OFF"}'

galzil, 2022-05-29:
After re-paring the dimmer for the third time it started working again, not sure what was the problem

closed 2022-05-29 by galzil

shanelord01, 2022-08-26:
Opening this as I am having the same issue with new dimmers (first two I've ever installed) right now.

Logs in my comment here:
https://github.com/Koenkk/zigbee2mqtt/discussions/11376