Koenkk/zigbee2mqtt #12598: Custom device doesn't update exposed entity states when physical controls are used

gid204, 2022-05-26:
### What happened?

I'm adding a new 'unsupported' device which I have received. It is a white label product directly from a Chinese factory.
I plan on selling this as my own product at some point, thus the reason I'm not submitting a new device request just yet.

When I connect it to my coordinator, the only messages I see in the debug logs are:
```
Debug 2022-05-26 16:24:27Received Zigbee message from 'Dimmer', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":268513281,"imageType":513,"manufacturerCode":4417}' from endpoint 1 with groupID 0
Debug 2022-05-26 16:24:27Device 'Dimmer' requested OTA
```

The above **DEBUG** output occurs with and without fz, tz and exposes added into the converter file.

I decided to add the following converter file, utilising the pre-configured light_onoff_brightness extend file:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Dimmer'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'K-DIM-1', // Vendor model number, look on the device for a model number
    vendor: 'Kaneo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Dimmer Mechanism', // Description of the device, copy from vendor site. (only used for documentation and startup logging)

    extend: extend.light_onoff_brightness()
};

module.exports = definition;

```

Now when I access the device in my zigbee2mqtt frontend, I can control and view the exposed elements.

[image]

Here's where my issue begins.
When I change the state of the bulb via the dimmers physical button, the state in zigbee2mqtt doesn't change until I press the little blue refresh button.

If I change any exposed setting in zigbee2mqtt frontend I can see the change physically on the light bulb, as well as in my logs (immediately).

If I change the state of the bulb using the physical button, the bulb state will physically change, but I won't see the exposed entity updated nor will I see anything in logs.

Once I press the blue refresh button as per the image above, then I'll see the updated state and the output logs.

For reference here are the logs for when I turn the bulb on from zigbee2mqtt frontend and also from the dimmers physical button:

**Frontend:**
```
Debug 2022-05-26 16:53:30Received MQTT message on 'zigbee2mqtt/Dimmer/set' with data '{"state":"ON"}'
Debug 2022-05-26 16:53:30Publishing 'set' 'state' to 'Dimmer'
Info 2022-05-26 16:53:30MQTT publish: topic 'zigbee2mqtt/Dimmer', payload '{"brightness":19,"linkquality":102,"state":"ON"}'
```

**Physical Button then press refresh on frontend**
```
Debug 2022-05-26 16:55:06Received MQTT message on 'zigbee2mqtt/Dimmer/get' with data '{"state":""}'
Debug 2022-05-26 16:55:06Publishing get 'get' 'state' to 'Dimmer'
Debug 2022-05-26 16:55:06Received Zigbee message from 'Dimmer', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-05-26 16:55:06MQTT publish: topic 'zigbee2mqtt/Dimmer', payload '{"brightness":53,"linkquality":102,"state":"ON"}'
```

### What did you expect to happen?

I expected zigbee2mqtt to update the exposed entity and verify the devices state change in logs when the real world button changed the state of the device.

### How to reproduce it (minimal and precise)

1. Add device to zigbee2mqtt
2. Add custom converter file and enable it in configuration.yaml
3. Change state of device with physical button and observe in zigbee2mqtt.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 Dongle

sjorge, 2022-05-26:
You need to configure reporting for the fields you care about. This is so go all bulbs.

generally you want it for genOnOff.onOff and genLevelCtrl.currentLevel

gid204, 2022-05-28:
Thanks for that!
So I got it working by adding the reporting in the Web UI.

[image]

Could you point me in the right direction for adding this by default in the .js file? Is this something that can be done? I assume so as all my other 'supported' devices update automatically once there is a state change.....

I added the following line but it still isn't working with this code so I've obviously done something wrong.
**I'm referring to "reporting: [reporting.genOnOff, reporting.genLevelCtrl]," below:**

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Dimmer'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'K-DIM-1', // Vendor model number, look on the device for a model number
    vendor: 'Kaneo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Dimmer Mechanism', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    reporting: [reporting.genOnOff, reporting.genLevelCtrl],
    extend: extend.light_onoff_brightness()

};

module.exports = definition;

```

Cheers!

sjorge, 2022-05-28:
You can add something like this to the configure function
```
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
```

Make sure to pick the correct endpoint.

gid204, 2022-05-29:
Sorry for making you spell it out...I'm a hardware engineer ☉_☉

I've tried a few ways and keep getting errors galore...

Would you be able to show me how it would fit into the code?

Cheers!

gid204, 2022-05-29:
Thanks @sjorge for your help, I seem to have gotten it working...
Did some "reverse engineering" and just looked at some of the code for an existing device(s) https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js

Cheers

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Dimmer'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'K-DIM-1', // Vendor model number, look on the device for a model number
    vendor: 'Kaneo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Dimmer Mechanism', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.power_on_behavior, fz.ignore_basic_report, fz.ignore_genOta], // We will add this later
    toZigbee: [tz.light_onoff_brightness, tz.ignore_transition, tz.ignore_rate, tz.light_brightness_move,
        tz.light_brightness_step, tz.level_config, tz.power_on_behavior], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.light_brightness()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;

```

closed 2022-05-29 by gid204