Koenkk/zigbee2mqtt #12604: Tradfri bulbs do not retain brightness / color temp after toggle on

jperquin, 2022-05-26:
### What happened?

Caveat: Issue appears to have been reported in 2020 and was solved back then, but am experiencing this on latest version of Z2M and controller FW.

I have several Tradfri bulbs (LED2003G10) and setting brightness and color temp in HA works as expected. When I toggle the bulb off and after a few seconds, minutes or hours toggle it back on it does not maintain the previously set brightness and color_temp.

### What did you expect to happen?

Would be nice to have previous settings be maintained.

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee USB Dongle 3.0 Plus

### Debug log

``debug 2022-05-26 09:48:22: Received MQTT message on 'zigbee2mqtt/IKEA White Hal Eerste/set' with data '{"state": "ON", "brightness": 165}'
debug 2022-05-26 09:48:22: Publishing 'set' 'brightness' to 'IKEA White Hal Eerste'
info  2022-05-26 09:48:22: MQTT publish: topic 'zigbee2mqtt/IKEA White Hal Eerste', payload '{"brightness":165,"color_mode":"color_temp","color_temp":454,"linkquality":149,"power_on_behavior":"on","state":"ON","update":{"state":null},"update_available":null}'
debug 2022-05-26 09:48:24: Received MQTT message on 'zigbee2mqtt/IKEA White Hal Eerste/set' with data '{"state": "OFF"}'
debug 2022-05-26 09:48:24: Publishing 'set' 'state' to 'IKEA White Hal Eerste'
info  2022-05-26 09:48:24: MQTT publish: topic 'zigbee2mqtt/IKEA White Hal Eerste', payload '{"brightness":165,"color_mode":"color_temp","color_temp":454,"linkquality":145,"power_on_behavior":"on","state":"OFF","update":{"state":null},"update_available":null}'
debug 2022-05-26 09:48:27: Received MQTT message on 'zigbee2mqtt/IKEA White Hal Eerste/set' with data '{"state": "ON", "brightness": 50}'
debug 2022-05-26 09:48:27: Publishing 'set' 'brightness' to 'IKEA White Hal Eerste'
info  2022-05-26 09:48:27: MQTT publish: topic 'zigbee2mqtt/IKEA White Hal Eerste', payload '{"brightness":50,"color_mode":"color_temp","color_temp":454,"linkquality":145,"power_on_behavior":"on","state":"ON","update":{"state":null},"update_available":null}'``

Koenkk, 2022-05-26:
Do you have a default transition set for these lights in your `configuration.yaml`? If yes, can you try removing this and see if the issue is fixed?

jperquin, 2022-05-26:
Here's my configuration.yaml, but don't think so..

homeassistant: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: XXXX
  password: XXXX
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_c468e1c1e1c9eb119b288b4f1d69213e-if00-port0
frontend:
  port: 8099
devices:
  '0x84ba20fffe5ec292':
    friendly_name: IKEA White Hal Eerste
    color_sync: true
  '0x00158d0007e627c4':
    friendly_name: Aqara Motion Hall Tweede
  '0x003c84fffe318320':
    friendly_name: ' IKEA White Hal Tweede'
    color_sync: true
  '0x003c84fffe318b9b':
    friendly_name: IKEA White WC Eerste
advanced:
  pan_id: XX
  ext_pan_id:
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
  channel: 25
  network_key:
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
    - XX
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
device_options:
  legacy: false
zigbee_herdsman_debug: true

jperquin, 2022-05-26:
Crickey! I feel rather inept...

Turns out that somewhere along the way I had added the following line to light_profiles.csv:

group.all_lights.default,0.6,0.2,50

After removal all works as expected.

Hope this helps someone else debugging their setup at some point in the future.

closed 2022-05-26 by jperquin