Koenkk/zigbee2mqtt #12610: HA created doubled entities

michalk-k, 2022-05-26:
### What happened?

Recently after adding and renaming a new Zigbee device (Aqara Door&Window sensor) Home Assistant in fact created doubled entities for all sensors of those devices. 

For example for contact sensor following entities have been created: `binary_sensor.door_storeroom_contact` and  `binary_sensor.door_storeroom_contact_2` while `door_storeroom` was requested device name.
Their friendly names are correct.

Entities without the suffix are ghost ones. Entities with the suffix are working ones. 
What is interesting, HA doesn't show non-suffixed entities in dev tools at all. but it has to provide them via API since NodeRed lists them in nodes configuration.

I was trying to rename it a few times like to `door_storeroom2`, 3 etc. But the result is always suffixed with `_2`.
Hope the log below will help find the root cause of this unexpected transition. BTW it reminds me similar issue from the past (more than a year ago, which was resulting in the failure of the renaming device due to entity name conflict (caused by race condition during renaming). It was the HA discovery bug AFAIK

I'm using HA Home Assistant Core 2022.5.5

With regards

### What did you expect to happen?

Flawless renaming, without additional suffixes.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

0x26700700

### Adapter

ConBee2

### Debug log

```
config $ cat ./home-assistant.log | egrep "0x00158d0007ef4f39|door_storeroom" > ./log_z2m.log
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0007ef4f39 battery
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.0x00158d0007ef4f39_battery
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: binary_sensor 0x00158d0007ef4f39 contact
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0007ef4f39 temperature
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0007ef4f39 voltage
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0007ef4f39 linkquality
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.mqtt entity: binary_sensor.0x00158d0007ef4f39_contact
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.0x00158d0007ef4f39_temperature
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 battery, queuing update
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: binary_sensor 0x00158d0007ef4f39 contact, queuing update
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 temperature, queuing update
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 voltage, queuing update
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 linkquality, queuing update
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.0x00158d0007ef4f39_voltage
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.0x00158d0007ef4f39_linkquality
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0007ef4f39 voltage
2022-05-26 20:19:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0007ef4f39 linkquality
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 battery, sending update
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.mixins] Got update for entity with hash: ('sensor', '0x00158d0007ef4f39 battery') '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt/0x00158d0007ef4f39/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0007ef4f39'], 'manufacturer': 'Xiaomi', 'model': 'Aqara door & window contact sensor (MCCGQ11LM)', 'name': '0x00158d0007ef4f39', 'sw_version': '3000-0001'}, 'device_class': 'battery', 'enabled_by_default': True, 'entity_category': 'diagnostic', 'name': '0x00158d0007ef4f39 battery', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/0x00158d0007ef4f39', 'unique_id': '0x00158d0007ef4f39_battery_zigbee2mqtt', 'unit_of_measurement': '%', 'value_template': '{{ value_json.battery }}', 'platform': 'mqtt'}'
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.mixins] Ignoring unchanged update for: sensor.0x00158d0007ef4f39_battery
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: binary_sensor 0x00158d0007ef4f39 contact, sending update
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.mixins] Got update for entity with hash: ('binary_sensor', '0x00158d0007ef4f39 contact') '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt/0x00158d0007ef4f39/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0007ef4f39'], 'manufacturer': 'Xiaomi', 'model': 'Aqara door & window contact sensor (MCCGQ11LM)', 'name': '0x00158d0007ef4f39', 'sw_version': '3000-0001'}, 'device_class': 'door', 'name': '0x00158d0007ef4f39 contact', 'payload_off': True, 'payload_on': False, 'state_topic': 'zigbee2mqtt/0x00158d0007ef4f39', 'unique_id': '0x00158d0007ef4f39_contact_zigbee2mqtt', 'value_template': '{{ value_json.contact }}', 'platform': 'mqtt'}'
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.mixins] Ignoring unchanged update for: binary_sensor.0x00158d0007ef4f39_contact
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 temperature, sending update
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.mixins] Got update for entity with hash: ('sensor', '0x00158d0007ef4f39 temperature') '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt/0x00158d0007ef4f39/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0007ef4f39'], 'manufacturer': 'Xiaomi', 'model': 'Aqara door & window contact sensor (MCCGQ11LM)', 'name': '0x00158d0007ef4f39', 'sw_version': '3000-0001'}, 'device_class': 'temperature', 'enabled_by_default': True, 'name': '0x00158d0007ef4f39 temperature', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/0x00158d0007ef4f39', 'unique_id': '0x00158d0007ef4f39_temperature_zigbee2mqtt', 'unit_of_measurement': 'Â°C', 'value_template': '{{ value_json.temperature }}', 'platform': 'mqtt'}'
2022-05-26 20:19:16 INFO (MainThread) [homeassistant.components.mqtt.mixins] Ignoring unchanged update for: sensor.0x00158d0007ef4f39_temperature
2022-05-26 20:28:05 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 battery, sending update
2022-05-26 20:28:05 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: binary_sensor 0x00158d0007ef4f39 contact, sending update
2022-05-26 20:28:05 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 temperature, sending update
2022-05-26 20:28:06 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 0x00158d0007ef4f39 battery, queuing update
[... cut ...] sion': '3000-0001'}, 'device_class': 'temperature', 'enabled_by_default': True, 'name': 'door_storeroom_temperature', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/door_storeroom', 'unique_id': '0x00158d0007ef4f39_temperature_zigbee2mqtt', 'unit_of_measurement': 'Â°C', 'value_template': '{{ value_json.temperature }}', 'platform': 'mqtt'}'
2022-05-26 20:46:50 INFO (MainThread) [homeassistant.components.mqtt.mixins] Ignoring unchanged update for: sensor.door_storeroom_temperature_2
```

dg1sek, 2022-05-31:
I experience the same problem. Pretty painful to debug the configuration manually and more importantly not clear to me from the documentation what exactly the expected behaviour is and why these are created with "_2"

Koenkk, 2022-06-01:
Pushed a possible fix, could you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-10 by github-actions[bot]