Koenkk/zigbee2mqtt #12611: IKEA TRADFRI 5 button remote: no action event (Remote Firmware 2.3.080, Adapter: RaspBee)

rtitmuss, 2022-05-27:
### What happened?

No action events are reported by z2m when pressing any button on the IKEA TRADFRI 5 button remote, when using the RaspBee coordinator and 2.3.080 firmware on the tradfri remote.

### What did you expect to happen?

Action events are reported by z2m when pressing the buttons on the IKEA TRADFRI 5 button remote.

### How to reproduce it (minimal and precise)

1. Remove the tradfri remote from z2m
2. Restart z2m
3. Enable permit join
4. Press the reset button on the remote 4 times
5. Wait for the remote to be interviewed
6. Disable permit join
7. Press the buttons on the remote

The remote is shown as bound to the Coordinator as shown in the screenshot below.

[image]

Note the remote has joined the network. Availability and battery status reports are working correctly. The remote can be bound to light bulbs. The only problem is no action events are emitted when the buttons are pressed.

I have tried multiple time unbinding and rebinding the Coordinator in the frontend. This does not fix the issue.

This is working correctly with a Philips Hue Remote and IKEA TRADFRI with firmware 1.2.223.

It does not appear to be related to https://github.com/Koenkk/zigbee2mqtt/issues/5350, as I don't see the `unknown device with address` error.

I have also tried binding using mqtt as described in https://github.com/Koenkk/zigbee2mqtt/issues/9439. The binding is successful but still no action events.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

0x26400500

### Adapter

RaspBee

### Debug log

Attached zigbee-herdsman logs from steps 3-7 above:

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/8784830/z2m.log)

MattWestb, 2022-05-27:
I think DE have not porting the last fixes from Rasp/CornBee II to first gen that is needed for getting IKEA remotes working OK.
`Improve network robustness for various devices from Ikea and Osram.`
Look on DE wiki https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog

cadavor, 2022-05-29:
Hi,

I've got the same problem with one of my 4 E1810 remote that I updated 4 days ago...
No more "action" event is received.
I tried to change the battery, unpair/repair a lot of times.

The other 3 (not updated) work well.

I use a CC2531 adapter updated with lastest firmware (20211115)

AlexKay88, 2022-06-22:
Did you disable legacy integration?

To disable the click event, set `legacy: false` for this device in configuration.yaml. Example:
```
devices:
  '0x12345678':
    friendly_name: my_device
    legacy: false
```

or via web gui:
[image]

Shamane2000, 2022-06-23:
I have two of this ikea devices and also a conbee1 here. (integrated in HASSIO) 
It makes no difference if i set it to true or false. No Messages from this tradfri device. :(  
Any news regarding this?

Shamane2000, 2022-06-23:
in addition to that, it seems to not working with more ikea devices. the same behavior appear with this shortcutbutton: https://www.zigbee2mqtt.io/devices/E1812.html#ikea-e1812
under "logs" in the zigbee2mqtt frontend is also nothing reportet, if i press any key (from tradfri buttons) An aquara switch ist working well.

ministryofsillywalks, 2022-07-05:
Happening to me aswell using ezsp coordinator.
All my IKEA Remotes can be bound but send no more actions.
It pairs perfectly fine. LQI and Battery are being shown but no actions show up in the log.
Has anyone figured out whats going on?

flechaig, 2022-07-28:
I have the Conbee stick (first versiion) and the same issue with IKEA remotes :(

flechaig, 2022-07-30:
I just figured it out...
As state in zigbee2mqtt documentation, the pairing MUST take place in contact with the controller.
It may need a few tries to pair correctly, but it seems that it works well if the pairing is done as advised.

amaximchuk, 2022-08-08:
I have the same issue with the IKEA 5-buttons remote. But with the Conbee II & Deconz works well.

ahd71, 2022-08-29:
+1 on the same issue. Have a number if them and some few works, most doesn't anymore  all are on the same firmware release

lbschenkel, 2022-09-08:
Same thing here. I was migrating from deCONZ to Z2M and none of my remotes work (they are all with 2.3.014 firmware). Using a Conbee (original model, not II). Debug logs show no activity when buttons are pressed.

lbschenkel, 2022-09-08:
Follow up: this is what happens when pressing buttons:
```
2022-09-08T11:35:30.093Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-09-08T11:35:30.093Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-09-08T11:35:30.097Z zigbee-herdsman:deconz:driver query aps data indication
2022-09-08T11:35:30.099Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 59
2022-09-08T11:35:30.125Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 59 srcAddr: 0xeaf8 destAddr: 0x0 profile id: 0x104 cluster id: 0x5 lqi: 255
2022-09-08T11:35:30.125Z zigbee-herdsman:deconz:frameParser response payload: 5,124,17,31,7,1,1,13,0
2022-09-08T11:35:30.133Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":31,"manufacturerCode":4476,"commandIdentifier":7},"Payload":{"value":257,"value2":13},"Command":{"ID":7,"parameters":[{"name":"value","type":33},{"name":"value2","type":33}],"name":"tradfriArrowSingle"}},"address":60152,"endpoint":1,"linkquality":255,"groupID":0,"wasBroadcast":true,"destinationEndpoint":53}'
2022-09-08T11:35:30.133Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '60152', skipping...
```
```
2022-09-08T11:35:32.510Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-09-08T11:35:32.510Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-09-08T11:35:32.513Z zigbee-herdsman:deconz:driver query aps data indication
2022-09-08T11:35:32.518Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 61
2022-09-08T11:35:32.542Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 61 srcAddr: 0xeaf8 destAddr: 0x0 profile id: 0x104 cluster id: 0x6 lqi: 255
2022-09-08T11:35:32.542Z zigbee-herdsman:deconz:frameParser response payload: 1,32,2
2022-09-08T11:35:32.546Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":32,"manufacturerCode":null,"commandIdentifier":2},"Payload":{},"Command":{"ID":2,"parameters":[],"name":"toggle"}},"address":60152,"endpoint":1,"linkquality":255,"groupID":0,"wasBroadcast":true,"destinationEndpoint":53}'
2022-09-08T11:35:32.547Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '60152', skipping...
```
Wild guess here, but could the problem be `zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '60152', skipping...`? so the "packets" (not sure what the right terminology is) are being discarded and not processed further?

Will try to take a look at the source code to find out more.

lbschenkel, 2022-09-08:
@Koenkk : the interesting thing is that even if I rejoin the remote, and the NWK changes (as it should), all `zcl` data keeps being received with `"address":60152`. According to https://github.com/Koenkk/zigbee-herdsman/blob/v0.14.53/src/controller/controller.ts#L601, this should be the network address (NWK), which is definitely *not* what's reported by the web interface nor what's stored in `database.db`.

60152 is `0xEAF8` and this remote's IEEE address is `0x..........eaf8..`. That can't be a coincidence, can it, especially if the address never changes on rejoins? But then the packet should have been logged with `"address":"..."` since the value should have been a string to trigger lookup by IEEE address instead of NWK.

lbschenkel, 2022-09-08:
@Koenkk:

I did a packet capture and the actual source is `0x9FA9` which matches the NWK in the database:
[image]

Therefore it seems that there's a bug in the packet decoding logic in Z2M. Probably around here? https://github.com/Koenkk/zigbee-herdsman/blob/b4cdf8ba6935ca20aa6967e1a0afed32f320e64b/src/adapter/deconz/driver/frameParser.ts#L241

ahd71, 2022-09-08:
Some of my remotes work, some doesn't.same model and firmware. Haven't done any packet trace.

lbschenkel, 2022-09-08:
The packet sniffing show that the remote is sending the right command, but it's being misinterpreted by the deconz driver in Z2M and it has a wrong source address. Since Z2M can't find the device it came from, it drops the packet. That's why nothing happens and nothing gets logged.

ahd71, 2022-09-08:
Great Finding! Shouldnt be that hard to identify and fix then. Can I help in any way?

ahd71, 2022-10-09:
But problem still exist. Actually the only severe problem I have in ny large (200 devices) installation at the moment. 
If I can help out (e.g with testing) or anything else, please let me know.

ahd71, 2022-10-09:
The problem perhaps is that in z2m but in a upstream system/driver. How do we identify that so it can be addressed to the right people?

lbschenkel, 2022-10-09:
Please don't close this. I've done a lot of troubleshooting, debugging and submitted the problem at https://github.com/Koenkk/zigbee-herdsman/issues/576 but I have heard no response from the maintainers so far.

lbschenkel, 2022-10-09:
> The problem perhaps is that in z2m but in a upstream system/driver. How do we identify that so it can be addressed to the right people?

I already did. All details are submitted at https://github.com/Koenkk/zigbee-herdsman/issues/576

ahd71, 2022-10-09:
Very good analysis! Shouldn't be too hard for someone written those code parts to adress it.

pjcarly, 2022-11-16:
Happening here as well. Original Conbee stick, 5-button remote.

mozzbozz, 2023-01-05:
I have two of these Ikea Tradfri Remotes and use the Raspbee addon shield. The remotes were on really old firmware (1.2.214). Both showed only N/A on every button press. So I went and updated them without a problem to 2.3.080.

After pressing the reconfigure button (after the update; I've tried it before with no luck), they started doing *something*: Now the toggle button (the middle button) actually sends a signal. But the other buttons still don't do anything.

konstantint, 2023-01-29:
The button worked for ages, then at some point just stopped working for seemingly no reason. Firmware 2.3.014, I don't think OTA ever succeeded, although I tried.

It works with ZHA without any problems, though.

konstantint, 2023-01-29:
... more news. I managed to force an OTA on the button (firware version became 24.4.5) and after a couple of "force remove / rejoin" iterations the "toggle" action started coming through.

netweaver1970, 2023-02-03:
My 5button ikea switch was generating action events in Z2M gui but HA doesn't see them. Was this also your problem? My ultimate goal was to use ControllerX to emulate the normal behavior.

ahd71, 2023-02-03:
In my case z2m didnt show it either (so not a HA issue). But with the latest firmware in Buttons and in the zzh stick and the z2m release everything works great!

closed 2023-03-14 by github-actions[bot]

ALCD-clio75, 2023-12-22:
Any solution on this issue ?

ChrisArgyle, 2024-06-12:
I have also experienced the degradation of IKEA remotes and sensors under Z2M (listed below).  I have tried all the various fixes, tricks, and black magic suggested here with no improvement.  _**I have resolved my issues by simply falling back to deCONZ.**_  I'll post all my makes, models, and versions below for posterity:

# gateway hardware
* Libre Computer Le Potato AML-S905X-CC
* dresden elektronik RaspBee 26400500

# gateway software
* Ubuntu (aarch64) 22.04.4
* Zigbee2MQTT 1.38.0
* deCONZ 2.26.3

# devices
* IKEA TRÅDFRI E1820 (5 button remote) build 20220928 version 24.4.5
    * no actions reported
* IKEA TRÅDFRI E1812 (shortcut button) build 20230308 version 24.4.6
    * no actions reported
* IKEA TRÅDFRI E1745 (motion sensor) build 20190308 version 2.0.022
    * no occupancy reported