Koenkk/zigbee2mqtt #12614: Sonoff USB Dongle - Adapter disconnected randomly

johannes-z, 2022-05-27:
### What happened?

Zigbee2MQTT in HA stops working at random times (every couple of minutes) with this error:

```
error 2022-05-27 11:04:55: Adapter disconnected, stopping
```

I only noticed because all the devices log `unavailable` to logbook.

### What did you expect to happen?

Adapter shouldn't disconnect.

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20220507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus - CC2652P

### Debug log

info  2022-05-27 10:47:26: Connected to MQTT server
info  2022-05-27 10:47:26: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2022-05-27 10:47:26: MQTT publish: topic 'zigbee2mqtt/901/availability', payload '{"state":"online"}'
info  2022-05-27 10:47:27: Started frontend on port 0.0.0.0:8099
...
error 2022-05-27 11:04:55: Adapter disconnected, stopping
info  2022-05-27 11:04:55: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2022-05-27 11:04:55: Disconnecting from MQTT server
info  2022-05-27 11:04:55: Stopping zigbee-herdsman...

closed 2022-07-04 by github-actions[bot]

link1873, 2022-08-22:
Having same issue here

swiergot, 2022-08-22:
I also had this. The problem disappeared completely after I replaced the USB cable.

cloudcosmonaut, 2023-04-06:
I have the same issue but don't have a USB cable. Do you have any idea how to debug this? I already flashed the stick with the latest firmware. After reconnecting, it's there but then is lost for some time again. The strange thing is it worked for over a month without any issues.

gkukurin, 2023-10-28:
@cloudcosmonaut i have exactly the same issue and I'm very frustrated about it. Did you manage to solve it?

cloudcosmonaut, 2023-10-28:
Hey Goran, 

I have yet to be able to resolve this. I also have no idea how I _could_ debug this to get more info on why the disconnect keeps happening. I would gladly try things out to resolve this...

I already tried a longer USB cable, but that doesn't change anything about the disconnect behaviour.

Fougues, 2024-01-18:
same problem for me. random disconnections every month or so.

markwalet, 2024-03-14:
I had it twice today. It seems to happen more frequently right now.