Koenkk/zigbee2mqtt #12617: SONOFF BASICZBR3 does not work.

Murianwind, 2022-05-28:
### What happened?

SONOFF BASICZBR3 connected to Z2M does not work since yesterday. The switch in SONOFF BASICZBR3 works normally. If the device is deleted and repaired, an interview will be conducted normally and a connection will be established. But on/off doesn't work.

### What did you expect to happen?

On/off operation is required.

### How to reproduce it (minimal and precise)

Execute on or off .
An error log is output and there is no operation.

### Zigbee2MQTT version

1.25.1-2

### Adapter firmware version

 20211217

### Adapter

SMARTLIGHT CC2652P (SLZB-02)

### Debug log

Zigbee2MQTT:error 2022-05-28 09:06:26: Publish 'set' 'state' to '별빛' failed: 'Error: Command 0x00124b0023437c8d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 261 - 1 - 101 - 6 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-05-28 09:06:50: Publish 'set' 'state' to '욕실 환풍기' failed: 'Error: Command 0x00124b00234311b4/0 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22074 - 0 - 105 - 6 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-05-28 09:07:26: Publish 'set' 'state' to '별빛' failed: 'Error: Command 0x00124b0023437c8d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 261 - 1 - 106 - 6 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-05-28 09:07:53: Publish 'set' 'state' to '별빛' failed: 'Error: Command 0x00124b0023437c8d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 261 - 1 - 107 - 6 - 11 after 10000ms)'

Murianwind, 2022-05-28:
I am using SONOFF BASICZBR3 with the following modifications in decices.yaml.

  friendly_name: 욕실 환풍기
  homeassistant:
    switch:
      type: fan
      object_id: fan
    fan:
      name: 욕실 환풍기
      value_template: null
      state_value_template: "{{ value_json.state }}"

Koenkk, 2022-05-28:
Does the on/off fail immediately after interviewing? Can you try to force remove the device, restart z2m, repair the device and see if on/off works now?

Murianwind, 2022-05-28:
> BASICZBR3

I've already uninstalled the device, restarted Z2M and repaired the device 3 times. After repairing the device and performing the on/off after the interview is successful, an error message appears immediately.

And if one BASICZBR3 fails, the other BASICZBR3 does not operate as the same message is output for a certain period of time.

What should I do? Should the device be replaced? The device was working normally until yesterday.

Koenkk, 2022-05-29:
Can you provide the herdsman debug log starting from the point where the device is paired until on/off fails?

Murianwind, 2022-05-29:
> Can you provide the herdsman debug log starting from the point where the device is paired until on/off fails?

I think it's a hardware problem. After replacing the device where the error occurred for the first time with a new device, it returned to normal with another device that had an error.
I connected the power to the device in which the error occurred and tried to repair it, but no other log is output.
I will close this thread.

closed 2022-05-29 by Murianwind