Koenkk/zigbee2mqtt #12619: Unstable connection with WOOX 7060 irrigation Valve

nielsvlaarhoven, 2022-05-28:
### What happened?

I recentle purchased a WOOX R7060 water valve for the irrigation of my garden.
I added it to HA using the deCONZ addon, which didn't want to accept it at first but worked after pairing a few times. 
After rebooting HA or deCONZ the WOOX R7060 loses connection to the zigbee network (see attached screendump)
[image]
Even after a day or two it loses connection by itsself, the closest (stable) zigbee device is not even 2.5 meters away from the WOOX R7060.
I have tried to integrate it in ZHA to rule out deCONZ which is losing the connection, but when using ZHA it loses connection as wel.
I have to pair the device again to make it work again.

### What did you expect to happen?

the connection with the WOOX R7060 remains 

### How to reproduce it (minimal and precise)

Pair the WOOX R7060 and restart HA or deCONZ makes it lose connection.

### Zigbee2MQTT version

n.a.

### Adapter firmware version

2.15.03 / 4/21/2022

### Adapter

CONBEE II

nielsvlaarhoven, 2022-05-28:
As you can see in the attached screendump, in the mesh map it only has one active connection (with a device quite far away eventhough there are several devices much closer). 
[image]

GSzabados, 2022-05-28:
Try to use the option to pair it to a specific device. But why are you posting here an issue if you do not even use Zigbee2MQTT???

Koenkk, 2022-05-29:
Indeed not a Zigbee2MQTT issue, please create it in to Deconz issue tracker: https://github.com/dresden-elektronik/deconz-rest-plugin/issues

closed 2022-05-29 by Koenkk

nielsvlaarhoven, 2022-05-29:
Sorry for my stupidity posting it in the wrong topic. @Koenkk thnx for the tip.