Koenkk/zigbee2mqtt #12622: [New device support]: TS0601 smoke detector (TZE200_ntcy3xu1)

FHannes, 2022-05-28:
### Link

https://www.aliexpress.com/item/1005003951429372.html

### Database entry

{"id":1,"type":"EndDevice","ieeeAddr":"0x0123456789abcdef","nwkAddr":24426,"manufId":4417,"manufName":"_TZE200_ntcy3xu1","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,100],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1653739221127,"defaultSendRequestWhen":"immediate"}

### Comments

I don't have a Tuya gateway, so I haven't been able to actually use the Tuya platform to check the attributes of this device. But with some trial and error I was able to sort out a couple of properties in an external convertor. As far as I can tell, the external convertor seems to work correctly. Maybe it can be merged with the existing TS0601 convertor. When you press the test button it reports as smoke. So far I have not been able to actually trip it with smoke, so it might just be a terrible smoke sensor...

Compared to other TS0601 smoke sensors, this one does not seem to support a battery level, or at least not as far as I could tell. It does have a low battery state property which seems to be in line with the product description. For the low battery state, I was only able to reproduce the values 0 for low battery and 2 with full batteries. Maybe there is an intermediate state 1, but I have not seen that occur so far. The device has a tamper switch on the back that is pressed when it is locked into the mounting plate.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 4:
                    result.tamper = value;
                    break;
                case 14:
                    result.battery_low = value === 0;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['ntcy3xu1'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ntcy3xu1'}],
    model: 'TS0601_smoke2',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [],
    exposes: [e.smoke(), e.tamper(), e.battery_low()],
};

module.exports = definition;
```

homuser2003, 2022-05-29:
Hi, 

we seem to look for something quite similar - I have AC powered TS0601 detectors. I adopted your converter and it helped me a bit, thanks! My difference is that I have battery state "0-100 %" on DP 15 while you seem to have low and not low only. Can you give me a hint what to edit but 15 instead of your dp 14? I erased ypur entries for dp 4 since I don´t have any "tamper"...

I also seem to have "remote muffling" on dp 16 which I guess is toZigbee, but no clue what data to sent to dp 16...

Your somke state is dp 1 also, right? 

Thanks!

FHannes, 2022-05-30:
> Hi,
>
> we seem to look for something quite similar - I have AC powered TS0601 detectors. I adopted your converter and it helped me a bit, thanks! My difference is that I have battery state "0-100 %" [...]

Your sensor sounds like what the current implementation of the convertor for the TS0601 supports. You'll probably be better off copying that implementation and adding a fingerprint with your manufacturer. Though this dp 16 is also not present there. Smoke is dp 1, yes.

ov1d1u, 2022-06-03:
My sensor variant doesn't seem to have a muffling dp, but I got a gas sensor with that, and it's also on dp 16. Try this modified converter, maybe the data matches the one for this sensor?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 4:
                    result.tamper = value;
                    break;
                case 14:
                    result.battery_low = value === 0;
                    break;
                case 16:
                	result.muffling = value;
                	break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['ntcy3xu1'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ntcy3xu1'}],
    model: 'TS0601_smoke2',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [
        {
            key: ['muffling'],
            convertSet: async (entity, key, value, meta) => {
                await tuya.sendDataPointBool(entity, 16, value);
            }
        }
    ],
    exposes: [e.smoke(), e.tamper(), e.battery_low(), exposes.binary('muffling', ea.STATE_SET, true, false)],
};

module.exports = definition;
```

SamMousa, 2022-07-02:
Did you guys test these converters btw? I just put my detector above some smoke and while the siren works for sure, there were no state changes in zigbee2mqtt.

FHannes, 2022-07-03:
> Did you guys test these converters btw? I just put my detector above some smoke and while the siren works for sure, there were no state changes in zigbee2mqtt.

I have tested it when I wrote it and everything seemed to be working fine.

SamMousa, 2022-07-03:
Hmm, then it seems we have different models identifying with the exact same manufacturer name.

homuser2003, 2022-07-09:
> Hi.
> I recently receive my TS0601 smoke detector and it shows on my HA as Not supported.
> Can you please help me?
> Thanks.
> Miguel

Hi,

 

I have no experience with HA but for Zigbee2MQTT this converter woks just fine for me:

 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    R7049_status: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp; // First we get the data point ID
                const value = tuya.getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 15:
                    result.battery = value;
                    break;
                case 16:
                    result.silence_siren = value;
                    break;
                case 20: {
                    const alarm = {0: true, 1: false};
                    result.alarm = alarm[value];
                    break;
                }
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:Woox Smoke Detector: NOT RECOGNIZED DP #${
                        dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
};

const tzLocal = {
    R7049_silenceSiren: {
        key: ['silence_siren'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 16, value);
        },
    },
    R7049_alarm: {
        key: ['alarm'],
        convertSet: async (entity, key, value, meta) => {
        key: ['alarm'],
        convertSet: async (entity, key, value, meta) => {
            const linkAlarm = {true: 0, false: 1};
            await tuya.sendDataPointEnum(entity, 20, linkAlarm[value]);
        },
    },
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_5d3vhjro'}],
        model: 'SA12IZL',
        vendor: 'TuYa',
        description: 'Smart smoke alarm',
        meta: {timeout: 30000, disableDefaultResponse: true},
        fromZigbee: [fzLocal.R7049_status, fz.ignore_tuya_set_time, fz.ignore_time_read],
        toZigbee: [tzLocal.R7049_silenceSiren, tzLocal.R7049_alarm,tz.tuya_data_point_test,],
        exposes: [e.battery(),
            exposes.binary('smoke', ea.STATE, true, false).withDescription('Smoke alarm status'),
            exposes.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state'),
            exposes.binary('alarm', ea.STATE_SET, true, false).withDescription('Alarm enable'),
            exposes.binary('silence_siren', ea.STATE_SET, true, false).withDescription('Silence siren')],
        onEvent: tuya.onEventsetTime,
    },
];
 

 

Hope this helps!

 

Most likely you have to change the manufacturerName. Put this in a seperate .js file, add the file to the configuration.yaml (see doku) and it might work...

 
 

Gesendet: Dienstag, 05. Juli 2022 um 18:44 Uhr

homuser2003, 2022-07-09:
Yes, work for me thanks!!

 
 

Gesendet: Sonntag, 03. Juli 2022 um 14:41 Uhr

Impact123, 2022-09-10:
Unstale.

Koenkk, 2022-09-10:
I've added the converter from @homuser2003 so the `_TZE200_5d3vhjro` will be supported now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Impact123, 2022-09-10:
@Koenkk Did you forget the `_TZE200_ntcy3xu1` or are they different?

Koenkk, 2022-09-11:
@Impact123 added `_TZE200_ntcy3xu1`!

Dunkelschunkel, 2022-09-13:
I also just purchased 2 of those little discs. Switched to the developer latest.

Impact123, 2022-09-15:
`_TZE200_ntcy3xu1` is not compatible with homuser2003's converter, FHhannes' works fine.

Dunkelschunkel, 2022-09-15:
@Impact123 @FHannes I am confused and astonished at the same time. Everything is working after adding the external converter. Thanks a lot.

Dunkelschunkel, 2022-10-01:
Mhh, I wonder, can we get a shiny device icon when using an external converter?

Koenkk, 2022-10-03:
To confirm; the converter in the OP (https://github.com/Koenkk/zigbee2mqtt/issues/12622#issue-1251618774) works correctly?

ov1d1u, 2022-10-03:
I can confirm that @FHannes's convertor is working correctly with _TZE200_ntcy3xu1. Not sure if it supports all the features, but at least smoke, tamper and battery_low are properly reported.

Dunkelschunkel, 2022-10-03:
> 
Same here. Tamper, smoke battery low arereportet

Koenkk, 2022-10-04:
Great, can you update to z2m 1.28.0 and check if it also work with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const utils = require('zigbee-herdsman-converters/lib/utils');
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ntcy3xu1']),
    model: 'TS0601_smoke_2',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.smoke(), e.tamper(), e.battery_low()],
    meta: {
        tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.trueFalse],
            [4, 'tamper', tuya.valueConverter.trueFalse],
            [14, 'battery_low', tuya.valueConverter.trueFalse],
        ],
    },
};

module.exports = definition;
```

Dunkelschunkel, 2022-10-04:
Detector gets recognized as TS0601_smoke_2 but zb2mqtt throws exceptions when either tamper or smoke are triggered.
error 2022-10-04 18:44:46Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of false,true}
error 2022-10-04 18:45:28Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of false,true}
error 2022-10-04 18:45:41Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of false,true}
error 2022-10-04 18:45:41Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of false,true}
error 2022-10-04 18:45:43Exception while calling fromZigbee converter: Value '1' is not allowed, expected one of false,true}
error 2022-10-04 18:45:43Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of false,true}

ov1d1u, 2022-10-04:
Same for me, I'm getting

```
Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of false,true}
```

when opening the cover of the smoke detector.

Koenkk, 2022-10-04:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/12622#issuecomment-1267240038

ov1d1u, 2022-10-04:
This one looks good, but I also had to change

```
[4, 'tamper', tuya.valueConverter.trueFalse]
```
to
```
[4, 'tamper', {from: (v) => v}],
```

so it shows Clear/Tampered instead of 1/0 for `tamper`.

Koenkk, 2022-10-04:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/12622#issuecomment-1267240038, can you check again?

ov1d1u, 2022-10-04:
I just checked, all good.

Koenkk, 2022-10-04:
Can you check if it also provides a value for battery with https://github.com/Koenkk/zigbee2mqtt/issues/12622#issuecomment-1267240038 ?

ov1d1u, 2022-10-04:
It says `Null` for `battery`. At least I can confirm that `battery_low` goes from `false` to `true` at about 2.4V.

Koenkk, 2022-10-05:
@ov1d1u thanks for checking.

Integrated the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-05 by Koenkk

lied, 2023-02-15:
Has somone succeeded to silence an alarm or to put siren into on/off with TZE200_ntcy3xu1?
I'm not sure if this is supported by the device, but without, I don't see much benefit of the device, as my plan was to use it as siren or connected alarms. So if one smoke detector alarms, the others do the same.
If it's not possbile, PM me, I have 8 devices to sell. :-D

adi90x, 2023-03-04:
> Has somone succeeded to silence an alarm or to put siren into on/off with TZE200_ntcy3xu1? I'm not sure if this is supported by the device, but without, I don't see much benefit of the device, as my [...]

Did you succed ?  Mine only report smoke not even battery...

almirus, 2023-04-12:
The device is often unavailable. Is this only a problem for me?

dspalu32, 2023-08-27:
> The device is often unavailable. Is this only a problem for me?

I see the same thing on a _TZE204_ntcy3xu1: starts off OK with periodic check-ins until about 1 day since the last manual event (tamper / smoke / power up) and is then flagged as 'offline'  by z2m after the default 25 hours.

Digging around, I found this: https://community.hubitat.com/t/release-tuya-zigbee-smoke-detector/104159

"ver. 1.1.0 2023-04-07 - extended tuyaMagic (hopefully activates check-in every 4 hours); added capability 'Health Check'; added ping() command and rtt measurement;"

...which seems to be a write to attribute 0xffde: https://raw.githubusercontent.com/kkossev/Hubitat/main/Drivers/Tuya_Zigbee_Smoke_Detector/Tuya_Zigbee_Smoke_Detector.groovy

I'm lost as to what to try next. Is it anything to do with this: https://github.com/Koenkk/zigbee2mqtt/discussions/12305

dspalu32, 2023-09-13:
Follow-up for posterity, as this thread shows up on a Google search for ntcy3xu1:

TLDR: it seems that pairing the smoke alarm with the baseplate attached (no tamper) might be key, but I don't have proof.

I tried lots of things to get this alarm to work relaiably. It would make approx eight 4-hour heartbeat/check-ins before going offline. Triggering the tamper microswitch would bring it back online, but triggering with actual smoke or test wouldn't. Useless!

I wrote a custom External Converter to make the 0xffde attribute write after the default Tuya Magic reads. Value = 0xd seems to be the original packet-logged value with a Tuya hub, but I've found plenty of references to value=0x13 too - with no source reference. In any case, this custom converter had no effect with either value. Heartbeats still stopped after ~32 hours.

Leaving the unit in "tamper" mode does keep the 4-hour heartbeat running, but also prevents low-power idle. Battery drain is 3mA and drains cells within a few days. You can hear a DCDC converter squealing inside the alarm.

Next, the other three identical alarms I ordered arrived. All work OK(!) and give continuous heartbeats, with the default converter and no 0xffde attribute write.

So, what's different? The first alarm is on a z2m instance with a remote USB-ser2net EZSP device. The next three alarms were on an identical z2m instance with a local identical EZSP device. Swapping a 'good' alarm with the 'bad' one in another location: all now work OK. The only thing I can think that's different is that I paired the alarm while the base plate was attached. Crazy.

floco, 2023-10-17:
Have the same issue with TS0205, going offline, seems like they don't send an heartbeat. 
Only way to put them back online is to press on the physical button.
Now I know why those detectors were so cheap...
Anyone found a solution ?

almirus, 2023-10-17:
I think the device is functioning properly, it just rarely sends a heartbeat to save battery. However, if smoke occurs, it will instantly send a status update. You can adjust the time when the device is considered Offline here

[image]

floco, 2023-10-17:
Thanks both for your insight :-) I will try your recommendation and see if it helps.