Koenkk/zigbee2mqtt #12630: zzh! Adaptor Becomes Unresponsive after a period of time

adamrutt, 2022-05-29:
### What happened?

I have been using Zigbee2MQTT successfully and happily for years and I have a zzh! Adaptor plugged into an Intel NUC running Home Assistant and I use the Zigbee2MQTT addon.
I haven't added any new devices recently and can't think of any changes to my setup recently either.

Things I have tried:
- Restarting the host Intel NUC.
- Switching to the dev version of Zigbee2MQTT.
- Trying different firmwares for the zzh! stick, current zStack stable and dev. 

I thought my issue might relate to this issue #12439 so I also tried the firmware provided in that issue but it didnt resolve my problem.

### What did you expect to happen?

Zigbee2MQTT would function and run as normal.

### How to reproduce it (minimal and precise)

1. Start zigbee2mqtt
2. Allow it to start and run normally
3. After a period of time (about 10 minutes), all zigbee devices become unresponsive.
4. Error messages are shown in the web frontend.

The only way to recover is by unplugging the zzh Adaptor from the USB port and plug it in again then restart zigbee2mqtt.

### Zigbee2MQTT version

1.25.1-dev

### Adapter firmware version

20220507

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8792932/log.txt)

Koenkk, 2022-05-29:
Looks that the adapter disconnects, is anything logged to `dmesg` on the moment of the crash?

webnar, 2022-05-29:
Same issue here. Since today my Zigbee network becomes unresponsive after x minutes. It works after restarting the container but breaks and stops working within a few minutes every time. It is worked for years without any issues

adamrutt, 2022-05-30:
Managed to get the herdsman log at the time it stopped working.

2022-05-30T00:37:07.498Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-30T00:37:07.509Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,71,68,129,0,0,0,0,206,75,1,1,0,109,0,146,232,31,0,0,51,28,95,17,160,10,1,255,66,42,1,33]
2022-05-30T00:37:07.509Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,71,68,129,0,0,0,0,206,75,1,1,0,109,0,146,232,31,0,0,51,28,95,17,160,10,1,255,66,42,1,33]
2022-05-30T00:37:07.511Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [179,11,3,40,22,4,33,168,67,5,33,92,6,6,36,1,0,0,0,0,10,33,166,237,151,33,0,0,152,33,245,3]
2022-05-30T00:37:07.512Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,71,68,129,0,0,0,0,206,75,1,1,0,109,0,146,232,31,0,0,51,28,95,17,160,10,1,255,66,42,1,33,179,11,3,40,22,4,33,168,67,5,33,92,6,6,36,1,0,0,0,0,10,33,166,237,151,33,0,0,152,33,245,3]
2022-05-30T00:37:07.514Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [153,33,198,3,154,33,4,0,166,237,28,104]
2022-05-30T00:37:07.515Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,71,68,129,0,0,0,0,206,75,1,1,0,109,0,146,232,31,0,0,51,28,95,17,160,10,1,255,66,42,1,33,179,11,3,40,22,4,33,168,67,5,33,92,6,6,36,1,0,0,0,0,10,33,166,237,151,33,0,0,152,33,245,3,153,33,198,3,154,33,4,0,166,237,28,104]
2022-05-30T00:37:07.515Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 71 - 2 - 4 - 129 - [0,0,0,0,206,75,1,1,0,109,0,146,232,31,0,0,51,28,95,17,160,10,1,255,66,42,1,33,179,11,3,40,22,4,33,168,67,5,33,92,6,6,36,1,0,0,0,0,10,33,166,237,151,33,0,0,152,33,245,3,153,33,198,3,154,33,4,0,166,237,28] - 104
2022-05-30T00:37:07.515Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":19406,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":109,"securityuse":0,"timestamp":2091154,"transseqnumber":0,"len":51,"data":{"type":"Buffer","data":[28,95,17,160,10,1,255,66,42,1,33,179,11,3,40,22,4,33,168,67,5,33,92,6,6,36,1,0,0,0,0,10,33,166,237,151,33,0,0,152,33,245,3,153,33,198,3,154,33,4,0]}}
2022-05-30T00:37:07.516Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":160,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":65281,"dataType":66,"attrData":{"1":2995,"3":22,"4":17320,"5":1628,"6":[0,1],"10":60838,"151":0,"152":1013,"153":966,"154":4}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":19406,"endpoint":1,"linkquality":109,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-05-30 10:37:07: Received Zigbee message from 'Living Room Cube', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2995,"10":60838,"151":0,"152":1013,"153":966,"154":4,"3":22,"4":17320,"5":1628,"6":[0,1]}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-05-30 10:37:07: lumi.sensor_cube,lumi.sensor_cube.aqgl01: unknown key 6 with value 0,1
Zigbee2MQTT:debug 2022-05-30 10:37:07: lumi.sensor_cube,lumi.sensor_cube.aqgl01: unknown key 153 with value 966
Zigbee2MQTT:debug 2022-05-30 10:37:07: lumi.sensor_cube,lumi.sensor_cube.aqgl01: unknown key 154 with value 4
Zigbee2MQTT:debug 2022-05-30 10:37:07: lumi.sensor_cube,lumi.sensor_cube.aqgl01: Processed data into payload {"voltage":2995,"battery":100,"device_temperature":22,"current":0,"power":1013}
Zigbee2MQTT:debug 2022-05-30 10:37:07: lumi.sensor_cube,lumi.sensor_cube.aqgl01: Processed data into payload {"voltage":2995,"battery":100,"device_temperature":22,"current":0,"power":1013}
2022-05-30T00:37:07.522Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2022-05-30 10:37:07: MQTT publish: topic 'zigbee2mqtt/Living Room Cube', payload '{"action":null,"action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"angle":7.72,"battery":100,"current":0,"device_temperature":22,"last_seen":"2022-05-30T10:37:07+10:00","linkquality":109,"power":1013,"power_outage_count":null,"side":3,"temperature":20,"voltage":2995}'
Zigbee2MQTT:debug 2022-05-30 10:37:10: Received MQTT message on 'zigbee2mqtt/Bathroom Lightstrip/set' with data '{"state": "OFF", "transition": 10.0}'
Zigbee2MQTT:debug 2022-05-30 10:37:10: Publishing 'set' 'transition' to 'Bathroom Lightstrip'
Zigbee2MQTT:debug 2022-05-30 10:37:10: Publishing 'set' 'state' to 'Bathroom Lightstrip'
2022-05-30T00:37:10.332Z zigbee-herdsman:controller:endpoint Command 0x60a423fffe10f6a4/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":100}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-05-30T00:37:10.333Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x60a423fffe10f6a4:34755/1 (0,0,1)
2022-05-30T00:37:10.333Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":34755,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":132,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,79,4,0,100,0]}}
[... cut ...]
Zigbee2MQTT:warn  2022-05-30 10:40:03: Failed to ping 'Study Grow Light' (attempt 2/2, Read 0x84fd27fffea15028/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
Zigbee2MQTT:debug 2022-05-30 10:40:06: Active device 'Study Grow Light' was last seen '10.32' minutes ago.
Zigbee2MQTT:info  2022-05-30 10:40:06: MQTT publish: topic 'zigbee2mqtt/Study Grow Light/availability', payload 'offline'

adamrutt, 2022-05-30:
I got the dmesg as well, this is from the Home Assistant OS host log output a couple of minutes after it stopped working. You can see at 172 and 176 is where I unplug and replug the USB adaptor. I don't see anything of interest.

I think the bluepy errors are from some bluetooth-le devices I am monitoring, not sure about the audit messages.

[host.txt](https://github.com/Koenkk/zigbee2mqtt/files/8794627/host.txt)

webnar, 2022-05-30:
@adamrutt I notice you have the same errors that I'm seeing in my log file (SRSP - AF - dataRequest after 6000ms)

I found the following page:
https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time

I hope this resolves the issue but its still strange that it started yesterday. It was working without any issues.

webnar, 2022-05-30:
> @adamrutt I notice you have the same errors that I'm seeing in my log file (SRSP - AF - dataRequest after 6000ms)
> 
> I found the following page: https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time
> 
> I hope this resolves the issue but its still strange that it started yesterday. It was working without any issues.
> 
> Disable the USB autosuspend feature, if cat /sys/module/usbcore/parameters/autosuspend returns 1 or 2 it is enabled; to disable execute:
> 
> ```
> sed -i 's/GRUB_CMDLINE_LINUX_DEFAULT="/&usbcore.autosuspend=-1 /' /etc/default/grub
> update-grub
> systemctl reboot
> ```

Okay, that didn't work. It took some more time. But my z2m stopped responding again.
I receive this error:
Error 2022-05-30 13:29:29Publish 'set' 'state' to 'sk4_bureauRechts' failed: 'Error: Command 0x0017880106925fa1/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8798020/log.txt)
.

Koenkk, 2022-05-30:
Could you check if modemmanager is installed? https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#modemmanager-is-installed

webnar, 2022-05-30:
Hi Koen,

I do not have modemmanager on my ubuntu.

[image]

webnar, 2022-05-31:
@Koenkk, any suggestions? Is there a way to repair the database to ensure the DB is not corrupted (although its far fetched I guess).

webnar, 2022-05-31:
Hi, @Koenkk I rolled back from 1.25.1 to 1.24 and Z2M looks stable now for almost 12 hours (Since the issue occurred it "normally" got unresponsive in a few minutes).

webnar, 2022-05-31:
It just broke down again:

`2022-05-31T13:02:30.481Z zigbee-herdsman:controller:endpoint CommandResponse 0x000b57fffed70bf4/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":8449,"fileVersion":587753009,"fileOffset":14050,"dataSize":50,"data":{"type":"Buffer","data":[112,71,10,33,16,251,1,240,255,231,248,181,0,245,215,114,64,242,119,35,154,66,1,211,139,32,242,189,223,248,108,70,223,248,76,88,34,107,10,35,0,42,30,191,174,126,6,240,192,6]}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-05-31T13:02:30.481Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x000b57fffed70bf4:48576/1 (0,0,1)
2022-05-31T13:02:30.481Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":48576,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":133,"options":0,"radius":30,"len":67,"data":{"type":"Buffer","data":[25,72,5,0,124,17,1,33,49,102,8,35,226,54,0,0,50,112,71,10,33,16,251,1,240,255,231,248,181,0,245,215,114,64,242,119,35,154,66,1,211,139,32,242,189,223,248,108,70,223,248,76,88,34,107,10,35,0,42,30,191,174,126,6,240,192,6]}}
2022-05-31T13:02:30.481Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,77,36,1,192,189,1,1,25,0,133,0,30,67,25,72,5,0,124,17,1,33,49,102,8,35,226,54,0,0,50,112,71,10,33,16,251,1,240,255,231,248,181,0,245,215,114,64,242,119,35,154,66,1,211,139,32,242,189,223,248,108,70,223,248,76,88,34,107,10,35,0,42,30,191,174,126,6,240,192,6,105]
2022-05-31T13:02:30.498Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-05-31T13:02:30.498Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-05-31T13:02:30.498Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-05-31T13:02:30.498Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-05-31T13:02:30.498Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-31T13:02:30.504Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,133,67]
2022-05-31T13:02:30.504Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,133,67]
2022-05-31T13:02:30.504Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,133] - 67
2022-05-31T13:02:30.504Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":133}
2022-05-31T13:02:30.504Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-31T13:02:30.520Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,192,189,0,255]
2022-05-31T13:02:30.520Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,192,189,0,255]
2022-05-31T13:02:30.521Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [192,189,0] - 255
2022-05-31T13:02:30.521Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48576,"relaycount":0,"relaylist":[]}
2022-05-31T13:02:30.521Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-31T13:02:30.582Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,37,68,129,0,0,25,0,192,189,1,13,0,0,0,19,154,75,0,0,17,1,73,3,0,124,17,1,33,49,102,8,35,20,55,0,0,63,192,189,29,93]
2022-05-31T13:02:30.582Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,37,68,129,0,0,25,0,192,189,1,13,0,0,0,19,154,75,0,0,17,1,73,3,0,124,17,1,33,49,102,8,35,20,55,0,0,63,192,189,29,93]
2022-05-31T13:02:30.582Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 37 - 2 - 4 - 129 - [0,0,25,0,192,189,1,13,0,0,0,19,154,75,0,0,17,1,73,3,0,124,17,1,33,49,102,8,35,20,55,0,0,63,192,189,29] - 93
2022-05-31T13:02:30.582Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":48576,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":4954643,"transseqnumber":0,"len":17,"data":{"type":"Buffer","data":[1,73,3,0,124,17,1,33,49,102,8,35,20,55,0,0,63]}}
2022-05-31T13:02:30.583Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":73,"manufacturerCode":null,"commandIdentifier":3},"Payload":{"fieldControl":0,"manufacturerCode":4476,"imageType":8449,"fileVersion":587753009,"fileOffset":14100,"maximumDataSize":63},"Command":{"ID":3,"response":5,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35},{"name":"fileOffset","type":35},{"name":"maximumDataSize","type":32}],"name":"imageBlockRequest"}},"address":48576,"endpoint":1,"linkquality":0,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
2022-05-31T13:02:30.584Z zigbee-herdsman:controller:log Skipping command 'imageBlockRequest' because it is missing from the lookup
2022-05-31T13:02:30.584Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-31T13:02:30.588Z zigbee-herdsman:controller:endpoint CommandResponse 0x00178801033b78d9/11 genOta.imageBlockResponse({"status":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16785664,"fileOffset":203580,"dataSize":50,"data":{"type":"Buffer","data":[193,44,165,109,57,142,27,14,149,177,49,242,251, [... cut ...]
2022-05-31T13:03:08.710Z zigbee-herdsman:controller:endpoint Read 0x84ba20fffe80ca3c/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-05-31T13:03:08.710Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x84ba20fffe80ca3c:3061/1 (0,0,6)
`

Koenkk, 2022-05-31:
@webnar I still expect that some other program on your computer is messing with the adapter, can you try to stop some other services on the computer and see if that helps

webnar, 2022-05-31:
@Koenkk hi thanks for the reply. I have the same setup for years without any issues. The only thing is that we had a power outage sunday night. Next day i powered the nas and the issues started. But the other devices (zwave and rfxcom) are working fine. The atrange thing is zigbee is working fine for x minutes and yesterday i rolled back to 1.24 after that ot worked almost 12 hours.

closed 2022-06-01 by adamrutt

webnar, 2022-06-01:
@Koenkk I followed the example of @adamrutt and rebuild my network with a new key and pan ID. And it seems to work again for almost 10 hours straight on.

pannal, 2022-06-23:
FYI, I had my zzh! in a similarly (but not same) wonky state recently. Got the new case for it that allows the little button to be accessed without opening the device. So I proceeded to unplug the stick and swap for the new case.

Seems like those zzh! devices are a little finicky with being pulled out of the USB live (or maybe I wiggled it by accident); nevertheless, after I did that, z2m errored approximately every 1-2 minutes, reconnecting to MQTT, leaving a socket error in its logs, re-reporting on/off for all devices to HA, essentially flashing all the lights in the flat.

No dmesg output, nothing out of the ordinary. Tried reflashing, different USB ports, rebooting the Ubuntu machine, to no avail.

**The only thing that helped was unplugging the zzh! stick for a couple of minutes (not just seconds) and plugging it back in.**

Works again, as if nothing ever happened.

Maybe this helps other zzh! users.