Koenkk/zigbee2mqtt #12633: CC2652P - network commissioning timed out (bdbComissioningNotifcation - {"status":1})

igvk, 2022-05-29:
### What happened?

The USB stick is CC2652P.
I've uploaded firmware CC1352P2_CC2652P_launchpad_coordinator_20220219.

Pan id is set to 6757.
Having started z2m, I get network commissioning error.

zigbee-herdsman debug log is attached
What can be done to troubleshoot this problem?

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

CC2562P USB Stick (noname)

### Debug log

```
2022-05-29T14:21:36.351Z` zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6757,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/var/lib/zigbee2mqtt/default/database.db","databaseBackupPath":"/var/lib/zigbee2mqtt/default/database.db.backup","backupPath":"/var/lib/zigbee2mqtt/default/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-05-29T14:21:36.354Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-05-29T14:21:36.367Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2022-05-29T14:21:36.370Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-05-29T14:21:36.372Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-05-29T14:21:36.379Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254]
2022-05-29T14:21:36.379Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254]
2022-05-29T14:21:36.380Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [2,97,1,89,6,61]
2022-05-29T14:21:36.381Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2022-05-29T14:21:36.382Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2022-05-29T14:21:36.384Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2022-05-29T14:21:36.386Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-29T14:21:36.386Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-05-29T14:21:36.386Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-05-29T14:21:36.389Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2022-05-29T14:21:36.389Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2022-05-29T14:21:36.390Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2022-05-29T14:21:36.390Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2022-05-29T14:21:36.390Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-29T14:21:36.390Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2022-05-29T14:21:36.390Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2022-05-29T14:21:36.393Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254]
2022-05-29T14:21:36.393Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254]
2022-05-29T14:21:36.393Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [10,97,2,2,1,2,7]
2022-05-29T14:21:36.394Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7]
2022-05-29T14:21:36.394Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [1,59,137,52,1,0,233]
2022-05-29T14:21:36.394Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7,1,59,137,52,1,0,233]
2022-05-29T14:21:36.394Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,59,137,52,1,0] - 233
2022-05-29T14:21:36.395Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219}
2022-05-29T14:21:36.395Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-29T14:21:36.396Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16
2022-05-29T14:21:36.396Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219})
2022-05-29T14:21:36.397Z zigbee-herdsman:adapter:zStack:startup beginning znp startup
2022-05-29T14:21:36.399Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {}
2022-05-29T14:21:36.399Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37]
2022-05-29T14:21:36.401Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254]
2022-05-29T14:21:36.401Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254]
2022-05-29T14:21:36.402Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [8,97,4,211,43,203,36,0,75,18,0]
2022-05-29T14:21:36.402Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,8,97,4,211,43,203,36,0,75,18,0]
2022-05-29T14:21:36.404Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [35]
2022-05-29T14:21:36.404Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,8,97,4,211,43,203,36,0,75,18,0,35]
2022-05-29T14:21:36.404Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 8 - 3 - 1 - 4 - [211,43,203,36,0,75,18,0] - 35
2022-05-29T14:21:36.405Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - getExtAddr - {"extaddress":"0x00124b0024cb2bd3"}
2022-05-29T14:21:36.405Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... 476 lines cut ...]
2022-05-29T14:21:41.655Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,111,5,0,107,254,3,79,128,1,2,4,203]
2022-05-29T14:21:41.655Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 15 - 5 - [0] - 107
2022-05-29T14:21:41.656Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- APP_CNF - bdbStartCommissioning - {"status":0}
2022-05-29T14:21:41.656Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,79,128,1,2,4,203]
2022-05-29T14:21:41.656Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 15 - 128 - [1,2,4] - 203
2022-05-29T14:21:41.657Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- APP_CNF - bdbComissioningNotifcation - {"status":1}
2022-05-29T14:21:41.658Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-29T14:21:41.862Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254]
2022-05-29T14:21:41.862Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254]
2022-05-29T14:21:41.863Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [1,69,192,8,140]
2022-05-29T14:21:41.864Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,192,8,140]
2022-05-29T14:21:41.864Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 192 - [8] - 140
2022-05-29T14:21:41.864Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8}
2022-05-29T14:21:41.865Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-29T14:21:42.164Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2022-05-29T14:21:42.164Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2022-05-29T14:21:44.217Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254]
2022-05-29T14:21:44.217Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,254]
2022-05-29T14:21:44.218Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [6,65,128,0,2,1,2,7,1,192]
2022-05-29T14:21:44.219Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,65,128,0,2,1,2,7,1,192]
2022-05-29T14:21:44.219Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
2022-05-29T14:21:44.219Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
2022-05-29T14:21:44.219Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

```

igvk, 2022-06-13:
Seems that the stick resets after StartComissioning.

Is it normal behavior?
How to make it work?

Koenkk, 2022-06-14:
This means a network is already running on this channel/panid, try a different channel or panid.

igvk, 2022-06-14:
There is no network on that channel and panid.
I know that because I specifically replaced the channel & changed panid sequentially.
The result was the same.

Then I tried flashing different firmware versions going back - and to my surprise it worked after several revisions with Z-Stack_Home_3.x.0-20210708.
Z1m started normally, and I changed channel & panid to starting values - and it continued to work.

After that, I flashed firmware 20211217 to check - and it failed again.
Reverted to 20210708 - all functioning as expected.

I wonder, what was changed in this version to have such behavior?

Koenkk, 2022-06-14:
Strange! We previously saw such behaviour when the firmware was not correctly flahsed, flashing with another tool solved the problem. See second point of https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel

igvk, 2022-06-14:
This is indeed strange.
As for flashing with different tools - I tried both Flash Programmer 2 & cc2538-bsl on Windows (with the latest firmware).
Down/up-grading software versions later was done on Raspberry Pi with cc2538-bsl only.

If it was incorrect flashing - I don't understand why the results are so reproducible.
When I thought it was bad flashing, I flashed the same version of firmware several times, it always verifies ok.
And the results are consistently the same.
It works only with 20210708 (I wasn't trying earlier versions).
Upgrading the firmware to a later version breaks it, downgrading again - fixes it.

It could be some kind of faulty flash.
But how does it verify correctly? And how does it work with that revision of firmware then?

Koenkk, 2022-06-15:
Does it also happen when no backup is restored? To test:
 - reflash stick
 - rename z2m data folder to `data_backup`
 - start z2m

To restore the backup:
- reflash stick
- rename z2m data folder back to data

closed 2022-07-24 by github-actions[bot]