Koenkk/zigbee2mqtt #12639: [New device support]: Aqara Curtain Motor E1 (ZNCLBL01LM) by Xiaomi

keith-kl, 2022-05-30:
### Link

https://www.aqara.com/cn/productDetail/curtain-controller-e1

### Database entry

{"id":6,"type":"EndDevice","ieeeAddr":"0x54ef4410004eef49","nwkAddr":32651,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.curtain.acn003","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,64704,258,10],"outClusterList":[3,25,64704,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.curtain.acn003"}},"aqaraOpple":{"attributes":{"238":6168,"247":{"type":"Buffer","data":[1,33,184,11,3,40,36,5,33,1,0,10,33,0,0,12,32,1,13,35,24,24,0,0,100,32,2,101,32,192,102,32,131,103,32,16,104,33,24,1,105,32,1,106,32,0,107,32,0]},"1026":1,"1027":16,"1034":192,"1035":3310,"1055":0,"1056":2,"1057":2,"1061":2,"1062":2,"1064":1,"1065":2,"1071":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":192,"batteryVoltage":32}},"closuresWindowCovering":{"attributes":{"configStatus":11,"currentPositionLiftPercentage":0}}},"binds":[{"cluster":64704,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b7ac52","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":24,"hwVersion":1,"dateCode":"Apr 22 2022","swBuildId":"0.0.0_0024","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1906590817},"lastSeen":1653895783338,"defaultSendRequestWhen":"immediate"}

### Comments

I have recently got a new curtain motor from Aqara which is only released in China at the moment. I tried to follow the simliar product Aqara roller shade companion E1 (ZNJLBL01LM) to create an external converter but it seems that the clusters are different.

The roller shade motor contains cluster "genAnalogOutput" and "genMultistateOutput" but they do not exist in the new curtain motor device.  I have exposed the "state" and "position" attribute like other curtain motor devices, but when I tried to refresh the state/position, the follow exception is thrown.

`Zigbee2MQTT:debug 2022-05-30 17:04:31: Received MQTT message on 'zigbee2mqtt/0x54ef4410004eef49/get' with data '{"state":""}'
Zigbee2MQTT:debug 2022-05-30 17:04:31: Publishing get 'get' 'state' to '0x54ef4410004eef49'
2022-05-30T09:04:31.248Z zigbee-herdsman:controller:endpoint Read 0x54ef4410004eef49/1 aqaraOpple([1055], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false})
2022-05-30T09:04:31.249Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x54ef4410004eef49:32651/1 (0,0,1)
2022-05-30T09:04:31.251Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":32651,"destendpoint":1,"srcendpoint":1,"clusterid":64704,"transid":7,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[20,95,17,5,0,31,4]}}
2022-05-30T09:04:31.252Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,139,127,1,1,192,252,7,0,30,7,20,95,17,5,0,31,4,166]
2022-05-30T09:04:31.263Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-05-30T09:04:31.264Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-05-30T09:04:31.264Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-05-30T09:04:31.264Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-05-30T09:04:31.265Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-30T09:04:31.537Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,7,193]
2022-05-30T09:04:31.538Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,7,193]
2022-05-30T09:04:31.538Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,7] - 193
2022-05-30T09:04:31.540Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":7}
2022-05-30T09:04:31.540Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-30T09:04:31.555Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,192,252,139,127,1,1,0,153,0,19,38,120,0,0,8,28,95,17,5,1,31,4,143,139,127,29,230]
2022-05-30T09:04:31.555Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,192,252,139,127,1,1,0,153,0,19,38,120,0,0,8,28,95,17,5,1,31,4,143,139,127,29,230]
2022-05-30T09:04:31.556Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,192,252,139,127,1,1,0,153,0,19,38,120,0,0,8,28,95,17,5,1,31,4,143,139,127,29] - 230
2022-05-30T09:04:31.557Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":32651,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":153,"securityuse":0,"timestamp":7874067,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[28,95,17,5,1,31,4,143]}}
2022-05-30T09:04:31.560Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":4447,"commandIdentifier":1},"Payload":[{"attrId":1055,"status":143}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":32651,"endpoint":1,"linkquality":153,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-05-30 17:04:31: Received Zigbee message from '0x54ef4410004eef49', type 'readResponse', cluster 'aqaraOpple', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-05-30 17:04:31: lumi.curtain.acn003: unknown key 1055 with value undefined
Zigbee2MQTT:debug 2022-05-30 17:04:31: lumi.curtain.acn003: Processed data into payload {}
2022-05-30T09:04:31.584Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-30T09:04:31.586Z zigbee-herdsman:controller:endpoint Read 0x54ef4410004eef49/1 aqaraOpple([1055], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'WRITE_ONLY')
Zigbee2MQTT:error 2022-05-30 17:04:31: Publish 'get' 'state' to '0x54ef4410004eef49' failed: 'Error: Read 0x54ef4410004eef49/1 aqaraOpple([1055], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'WRITE_ONLY')'
Zigbee2MQTT:debug 2022-05-30 17:04:32: Error: Read 0x54ef4410004eef49/1 aqaraOpple([1055], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'WRITE_ONLY')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.convertGet (/app/dist/util/externally-loaded.js:76:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:272:21)`

I have derived the following key and values from my obversation but i cannot go further
1055: % of open  0 is close 100 is full open
1057: 0=CLOSE, 1=OPEN, 2=stop  ??

I would like to contribute but can someone shade some lights on how I can start?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
[... cut ...]
    onEvent: async (type, data, device) => {        
        if (data.type === 'attributeReport' && data.cluster === 'genAnalogOutput') {
            await device.endpoints[0].read('genAnalogOutput', ['presentValue']);
        }
    },
    meta: {battery: {dontDividePercentage: 'false'}},
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage']);
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Otnow, 2022-05-30:
Ideally, we need to [sniff the traffic](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) between the device and the native Aqara hub in order to see the control commands sent using _Aqara Home_ app.

Do you have such an opportunity?

I also suggest changing the device name to `Aqara curtain driver E1` as it is indicated in the _Aqara Home_ app:

[image]

keith-kl, 2022-05-30:
Unfortunately, I do not have any Aqara hub since i use zigbee2mqtt in Day 1.  I think I can only do in a trial-and-error way. 
I found the model number in this website which seems to hosts the Miot spec, which I know is different from Zigbee. 
[https://home.miot-spec.com/spec?type=urn%3Amiot-spec-v2%3Adevice%3Acurtain%3A0000A00C%3Alumi-acn003%3A1](url)

Do people usually refer to this spec when they develop the converter?

Is it possible to trace the traffic by manually pulling the curtain to trigger the movement of the motor ? I can see the position attribute is being updated, but I do not know how to set the position

Otnow, 2022-05-30:
> I found the model number in this website which seems to hosts the Miot spec, which I know is different from Zigbee. [https://home.miot-spec.com/spec?type=urn%3Amiot-spec-v2%3Adevice%3Acurtain%3A0000A00C%3Alumi-acn003%3A1](url)
> 
> Do people usually refer to this spec when they develop the converter?

Yes, this information gives a rough idea of the functionality of the device.

***

> Is it possible to trace the traffic by manually pulling the curtain to trigger the movement of the motor ?

I do not know, I do not have such devices.

> I can see the position attribute is being updated, but I do not know how to set the position

Please post debug logs here with your comments so everyone can see and analyze them.

keith-kl, 2022-07-01:
I have submitted the converter several weeks ago. I guess it will be available in the coming release from the author.

closed 2022-08-09 by github-actions[bot]

chanphillip, 2022-10-07:
Just got mine yesterday and it is working great with my HA! Would be great to also have the illuminance level exposed 👀

keith-kl, 2022-10-07:
> Just got mine yesterday and it is working great with my HA! Would be great to also have the illuminance level exposed 👀

I hope so. But unfortunately I don't have a Aqara Hub so i cannot enable this from the Aqara app and test.  Will need somebody else to enhance it.  ;)