Koenkk/zigbee2mqtt #12641: Battery device: iCasa ICZB-KPD18S shows 200% battery

webnar, 2022-05-30:
### What happened?

The iCasa ICZB-KPD18S shows a battery life of 200% (switch firmware 2.7.6_r25)

[image]

### What did you expect to happen?

Showing 100% as my other switches do. (switch firmware: 2.5.3_r21)

[image]

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20211217

### Adapter

Sonoss

Koenkk, 2022-05-30:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-30 by Koenkk

webnar, 2022-05-30:
Thanks, But i'm still seeing the 200%

[image]

[image]

Koenkk, 2022-05-31:
@webnar it can take some time before the device reports a new battery value