Koenkk/zigbee2mqtt #12649: LIDL HG06336 Door sensor N/A contact, battery_low, tamper and LQI

seederp2p, 2022-05-30:
### What happened?

I'm using LIDL HG06336 Door sensor with HA 2022.5.5 and Zigbee2mqtt 1.18.1-1.

My coordinator is SONOFF Zigbee 3.0 firmware version 20220219

The device is listed but 

### What did you expect to happen?

It should show real data and not "N/A" on contact, battery_low, tamper and LQI

### How to reproduce it (minimal and precise)

I can remove the device and try again with the same result.

### Zigbee2MQTT version

1.18.1-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 firmware version 20220219

### Debug log

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[20:49:08] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[20:49:09] INFO: MQTT available, fetching server detail ...
[20:49:09] INFO: Previous config file found, checking backup
[20:49:09] INFO: Creating backup config in '/share/zigbee2mqtt/.configuration.yaml.bk'
[20:49:09] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[20:49:09] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
Zigbee2MQTT:debug 2022-05-30 20:49:12: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-05-30 20:49:12: Logging to console and directory: '/share/zigbee2mqtt/log/2022-05-30.20-49-11' filename: log.txt
Zigbee2MQTT:debug 2022-05-30 20:49:12: Removing old log directory '/share/zigbee2mqtt/log/2022-05-30.18-36-10'
Zigbee2MQTT:info  2022-05-30 20:49:12: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
Zigbee2MQTT:info  2022-05-30 20:49:12: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:debug 2022-05-30 20:49:12: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:info  2022-05-30 20:49:24: zigbee-herdsman started
Zigbee2MQTT:info  2022-05-30 20:49:24: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:debug 2022-05-30 20:49:24: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2022-05-30 20:49:24: Set transmit power to '20'
Zigbee2MQTT:info  2022-05-30 20:49:24: Currently 1 devices are joined:
Zigbee2MQTT:info  2022-05-30 20:49:24: 0x680ae2fffe0a2c21 (0x680ae2fffe0a2c21): HG06336 - Lidl Silvercrest smart window and door sensor (EndDevice)
Zigbee2MQTT:info  2022-05-30 20:49:24: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-05-30 20:49:24: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:info  2022-05-30 20:49:24: Connecting to MQTT server at mqtt://192.168.101.9:1883
Zigbee2MQTT:debug 2022-05-30 20:49:24: Using MQTT login with username: mosquitto
Zigbee2MQTT:info  2022-05-30 20:49:24: Connected to MQTT server
Zigbee2MQTT:info  2022-05-30 20:49:24: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-05-30 20:49:24: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe0a2c21', payload '{"battery":12}'
Zigbee2MQTT:debug 2022-05-30 20:49:24: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"1.18.1","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/share/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.txt","log_level":"warn","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x680ae2fffe0a2c21":{"friendly_name":"0x680ae2fffe0a2c21"}},"experimental":{"output":"json","transmit_power":20},"external_converters":[],"frontend":{"port":8099},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://192.168.101.9:1883","user":"mosquitto"},"ota":{"disable_automatic_update_check":false,"update_check_interval":10},"passlist":[],"permit_join":true,"queue":{},"serial":{"disable_led":false,"port":"/dev/ttyUSB0"},"whitelist":[]},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Allows to prevent certain attributes from being published","examples":["temperature","battery","action"],"items":{"type":"string"},"title":"Filtered attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By default this is the device ID","readOnly":true,"title":"Friendly name","type":"string"},"optimistic":{"description":"Publish optimistic state after set (default true)","title":"Optimistic","type":"boolean"},"qos":{"descritption":"QoS level for MQTT messages of this device","title":"QoS","type":"number"},"retain":{"description":"Retain MQTT messages of this device","title":"Retain","type":"boolean"},"retention":{"description":"Sets the MQTT Message Expiry in seconds, Make sure to set mqtt.version to 5","title":"Retention","type":"number"}},"required":["friendly_name"],"type":"object"},"group":{"properties":{"devices":{"items":{"type":"string"},"type":"array"},"filtered_attributes":{"items":{"type":"string"},"type":"array"},"friendly_name":{"type":"string"},"optimistic":{"type":"boolean"},"qos":{"type":"number"},"retain":{"type":"boolean"}},"required":["friendly_name"],"type":"object"}},"properties":{"advanced":{"properties":{"adapter_concurrent":{"description":"Adapter concurrency (e.g. 2 for CC2531 or 16 for CC26X2R1) (default: null, uses recommended value)","requiresRestart":true,"title":"Adapter concurrency","type":["number","null"]},"adapter_delay":{"description":"Adapter delay","requiresRestart":true,"title":"Adapter delay","type":["number","null"]},"availability_blacklist":{"items":{"type":"string"},"readOnly":true,"requiresRestart":true,"title":"Availability blacklist (deprecated, use availability_blocklist)","type":"array"},"availability_blocklist":{"description":"Prevent devices from being checked for availability","items":{"type":"string"},"requiresRestart":true,"title":"Availability Blocklist","type":"array"},"availability_passlist":{"description":"Only enable availability check for certain devices","items":{"type":"string"},"requiresRestart":true,"title":"Availability passlist","type":"array"},"availability_timeout":{"default":0,"description":"Availability timeout in seconds when enabled, devices will be checked if they are still online. Only AC powered routers are checked for availability","minimum":0,"requiresRestart":true,"title":"Availability Timeout","type":"number"},"availability_whitelist":{"items":{"type":"string"},"readOnly":true,"requiresRestart":true,"title":"Availability whitelist (deprecated, use passlist)","type":"array"},"baudrate":{"description":"Baudrate for serial p [... cut ...] "Zigbee2MQTT 1.18.1"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x680ae2fffe0a2c21","name":"0x680ae2fffe0a2c21 linkquality","state_topic":"zigbee2mqtt/0x680ae2fffe0a2c21","unique_id":"0x680ae2fffe0a2c21_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2022-05-30 20:54:12: Saving state to file /share/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2022-05-30 20:59:12: Saving state to file /share/zigbee2mqtt/state.json

seederp2p, 2022-05-31:
Update:

Using ZHA works perfectly.

So, maybe, this is a Z2M bug?

Koenkk, 2022-05-31:
You are using a very old z2m version, try updating to the latest and wait 24 hours so that the sensor can report its data.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

seederp2p, 2022-05-31:
Hi ! @Koenkk 

Ok. So I fresh installed my Z2M.

Right now is running version 1.25.1-2

Re-added my door contact sensor and got the same results:

contact
N/A

battery_low
N/A

tamper
N/A

battery
Remaining battery in %
100%

linkquality
145lqi

Koenkk, 2022-06-02:
Can you provide the data/database.db entry of this device?

seederp2p, 2022-06-03:
Hello,

After upgrading Z2M to 1.25.2-1 and today, HA to 2022.6.1, I re-added the device.

It looked like the same but I tried to reconfigure it as your said previously.

For some reason it now works.

Anyway, how can I retrieve the data/database.db inside HA?

closed 2022-07-12 by github-actions[bot]